OpenPrinting/ipp-usb #2: process "ipp-usb udev" still exists after disconnecting two USB printers

tangyanli, 2020-06-01:
Hi,

I used the ipp-usb on my mips os. And find below issue, please have a look.

Two printers are connected to PC via USB.
When I unplugged the USB cables, the process "ipp-usb udev" still exists.
When I plugged the USB again, the ipp-usb cannot works well.

[main.log](https://github.com/OpenPrinting/ipp-usb/files/4709780/main.log)
[04a9-2828-74bfc0146ffa-iR2625-2630-UFR.log](https://github.com/OpenPrinting/ipp-usb/files/4709782/04a9-2828-74bfc0146ffa-iR2625-2630-UFR.log)
[04a9-27fa-f4a997cf9d6c-MF745C-746C.log](https://github.com/OpenPrinting/ipp-usb/files/4709783/04a9-27fa-f4a997cf9d6c-MF745C-746C.log)

alexpevzner, 2020-06-01:
Hi,

I don't see "device removed" events in the `main.log`, so either log is incomplete for some reason, or this event was actually missed.

I want to ask you to perform the following experiment:
1. Diconnect all devices
2. Erase `main.log`
3. Start `ipp-dev` manually, in debug mode
4. Connect one device
5. Disconnect it
6. Terminate `ipp-usb` by pressing ctrl-c
7. Send me a resulting `main.log`

tangyanli, 2020-06-01:
Hi 
if just connect one usb printer, the process “ipp-usb udev” always exit after unplugged the usb.
I will use “ipp-usb debug” to connect two USB printers and catch the debug infor, is it OK?

But the debug info will be provided 12 hours later.

alexpevzner, 2020-06-01:
OK

tangyanli, 2020-06-02:
Sorry, alexpevzner.

I cannot reproduce this issue on Ubuntu16.04 & Debian10 amd64 architecture.
This issue probably caused by my mips os. It may have some wrong to catch the "device removed" events.
So just let it be, and please close this issue.

alexpevzner, 2020-06-02:
If you can reproduce it on MIPS, I'd be glad to fix it there.

tangyanli, 2020-06-02:
Yes, I can reproduce it on MIPS.
But from the log, there is no "device removed" event when I unplug the USB cables.
Is there some method to fix the OS issue?

alexpevzner, 2020-06-02:
May you rerun it as following:

    LIBUSB_DEBUG=1000 ipp-usb debug 2>libusb.log

tangyanli, 2020-06-02:
[libusb.log](https://github.com/OpenPrinting/ipp-usb/files/4714385/libusb.log)

The OS can catch the "device remove" event when I unplug the first USB cable, but it fails to catch the "device remove" event again when I unplug the second USB cables.

alexpevzner, 2020-06-02:
Hi @tangyanli,

thank a lot for your logs. I see, "remove" event comes to the libusb, but doesn't come to ipp-usb. I need to think a little bit about this problem, it will take some time.

alexpevzner, 2020-06-02:
Fixed (I hope). Please, try again (on MIPS). Note, I didn't set release tag yet, waiting for you feedback.

tangyanli, 2020-06-02:
All right, I will test it later. Thanks!

tangyanli, 2020-06-03:
Hi, alexpevzner

It's so great, the issue is fixed.
And the phenomenon bothered me is also fixed.

> Besides
> There is another phenomenon bothered me.
> First, unplug all USB printer,
> and set the port in the MF745C.state and iR2625.state are the same, 60000.
> If plug the MF745C firstly, and then plug the iR2625 later, the port in the iR2625.state would become 60001.
> But if plug the iR2625 firstly, and then plug the MF745C later, the port in the MF745C.state is still 60000, and ipp-usb fails to convert MF745C's USB URI to IPP URI. And no information is outputted to the MF745C.log file.

tangyanli, 2020-06-03:
Hi, alexpevzner

If it's possible, may I ask your help to supply the aarch64 binary based on Ubuntu16.04 and Debian10?

tangyanli, 2020-06-03:
Hi, alexpevzner

Could I ask you a question..

I thinked the below two URIs are equal:
http://localhost:60000/ipp/print
ipp://localhost:60000/ipp/print

Towards Canon MF745x and iR2625, driverless works well by the both URI.

Today I tested another Canon Device, the ipp URI always works. But sometime it fails to create driverless queue by the http URI.

Is there some difference between the two URI?
Waiting your reply, though I think I asked a stupid question...

alexpevzner, 2020-06-03:
Hi @tangyanli,

first of all, thank you for your feedback. You helped me to fix a real bug, which accidentally did not appear on Intel CPUs.

Regarding aarch64, I've tried to enable this build, but it failed with different reasons for Debian 10 and for Ubuntu 16.04.

On Debian, Go compiler is quite old, and its standard library misses the syscall.Dup2() function. I will implement workaround, but a bit later. But sane-airscan was build successfully for this target :-) 

On Ubunty 16.04/aarch64, SuSE build system misses many essential packages, and I can't resolve this issue.

Regarding your question, IPP uses HTTP, and at the HTTP level, there are only `http://` and `https://`. The `ipp://` and `ipps://` schemes are interpreted by CUPS, and when it comes to the HTTP, CUPS internally translates them to appropriate HTTP schemes. This is better to ask @tillkamppeter about CUPS, but if it sometimes work, sometimes not, it looks as a CUPS bug.

tangyanli, 2020-06-04:
Hi, alexpevzner 

The Debian10 aarch64 binary package is enough.
(waiting for it :-)). It can also works well on Ubuntu16.04. For I have tested the ipp-usb aarch64 on Ubuntu16.04 which built on Deepin (compatible with Debian10), but I just built the ipp-usb, not the whole package.

And thanks for your reply of the question.

alexpevzner, 2020-06-04:
Hi @tangyanli,

I've created ARM64 package for Debian 10

tangyanli, 2020-06-04:
Got it! Thank you so much

alexpevzner, 2020-06-04:
You are always welcome!

closed 2020-06-04 by alexpevzner