OpenPrinting/ipp-usb #3: Unstable connection

scubajeff, 2020-06-04:
Compile latest ipp-usb from source on a Raspberry Pi 3B+ running Ubuntu 20.04 which connects to a HP LaserJet MFP M28a via usb. Printing is OK, but sometimes hang, and I have to switch off and switch on the printer to get it back to work. Browsing to printer sever at port 60000 hang every now and then, during which I see lot of messages saying "`kernel: usb 1-1.1.2: usbfs: usb_submit_urb returned -121`" in `journalctl` output. With `sane-airscan`, it reports finding a scanner, but fail to scan, the scanning bar in the machine started moving from the left to right, but stop in the middle.

scubajeff, 2020-06-04:
I have latest `sane` 1.0.30 installed from `ppa:rolfbensch/sane-git`, so I try the escl backend, it also found the scanner, but `scanimage -d escl:http://127.0.0.1:60000 > a.pnm` gave this output:
```
Output format is not set, using pnm as a default.
scanimage: rounded value of br-x from 215.9 to 215.9
scanimage: rounded value of br-y from 297.011 to 297.011
scanimage: sane_start: Invalid argument
```
And the printer has to be reboot again.

alexpevzner, 2020-06-04:
Hi @scubajeff,

I've tried to blindly fix it, please test again

scubajeff, 2020-06-05:
My bad. With -o file the scanner is working perfectly. When the console hangup, `/var/log/ipp-usb/` has multiple entries of `USB[1]: zero-size read`. How can I send you the log file?

scubajeff, 2020-06-05:
Did you mean that I don't need any CUPS package installed? Not just `cups`, but also `cups-browsed`, `cups-daemon`, `cups-filters`, `cups-ipp-utils`, `cups-server-common`, etc.?

scubajeff, 2020-06-05:
Like I said, scanning locally with `scanimage` works. But trying to scan using `simple-scan` on another machine failed. The scanner got listed, but failed to scan. ipp-usb log shows that `simple-scan` tried to get `GET /eSCL/ScanJobs/1.0/NextDocument HTTP/1.1` after a scan job created, that failed with 404. While local `scanimage` was doing `GET /eSCL/ScanJobs/yk1lecc6-b1yz-v2zm-1005-9wqtyq05/NextDocument HTTP/1.1`, it also failed with 404, but scan completed.

scubajeff, 2020-06-05:
By the way, I'm using eSCL backend.

scubajeff, 2020-06-05:
Some more observation fyi. If only `ipp-usb` is annoucing print queue, Android and Ubuntu machine on the same network will pick it up nicely, but iOS devices will not. If I create an avahi service file, `avahi` will announce it, it will show up in iOS printer list, but then Android and Ubuntu will list two print queues.

It would be nice to have `ipp-usb` to announce Airprint print queue to iOS devices.

alexpevzner, 2020-06-05:
You may send me log files directly by e-main: pzz@apevzner.com

Please, try scanning with my eSCL backend:
* https://github.com/alexpevzner/sane-airscan -- project page
* https://download.opensuse.org/repositories/home:/pzz/ -- binary packages

May I ask you to send me Avahi service file, that works with iOS?

scubajeff, 2020-06-05:
I tried your backend this morning. `airscan-discover` can find the scanner, but `scanimage` with `airscan` backend can't find any.

I will send you the log file and avahi service file when I get back home later.

scubajeff, 2020-06-06:
I upgrade the Ubuntu client machine to sane 1.0.30, `simple-scan` works now. I had setup `saned` in the print server, and my scan client in Android phone also works. So the issues unsolved are hang in web console and proper way to share print queue and scan device. I've sent you email with log file and avahi service file. During this logging, I experienced hang when I try to open web console home tab.

alexpevzner, 2020-06-06:
Hi @scubajeff,

I've created 3 new issues, based on your feedback: #4, #5 and #6 

#4 and #6 are now hopefully fixed (please, test and confirm)

#5 is work in progress

alexpevzner, 2020-06-08:
Please look to my comment in the #5

commit 10fed649 (2020-06-04): Bulk recv size limited to 16kb per request, as attempt to fix #3
closed 2020-06-18 by alexpevzner