OpenPrinting/ipp-usb #7: Option to allow customizing share print queue name

scubajeff, 2020-06-08:
An option in `ipp-usb.conf` to allow user specify a more meaningful print queue name rather than the default.

alexpevzner, 2020-06-08:
Hi,

`ipp-usb` actually doesn't create a print queue. It takes printer's DND-SD name (usually it is configurable via printer's web console), adds `" (USB)"` suffix to it, and uses the resulting string for announcing the printer.

How CUPS converts the resulting DNS-SD name into the print queue name is out of control of the `ipp-usb`

scubajeff, 2020-06-09:
Actually, as you advised earlier, I have stop and remove CUPS from my server. I'm the guy who don't like a lot of daemon running in the background. My print queue is now advertised by `ipp-usb` I believe, since I have to set `interface = all` to make it works. The name is exactly the return of the txt record of `ty` plus `usb`. It's too long, client side software, no matter ubuntu or android, can't show them all in one line and start using `…`. As a admin, this is not a big deal, but end-user kind of think this is weird.

alexpevzner, 2020-06-09:
End user may change `ty` using printer's web console. Usually this parameter is called like "Bonjour name" or similar. I believe, for end user it is much simpler to use printer's web console rather that tweaking `ipp-usb.conf`

scubajeff, 2020-06-09:
Android and iOS user can't change the print queue name pn their phone. Besides, I tried to change the print queue name in web console's system tab. It accepted and the new name seemed to be saved. But nothing changed on the client side.

alexpevzner, 2020-06-10:
Android and iOS user can't change ipp-usb.conf file too :-)

Changing printer's DNS-SD (Bonjour) name should take effect only after ipp-usb restart or printer disconnect/connect. ipp-usb has no notification that this parameter has been changed on printer.

My main concern regarding this option, is how to implement and explain it so it will be convenient and understandable to the "normal" user.

scubajeff, 2020-06-10:
You are right. Considering this should also work in situation that there are more than one printer, setting names for each in a conf file is itself a headache. But there must be a way, or we will have users scratching their heads on what the hell is M28a_0ABF83_USB means.

scubajeff, 2020-06-11:
Change it in Web console would be nice.

alexpevzner, 2020-08-05:
Closed - have no idea how to implement

closed 2020-08-05 by alexpevzner