OpenPrinting/ipp-usb #10: systemd service: should start verbose in foreground

OdyX, 2020-07-09:
When run in systemd service, `ipp-usb udev` forks in background, systemd is configured to try catching up with it by gessing it's main PID, and the type is set to forking:
https://github.com/OpenPrinting/ipp-usb/blob/af7189461a8f742a830e41e4b29c225ead4b5144/systemd-udev/ipp-usb.service#L7-L9

This seems overly complicated while a simple oneshot service would achieve pretty much the same effect, with additional journal support:
```
 Type=oneshot
 ExecStart=/sbin/ipp-usb udev
```
?

alexpevzner, 2020-07-09:
Well, I'm not udev expert :-)

Is my understanding correct, you suggest to replace the entire content of `[Service]` section with the following:

    [Service]
    Type=oneshot
    ExecStart=/sbin/ipp-usb udev

OdyX, 2020-07-10:
I think the following full-file should work (note I've added the Documentation line):

```
[Unit]
Description=Daemon for IPP over USB printer support
Documentation=man:ipp-usb(8)
After=cups.service avahi-daemon.service
Wants=avahi-daemon.service

[Service]
Type=oneshot
ExecStart=/sbin/ipp-usb udev
```
The only printer I have a USB connection on is an HP LaserJet Professional P1102W, which doesn't seem to do anything with ipp-usb, so you might want to test this.

alexpevzner, 2020-07-10:
I'm currently on my countryside home, while my 25 kg weigh printer is not, so I can test on a real hardware only at the end of next week.

@tangyanli, could you please help us with testing? If yes, please replace the content of the `/lib/systemd/system/ipp-usb.service`file with the content included into the previous message, and verify that `ipp-usb` successfully starts when printer is connected and finishes when printer is disconnected.

This effort is a part of `ipp-usb` integration into the official Debian distro.

Thanks in advance!

tangyanli, 2020-07-10:
@alexpevzner I am glad to do it. I will commit test result here next monday (the prints are not around at the weekend)

alexpevzner, 2020-07-13:
Thank a lot, Tess!

@OdyX, I've updated the file

alexpevzner, 2020-08-05:
Closed - fixed

closed 2020-08-05 by alexpevzner