OpenPrinting/ipp-usb #14: Trying to connect a usb to cups

EduardVidalTulsa, 2020-09-24:
im really not sure about your software work for that.
i have a usb printer (dymo labelwriter 450) connect to my rpi with usb. 
Cups in my notebook x86_64 detect and work with this printer without problems  
but my arm (rpi4) dont. 
So i find your software and i will ask about. 
I can share my usb printer across ipp protocol and conect cups to a ipp://localhost/anything
or this software is not intended for that?

alexpevzner, 2020-09-24:
The answer depends on the capabilities of your printer

Please, run the following command as root, and show me its output:

    lsusb -v

EduardVidalTulsa, 2020-09-28:
sure, my lsusb -v is very extensive, i put entry of my printer at top of my out:
*** this are the out for my rpi and my not detected printer by cups, after that you have equivalent in my notebook what really works with cups ***
Ty for your help 👍 
Bus 001 Device 004: ID 0922:0020 Dymo-CoStar Corp. LabelWriter 450
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0922 Dymo-CoStar Corp.
  idProduct          0x0020 LabelWriter 450
  bcdDevice            1.00
  iManufacturer           1 DYMO
  iProduct                2 DYMO LabelWriter 450
  iSerial                 3 19120604380561
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                4mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         7 Printer
      bInterfaceSubClass      1 Printer
      bInterfaceProtocol      2 Bidirectional
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0101
  Self Powered
** there is my notebook (x86_64 lsusb -v out for my printer (i have 2 of them)
Bus 002 Device 006: ID 0922:0020 Dymo-CoStar Corp. LabelWriter 450
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
[... 713 lines cut ...]
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

alexpevzner, 2020-10-07:
Hi @EduardVidalTulsa,

sorry for long delay with reply. Your device doesn't support IPP over USB, so it will not work with this software. IPP over USB support is indicated by combination of:

    bInterfaceClass       7 Printer
    bInterfaceSubClass    1 Printer
    bInterfaceProtocol    4

closed 2020-10-07 by alexpevzner