OpenPrinting/ipp-usb #16: Printing stuck at "Rendering completed"

hourna, 2020-12-10:
Hi!
I have a Canon E204 printer connected to a raspberry pi. I compiled the latest version, and added the printer to CUPS using generic IPP driver. Connection address is http://localhost:60000/ipp/print. When I send a test page to the printer, it doesn't print anything and status is stuck at "Processing - "Rendering completed""

Maybe I did something wrong. I am not totally sure about how to set up CUPS.

Thanks!

alexpevzner, 2020-12-10:
Hi!

May I ask you to show me logs -- all files from the `/var/log/ipp-usb/` directory?

hourna, 2020-12-10:
I attached everything in that directory.

[04a9-182c-601520-Canon-E200-series.log.2.gz](https://github.com/OpenPrinting/ipp-usb/files/5675215/04a9-182c-601520-Canon-E200-series.log.2.gz)
[04a9-182c-601520-Canon-E200-series.log.3.gz](https://github.com/OpenPrinting/ipp-usb/files/5675216/04a9-182c-601520-Canon-E200-series.log.3.gz)
[04a9-182c-601520-Canon-E200-series.log.4.gz](https://github.com/OpenPrinting/ipp-usb/files/5675217/04a9-182c-601520-Canon-E200-series.log.4.gz)
[main.log](https://github.com/OpenPrinting/ipp-usb/files/5675218/main.log)
[04a9-182c-601520-Canon-E200-series.log](https://github.com/OpenPrinting/ipp-usb/files/5675219/04a9-182c-601520-Canon-E200-series.log)
[04a9-182c-601520-Canon-E200-series.log.0.gz](https://github.com/OpenPrinting/ipp-usb/files/5675220/04a9-182c-601520-Canon-E200-series.log.0.gz)
[04a9-182c-601520-Canon-E200-series.log.1.gz](https://github.com/OpenPrinting/ipp-usb/files/5675221/04a9-182c-601520-Canon-E200-series.log.1.gz)

alexpevzner, 2020-12-10:
I guess you are running `ipp-usb` directly from the build directory.

Please, edit the file `ipp-usb-quirks/default.conf` (in the build directory). Currently it has the following content:
```
[*]
  # Drop Connection: header by default
  http-connection = ""
```
Please, try to replace `""` with `keep-alive` or `close`, and tell me, does it help

If you don't run `ipp-usb` directly from the build directory, but have installed it before, you will find this file here: `/usr/share/ipp-usb/quirks/default.conf`

hourna, 2020-12-10:
I tried both, this time status is stuck at "Connected to printer."

alexpevzner, 2020-12-10:
At previous set of logs, there were too many messages, so some information has gone away.

May I ask you to do the following:

1. Stop `ipp-usb`
2. Set http-connection = keep-alive in the file I've previously mentioned
3. rm /var/log/ipp-usb/*
4. Start `ipp-usb`
5. Reproduce the problem
6. Stop `ipp-usb`
7. Send me all logs again

Note, it may be more convenient for you to zip logs before uploading to github, so you will have to upload only a single file (gitgub doesn't allow to attach .tar files, but allows to attach .zip)

alexpevzner, 2021-01-29:
Hi,

is this issue still relevant?

alexpevzner, 2023-02-25:
Closing, as there is no activity on this issue during very long time

closed 2023-02-25 by alexpevzner