OpenPrinting/ipp-usb #17: No printer found if printer is already on

heiner99, 2020-12-11:
Hi,

Debian 11 testing with printer EPSON ET-4750.
The printer is already on.
Now I switch on my computer.
Cups (localhost:631) does not find the printer.

root@pc4:~# driverless
ERROR: ippfind (PID 2542) stopped with status 1!

main.log:
11-12-2020 12:28:39:   ===============================
11-12-2020 12:28:39:   ipp-usb started in "udev" mode, pid=2444
11-12-2020 12:28:39: + PNP Bus 001 Device 006: added

04b8...log:
11-12-2020 12:28:39:
11-12-2020 12:28:39:   ===============================
11-12-2020 12:28:39: + Bus 001 Device 006: added ET-4750 Series
11-12-2020 12:28:39:   Device info:
11-12-2020 12:28:39:     Ident:         04b8-1129-583443523037373398-EPSON-ET-4750-Series
11-12-2020 12:28:39:     Manufacturer:  EPSON
11-12-2020 12:28:39:     Product:       ET-4750 Series
11-12-2020 12:28:39:     SerialNumber:  583443523037373398
11-12-2020 12:28:39:     MfgAndProduct: EPSON ET-4750 Series
11-12-2020 12:28:39:
11-12-2020 12:28:39:   Device quirks:
11-12-2020 12:28:39:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
11-12-2020 12:28:39:       blacklist = false
11-12-2020 12:28:39:       http-connection = ""
11-12-2020 12:28:39:
11-12-2020 12:28:39:   USB stack parameters
11-12-2020 12:28:39:     authorized_default:          -1
11-12-2020 12:28:39:     autosuspend:                 2
11-12-2020 12:28:39:     blinkenlights:               N
11-12-2020 12:28:39:     initial_descriptor_timeout:  5000
11-12-2020 12:28:39:     nousb:                       N
11-12-2020 12:28:39:     old_scheme_first:            N
11-12-2020 12:28:39:     quirks:
11-12-2020 12:28:39:     usbfs_memory_mb:             16
11-12-2020 12:28:39:     usbfs_snoop:                 N
11-12-2020 12:28:39:     usbfs_snoop_max:             65536
11-12-2020 12:28:39:     use_both_schemes:            Y
11-12-2020 12:28:39:
11-12-2020 12:28:39:   USB interfaces:
11-12-2020 12:28:39:     Config Interface Alt Class SubClass Proto
11-12-2020 12:28:39:        1       0      0   255    255     255
11-12-2020 12:28:39:        1       1      0   7      1       2
11-12-2020 12:28:39: *      1       1      1   7      1       4
11-12-2020 12:28:39:        1       2      0   255    170     1
11-12-2020 12:28:39:        1       3      0   7      1       2
11-12-2020 12:28:39: *      1       3      1   7      1       4
11-12-2020 12:28:39:        1       4      0   255    1       0
11-12-2020 12:28:39: *      1       4      1   7      1       4
11-12-2020 12:28:39:        1       5      0   255    1       0
11-12-2020 12:28:39: *      1       5      1   7      1       4
11-12-2020 12:28:39:
11-12-2020 12:28:39:   USB[0]: open: Bus 001 Device 006 Interface 1 Alt 1
11-12-2020 12:28:39:   USB[1]: open: Bus 001 Device 006 Interface 3 Alt 1
11-12-2020 12:28:39:   USB[2]: open: Bus 001 Device 006 Interface 4 Alt 1
11-12-2020 12:28:39:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
11-12-2020 12:28:39: > IPP request:
11-12-2020 12:28:39: > {
11-12-2020 12:28:39: >     VERSION 2.0
11-12-2020 12:28:39: >     OPERATION Get-Printer-Attributes
11-12-2020 12:28:39: >
11-12-2020 12:28:39: >     GROUP operation-attributes-tag
11-12-2020 12:28:39: >     ATTR "attributes-charset" charset: utf-8
11-12-2020 12:28:39: >     ATTR "attributes-natural-language" naturalLanguage: en-US
11-12-2020 12:28:39: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
[... 622 lines cut ...]
11-12-2020 12:38:14: < HTTP[003]: response body: got 65636 bytes; EOF
11-12-2020 12:38:14:   USB[3]: connection released, 0 in use: --- --- --- ---
11-12-2020 12:38:14: > HTTP[004]: GET http://localhost:60000/LANGUAGES/IPP?LANG=en
11-12-2020 12:38:14: > HTTP[004]: HTTP request header:
11-12-2020 12:38:14: >   GET /LANGUAGES/IPP?LANG=en HTTP/1.1
11-12-2020 12:38:14: >   Host: localhost:60000
11-12-2020 12:38:14: >   User-Agent: CUPS/2.3.3op1 (Linux 5.9.0-4-amd64; x86_64) IPP/2.0
11-12-2020 12:38:14: >
11-12-2020 12:38:14:   USB[0]: connection allocated, 1 in use: a-- --- --- ---
11-12-2020 12:38:14:   HTTP[004]: connection 0 allocated
11-12-2020 12:38:14: > HTTP[004]: request body: got 0 bytes; EOF
11-12-2020 12:38:14: > USB[0]: write: wanted 127 sent 127 total 127
11-12-2020 12:38:14: < USB[0]: read: wanted 4096 got 179 total 179
11-12-2020 12:38:14: < HTTP[004]: GET http://localhost:60000/LANGUAGES/IPP?LANG=en - 200 OK
11-12-2020 12:38:14: < HTTP[004]: HTTP response header:
11-12-2020 12:38:14: <   HTTP/1.1 200 OK
11-12-2020 12:38:14: <   Cache-Control: no-cache
11-12-2020 12:38:14: <   Content-Length: 175
11-12-2020 12:38:14: <   Content-Type: text/strings
11-12-2020 12:38:14: <   Pragma: no-cache
11-12-2020 12:38:14: <   Server: EPSON_Linux UPnP/1.0 Epson UPnP SDK/1.0
11-12-2020 12:38:14: <
11-12-2020 12:38:14: < USB[0]: read: wanted 4096 got 175 total 354
11-12-2020 12:38:14: < HTTP[004]: response body: got 175 bytes; EOF
11-12-2020 12:38:14:   USB[0]: connection released, 0 in use: --- --- --- ---

alexpevzner, 2020-12-11:
Hi,

log from the first (failed) attempt looks truncated. 

Can you please try to reproduce the problem and to resend the log?

heiner99, 2020-12-11:
Log again:
11-12-2020 15:28:58:   ===============================
11-12-2020 15:28:58: + Bus 001 Device 006: added ET-4750 Series
11-12-2020 15:28:58:   Device info:
11-12-2020 15:28:58:     Ident:         04b8-1129-583443523037373398-EPSON-ET-4750-Series
11-12-2020 15:28:58:     Manufacturer:  EPSON
11-12-2020 15:28:58:     Product:       ET-4750 Series
11-12-2020 15:28:58:     SerialNumber:  583443523037373398
11-12-2020 15:28:58:     MfgAndProduct: EPSON ET-4750 Series
11-12-2020 15:28:58:
11-12-2020 15:28:58:   Device quirks:
11-12-2020 15:28:58:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
11-12-2020 15:28:58:       blacklist = false
11-12-2020 15:28:58:       http-connection = ""
11-12-2020 15:28:58:
11-12-2020 15:28:58:   USB stack parameters
11-12-2020 15:28:58:     authorized_default:          -1
11-12-2020 15:28:58:     autosuspend:                 2
11-12-2020 15:28:58:     blinkenlights:               N
11-12-2020 15:28:58:     initial_descriptor_timeout:  5000
11-12-2020 15:28:58:     nousb:                       N
11-12-2020 15:28:58:     old_scheme_first:            N
11-12-2020 15:28:58:     quirks:
11-12-2020 15:28:58:     usbfs_memory_mb:             16
11-12-2020 15:28:58:     usbfs_snoop:                 N
11-12-2020 15:28:58:     usbfs_snoop_max:             65536
11-12-2020 15:28:58:     use_both_schemes:            Y
11-12-2020 15:28:58:
11-12-2020 15:28:58:   USB interfaces:
11-12-2020 15:28:58:     Config Interface Alt Class SubClass Proto
11-12-2020 15:28:58:        1       0      0   255    255     255
11-12-2020 15:28:58:        1       1      0   7      1       2
11-12-2020 15:28:58: *      1       1      1   7      1       4
11-12-2020 15:28:58:        1       2      0   255    170     1
11-12-2020 15:28:58:        1       3      0   7      1       2
11-12-2020 15:28:58: *      1       3      1   7      1       4
11-12-2020 15:28:58:        1       4      0   255    1       0
11-12-2020 15:28:58: *      1       4      1   7      1       4
11-12-2020 15:28:58:        1       5      0   255    1       0
11-12-2020 15:28:58: *      1       5      1   7      1       4
11-12-2020 15:28:58:
11-12-2020 15:28:58:   USB[0]: open: Bus 001 Device 006 Interface 1 Alt 1
11-12-2020 15:28:58:   USB[1]: open: Bus 001 Device 006 Interface 3 Alt 1
11-12-2020 15:28:58:   USB[2]: open: Bus 001 Device 006 Interface 4 Alt 1
11-12-2020 15:28:58:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
11-12-2020 15:28:58: > IPP request:
11-12-2020 15:28:58: > {
11-12-2020 15:28:58: >     VERSION 2.0
11-12-2020 15:28:58: >     OPERATION Get-Printer-Attributes
11-12-2020 15:28:58: >
11-12-2020 15:28:58: >     GROUP operation-attributes-tag
11-12-2020 15:28:58: >     ATTR "attributes-charset" charset: utf-8
11-12-2020 15:28:58: >     ATTR "attributes-natural-language" naturalLanguage: en-US
11-12-2020 15:28:58: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
11-12-2020 15:28:58: >     ATTR "requested-attributes" keyword: all
11-12-2020 15:28:58: > }
11-12-2020 15:28:58:
11-12-2020 15:28:58: > HTTP[000]: POST http://localhost:60000/ipp/print
11-12-2020 15:28:58: > HTTP[000]: request body: got 151 bytes; closed
11-12-2020 15:28:58: > HTTP[000]: body is small (151 bytes), prefetched before sending
11-12-2020 15:28:58: > HTTP[000]: HTTP request header:
11-12-2020 15:28:58: >   POST /ipp/print HTTP/1.1
11-12-2020 15:28:58: >   Host: localhost:60000
11-12-2020 15:28:58: >   User-Agent: ipp-usb
11-12-2020 15:28:58: >   Content-Length: 151
11-12-2020 15:28:58: >   Content-Type: application/ipp
11-12-2020 15:28:58: >
11-12-2020 15:28:58:   USB[0]: connection allocated, 1 in use: a-- --- --- ---
11-12-2020 15:28:58:   HTTP[000]: connection 0 allocated

heiner99, 2020-12-11:
After reboot:
11-12-2020 15:28:58:
11-12-2020 15:28:58:   ===============================
11-12-2020 15:28:58: + Bus 001 Device 006: added ET-4750 Series
11-12-2020 15:28:58:   Device info:
11-12-2020 15:28:58:     Ident:         04b8-1129-583443523037373398-EPSON-ET-4750-Series
11-12-2020 15:28:58:     Manufacturer:  EPSON
11-12-2020 15:28:58:     Product:       ET-4750 Series
11-12-2020 15:28:58:     SerialNumber:  583443523037373398
11-12-2020 15:28:58:     MfgAndProduct: EPSON ET-4750 Series
11-12-2020 15:28:58:
11-12-2020 15:28:58:   Device quirks:
11-12-2020 15:28:58:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
11-12-2020 15:28:58:       blacklist = false
11-12-2020 15:28:58:       http-connection = ""
11-12-2020 15:28:58:
11-12-2020 15:28:58:   USB stack parameters
11-12-2020 15:28:58:     authorized_default:          -1
11-12-2020 15:28:58:     autosuspend:                 2
11-12-2020 15:28:58:     blinkenlights:               N
11-12-2020 15:28:58:     initial_descriptor_timeout:  5000
11-12-2020 15:28:58:     nousb:                       N
11-12-2020 15:28:58:     old_scheme_first:            N
11-12-2020 15:28:58:     quirks:
11-12-2020 15:28:58:     usbfs_memory_mb:             16
11-12-2020 15:28:58:     usbfs_snoop:                 N
11-12-2020 15:28:58:     usbfs_snoop_max:             65536
11-12-2020 15:28:58:     use_both_schemes:            Y
11-12-2020 15:28:58:
11-12-2020 15:28:58:   USB interfaces:
11-12-2020 15:28:58:     Config Interface Alt Class SubClass Proto
11-12-2020 15:28:58:        1       0      0   255    255     255
11-12-2020 15:28:58:        1       1      0   7      1       2
11-12-2020 15:28:58: *      1       1      1   7      1       4
11-12-2020 15:28:58:        1       2      0   255    170     1
11-12-2020 15:28:58:        1       3      0   7      1       2
11-12-2020 15:28:58: *      1       3      1   7      1       4
11-12-2020 15:28:58:        1       4      0   255    1       0
11-12-2020 15:28:58: *      1       4      1   7      1       4
11-12-2020 15:28:58:        1       5      0   255    1       0
11-12-2020 15:28:58: *      1       5      1   7      1       4
11-12-2020 15:28:58:
11-12-2020 15:28:58:   USB[0]: open: Bus 001 Device 006 Interface 1 Alt 1
11-12-2020 15:28:58:   USB[1]: open: Bus 001 Device 006 Interface 3 Alt 1
11-12-2020 15:28:58:   USB[2]: open: Bus 001 Device 006 Interface 4 Alt 1
11-12-2020 15:28:58:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
11-12-2020 15:28:58: > IPP request:
11-12-2020 15:28:58: > {
11-12-2020 15:28:58: >     VERSION 2.0
11-12-2020 15:28:58: >     OPERATION Get-Printer-Attributes
11-12-2020 15:28:58: >
11-12-2020 15:28:58: >     GROUP operation-attributes-tag
11-12-2020 15:28:58: >     ATTR "attributes-charset" charset: utf-8
11-12-2020 15:28:58: >     ATTR "attributes-natural-language" naturalLanguage: en-US
11-12-2020 15:28:58: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
11-12-2020 15:28:58: >     ATTR "requested-attributes" keyword: all
11-12-2020 15:28:58: > }
11-12-2020 15:28:58:
11-12-2020 15:28:58: > HTTP[000]: POST http://localhost:60000/ipp/print
11-12-2020 15:28:58: > HTTP[000]: request body: got 151 bytes; closed
11-12-2020 15:28:58: > HTTP[000]: body is small (151 bytes), prefetched before sending
11-12-2020 15:28:58: > HTTP[000]: HTTP request header:
11-12-2020 15:28:58: >   POST /ipp/print HTTP/1.1
11-12-2020 15:28:58: >   Host: localhost:60000
11-12-2020 15:28:58: >   User-Agent: ipp-usb
11-12-2020 15:28:58: >   Content-Length: 151
11-12-2020 15:28:58: >   Content-Type: application/ipp
11-12-2020 15:28:58: >
11-12-2020 15:28:58:   USB[0]: connection allocated, 1 in use: a-- --- --- ---
11-12-2020 15:28:58:   HTTP[000]: connection 0 allocated
[... 44 lines cut ...]
11-12-2020 15:38:16:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
11-12-2020 15:38:16: > IPP request:
11-12-2020 15:38:16: > {
11-12-2020 15:38:16: >     VERSION 2.0
11-12-2020 15:38:16: >     OPERATION Get-Printer-Attributes
11-12-2020 15:38:16: >
11-12-2020 15:38:16: >     GROUP operation-attributes-tag
11-12-2020 15:38:16: >     ATTR "attributes-charset" charset: utf-8
11-12-2020 15:38:16: >     ATTR "attributes-natural-language" naturalLanguage: en-US
11-12-2020 15:38:16: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
11-12-2020 15:38:16: >     ATTR "requested-attributes" keyword: all
11-12-2020 15:38:16: > }
11-12-2020 15:38:16:
11-12-2020 15:38:16: > HTTP[000]: POST http://localhost:60000/ipp/print
11-12-2020 15:38:16: > HTTP[000]: request body: got 151 bytes; closed
11-12-2020 15:38:16: > HTTP[000]: body is small (151 bytes), prefetched before sending
11-12-2020 15:38:16: > HTTP[000]: HTTP request header:
11-12-2020 15:38:16: >   POST /ipp/print HTTP/1.1
11-12-2020 15:38:16: >   Host: localhost:60000
11-12-2020 15:38:16: >   User-Agent: ipp-usb
11-12-2020 15:38:16: >   Content-Length: 151
11-12-2020 15:38:16: >   Content-Type: application/ipp
11-12-2020 15:38:16: >
11-12-2020 15:38:16:   USB[0]: connection allocated, 1 in use: a-- --- --- ---
11-12-2020 15:38:16:   HTTP[000]: connection 0 allocated

alexpevzner, 2020-12-11:
Do you have proprietary Epson drivers installed?

heiner99, 2020-12-11:
Yes, but I purged them after ipp-usb was installed automatically by the debian.
(apt-get purge printer-driver-escpr)

alexpevzner, 2020-12-15:
Hi!

Looks that somebody (CUPS?) steals and reconfigures the device, which causes device to hangup in the middle of conversation.

I have implemented some countermeasures against situations like this. Could you please rebuild `ipp-usb` and retest?

zdohnal, 2020-12-15:
Hi @alexpevzner,

there is a automatic usb installer within system-config-printer-udev, which tries to install/remove print queue for USB printer.
I'll try to change its udev rule to ignore 7/1/4 interfaces. 

Hi @heiner99,
please check whether you have the package above installed in Debian, that could be a service which meddles with ipp-usb.

heiner99, 2020-12-15:
No change after installation:
apt-get install system-config-printer
apt-get install system-config-printer-udev

heiner99, 2020-12-15:
With the latest version of ipp-usb:
Endless loop

main.log:
/home/user/main.log   [B---]  1 L:[  1+ 0   1/ 12] *(1   / 701b) 0053 0x035                                       [*][X]
15-12-2020 15:54:00:   ===============================
15-12-2020 15:54:00:   ipp-usb started in "debug" mode, pid=1996
15-12-2020 15:54:00: + PNP Bus 001 Device 006: added
15-12-2020 15:54:03: ! PNP Bus 001 Device 006: Device initialization timed out
15-12-2020 15:54:04: + PNP Bus 001 Device 006: retry
15-12-2020 15:54:07: ! PNP Bus 001 Device 006: Device initialization timed out
etc...

04b8...log:
15-12-2020 15:54:00:
15-12-2020 15:54:00:   ===============================
15-12-2020 15:54:00: + Bus 001 Device 006: added ET-4750 Series
15-12-2020 15:54:00:   Device info:
15-12-2020 15:54:00:     Ident:         04b8-1129-583443523037373398-EPSON-ET-4750-Series
15-12-2020 15:54:00:     Manufacturer:  EPSON
15-12-2020 15:54:00:     Product:       ET-4750 Series
15-12-2020 15:54:00:     SerialNumber:  583443523037373398
15-12-2020 15:54:00:     MfgAndProduct: EPSON ET-4750 Series
15-12-2020 15:54:00:
15-12-2020 15:54:00:   Device quirks:
15-12-2020 15:54:00:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
15-12-2020 15:54:00:       blacklist = false
15-12-2020 15:54:00:       http-connection = ""
15-12-2020 15:54:00:
15-12-2020 15:54:00:   USB stack parameters
15-12-2020 15:54:00:     authorized_default:          -1
15-12-2020 15:54:00:     autosuspend:                 2
15-12-2020 15:54:00:     blinkenlights:               N
15-12-2020 15:54:00:     initial_descriptor_timeout:  5000
15-12-2020 15:54:00:     nousb:                       N
15-12-2020 15:54:00:     old_scheme_first:            N
15-12-2020 15:54:00:     quirks:
15-12-2020 15:54:00:     usbfs_memory_mb:             16
15-12-2020 15:54:00:     usbfs_snoop:                 N
15-12-2020 15:54:00:     usbfs_snoop_max:             65536
15-12-2020 15:54:00:     use_both_schemes:            Y
15-12-2020 15:54:00:
15-12-2020 15:54:00:   USB interfaces:
15-12-2020 15:54:00:     Config Interface Alt Class SubClass Proto
15-12-2020 15:54:00:        1       0      0   255    255     255
15-12-2020 15:54:00:        1       1      0   7      1       2
15-12-2020 15:54:00: *      1       1      1   7      1       4
15-12-2020 15:54:00:        1       2      0   255    170     1
15-12-2020 15:54:00:        1       3      0   7      1       2
15-12-2020 15:54:00: *      1       3      1   7      1       4
15-12-2020 15:54:00:        1       4      0   255    1       0
15-12-2020 15:54:00: *      1       4      1   7      1       4
15-12-2020 15:54:00:        1       5      0   255    1       0
15-12-2020 15:54:00: *      1       5      1   7      1       4
15-12-2020 15:54:00:
15-12-2020 15:54:00:   USB[0]: open: Bus 001 Device 006 Interface 1 Alt 1
15-12-2020 15:54:00:   USB[1]: open: Bus 001 Device 006 Interface 3 Alt 1
15-12-2020 15:54:00:   USB[2]: open: Bus 001 Device 006 Interface 4 Alt 1
15-12-2020 15:54:00:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
15-12-2020 15:54:00: > IPP request:
15-12-2020 15:54:00: > {
15-12-2020 15:54:00: >     VERSION 2.0
15-12-2020 15:54:00: >     OPERATION Get-Printer-Attributes
15-12-2020 15:54:00: >
15-12-2020 15:54:00: >     GROUP operation-attributes-tag
15-12-2020 15:54:00: >     ATTR "attributes-charset" charset: utf-8
15-12-2020 15:54:00: >     ATTR "attributes-natural-language" naturalLanguage: en-US
15-12-2020 15:54:00: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
15-12-2020 15:54:00: >     ATTR "requested-attributes" keyword: all
15-12-2020 15:54:00: > }
[... 81 lines cut ...]
15-12-2020 15:54:05: > HTTP[001]: POST http://localhost:60000/ipp/print
15-12-2020 15:54:05: > HTTP[001]: request body: got 151 bytes; closed
15-12-2020 15:54:05: > HTTP[001]: body is small (151 bytes), prefetched before sending
15-12-2020 15:54:05: > HTTP[001]: HTTP request header:
15-12-2020 15:54:05: >   POST /ipp/print HTTP/1.1
15-12-2020 15:54:05: >   Host: localhost:60000
15-12-2020 15:54:05: >   User-Agent: ipp-usb
15-12-2020 15:54:05: >   Content-Length: 151
15-12-2020 15:54:05: >   Content-Type: application/ipp
15-12-2020 15:54:05: >
15-12-2020 15:54:05:   USB[0]: connection allocated, 1 in use: a-- --- --- ---
15-12-2020 15:54:05:   HTTP[001]: connection 0 allocated
15-12-2020 15:54:07: > USB[0]: write: wanted 275 sent 0 total 0
15-12-2020 15:54:07: ! USB[0]: send: libusb_bulk_transfer: Operation timed out
15-12-2020 15:54:07: ! HTTP[001]: libusb_bulk_transfer: Operation timed out
15-12-2020 15:54:07:   USB[0]: connection released, 0 in use: --- --- --- ---
15-12-2020 15:54:07: ! IPP: HTTP: Post "http://localhost:60000/ipp/print": libusb_bulk_transfer: Operation timed out
15-12-2020 15:54:07: - Bus 001 Device 006: resetting ET-4750 Series
15-12-2020 15:54:07:   USB[0]: closed
15-12-2020 15:54:07:   USB[1]: closed
15-12-2020 15:54:07:   USB[2]: closed
15-12-2020 15:54:07:   USB[3]: closed
15-12-2020 15:54:07: - Bus 001 Device 006: removed ET-4750 Series
15-12-2020 15:54:08:
etc...

alexpevzner, 2020-12-15:
I've noticed that this device takes A LOT of time to respond for the first time, so I've increased initialization timeout. Please, rebuild and retest.

heiner99, 2020-12-15:
Like last version except slower:

main.log
15-12-2020 16:30:46:   ===============================
15-12-2020 16:30:46:   ipp-usb started in "debug" mode, pid=2332
15-12-2020 16:30:46: + PNP Bus 001 Device 006: added
15-12-2020 16:31:07: ! PNP Bus 001 Device 006: Device initialization timed out
15-12-2020 16:31:08: + PNP Bus 001 Device 006: retry
15-12-2020 16:31:28: ! PNP Bus 001 Device 006: Device initialization timed out
15-12-2020 16:31:29: + PNP Bus 001 Device 006: retry
15-12-2020 16:31:50: ! PNP Bus 001 Device 006: Device initialization timed out
15-12-2020 16:31:51: + PNP Bus 001 Device 006: retry
etc...

04b8...log:
15-12-2020 16:30:46:
15-12-2020 16:30:46:   ===============================
15-12-2020 16:30:46: + Bus 001 Device 006: added ET-4750 Series
15-12-2020 16:30:46:   Device info:
15-12-2020 16:30:46:     Ident:         04b8-1129-583443523037373398-EPSON-ET-4750-Series
15-12-2020 16:30:46:     Manufacturer:  EPSON
15-12-2020 16:30:46:     Product:       ET-4750 Series
15-12-2020 16:30:46:     SerialNumber:  583443523037373398
15-12-2020 16:30:46:     MfgAndProduct: EPSON ET-4750 Series
15-12-2020 16:30:46:
15-12-2020 16:30:46:   Device quirks:
15-12-2020 16:30:46:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
15-12-2020 16:30:46:       blacklist = false
15-12-2020 16:30:46:       http-connection = ""
15-12-2020 16:30:46:
15-12-2020 16:30:46:   USB stack parameters
15-12-2020 16:30:46:     authorized_default:          -1
15-12-2020 16:30:46:     autosuspend:                 2
15-12-2020 16:30:46:     blinkenlights:               N
15-12-2020 16:30:46:     initial_descriptor_timeout:  5000
15-12-2020 16:30:46:     nousb:                       N
15-12-2020 16:30:46:     old_scheme_first:            N
15-12-2020 16:30:46:     quirks:
15-12-2020 16:30:46:     usbfs_memory_mb:             16
15-12-2020 16:30:46:     usbfs_snoop:                 N
15-12-2020 16:30:46:     usbfs_snoop_max:             65536
15-12-2020 16:30:46:     use_both_schemes:            Y
15-12-2020 16:30:46:
15-12-2020 16:30:46:   USB interfaces:
15-12-2020 16:30:46:     Config Interface Alt Class SubClass Proto
15-12-2020 16:30:46:        1       0      0   255    255     255
15-12-2020 16:30:46:        1       1      0   7      1       2
15-12-2020 16:30:46: *      1       1      1   7      1       4
15-12-2020 16:30:46:        1       2      0   255    170     1
15-12-2020 16:30:46:        1       3      0   7      1       2
15-12-2020 16:30:46: *      1       3      1   7      1       4
15-12-2020 16:30:46:        1       4      0   255    1       0
15-12-2020 16:30:46: *      1       4      1   7      1       4
15-12-2020 16:30:46:        1       5      0   255    1       0
15-12-2020 16:30:46: *      1       5      1   7      1       4
15-12-2020 16:30:46:
15-12-2020 16:30:46:   USB[0]: open: Bus 001 Device 006 Interface 1 Alt 1
15-12-2020 16:30:46:   USB[1]: open: Bus 001 Device 006 Interface 3 Alt 1
15-12-2020 16:30:46:   USB[2]: open: Bus 001 Device 006 Interface 4 Alt 1
15-12-2020 16:30:46:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
15-12-2020 16:30:46: > IPP request:
15-12-2020 16:30:46: > {
15-12-2020 16:30:46: >     VERSION 2.0
15-12-2020 16:30:46: >     OPERATION Get-Printer-Attributes
15-12-2020 16:30:46: >
15-12-2020 16:30:46: >     GROUP operation-attributes-tag
15-12-2020 16:30:46: >     ATTR "attributes-charset" charset: utf-8
15-12-2020 16:30:46: >     ATTR "attributes-natural-language" naturalLanguage: en-US
15-12-2020 16:30:46: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
15-12-2020 16:30:46: >     ATTR "requested-attributes" keyword: all
[... 229 lines cut ...]
15-12-2020 16:31:51:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
15-12-2020 16:31:51: > IPP request:
15-12-2020 16:31:51: > {
15-12-2020 16:31:51: >     VERSION 2.0
15-12-2020 16:31:51: >     OPERATION Get-Printer-Attributes
15-12-2020 16:31:51: >
15-12-2020 16:31:51: >     GROUP operation-attributes-tag
15-12-2020 16:31:51: >     ATTR "attributes-charset" charset: utf-8
15-12-2020 16:31:51: >     ATTR "attributes-natural-language" naturalLanguage: en-US
15-12-2020 16:31:51: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
15-12-2020 16:31:51: >     ATTR "requested-attributes" keyword: all
15-12-2020 16:31:51: > }
15-12-2020 16:31:51:
15-12-2020 16:31:51: > HTTP[003]: POST http://localhost:60000/ipp/print
15-12-2020 16:31:51: > HTTP[003]: request body: got 151 bytes; closed
15-12-2020 16:31:51: > HTTP[003]: body is small (151 bytes), prefetched before sending
15-12-2020 16:31:51: > HTTP[003]: HTTP request header:
15-12-2020 16:31:51: >   POST /ipp/print HTTP/1.1
15-12-2020 16:31:51: >   Host: localhost:60000
15-12-2020 16:31:51: >   User-Agent: ipp-usb
15-12-2020 16:31:51: >   Content-Length: 151
15-12-2020 16:31:51: >   Content-Type: application/ipp
15-12-2020 16:31:51: >
15-12-2020 16:31:51:   USB[0]: connection allocated, 1 in use: a-- --- --- ---
15-12-2020 16:31:51:   HTTP[003]: connection 0 allocated

alexpevzner, 2020-12-15:
Hi @zdohnal,

this is not enough to simple ignore 7/1/4 interfaces, because the same interface may implement, for example. 7/1/4 and 7/1/2 protocols, depending on its alt setting, as with this printer, and this is very common for many devices.

The situation becomes even more complex, because there are devices, that advertises 7/1/4 interfaces, but IPP-over-USB actually doesn't work (for now I know only one of them, `HP Laser MFP 135a` see https://github.com/alexpevzner/sane-airscan/issues/68 for details)

`ipp-usb` blacklists such a devices, in a hope they will be handled by somebody else (and in order not to interfere with this "somebody else"), but if this list of exceptions will grow, it would be nice to find a way to share it between `ipp-usb` and `udev`

alexpevzner, 2020-12-15:
@heiner99,

if you restart failed `ipp-usb`, does work after that?

heiner99, 2020-12-15:
Same behaviour.
I am going to install now a new fresh Debian 11 Testing on my PC.
Maybe my current installtion is broken.
I will report tomorrow.

alexpevzner, 2020-12-15:
Installation is actually very simple, there is nothing to break.

Is my understanding correct, that latest "official" release doesn't work if system boots when printer is already on, while latest patched version doesn't work at all?

heiner99, 2020-12-15:
The latest patch does not work if printer is already on.
If I switched printer off and on,  cups will find the printer:

main.log:
15-12-2020 16:53:28:   ===============================
15-12-2020 16:53:28:   ipp-usb started in "udev" mode, pid=689
15-12-2020 16:53:28: + PNP Bus 001 Device 006: added
15-12-2020 17:15:25: + HOTPLUG: added Bus 002 Device 003
15-12-2020 17:20:25: - HOTPLUG: removed Bus 001 Device 006
15-12-2020 17:20:34: + HOTPLUG: added Bus 001 Device 009
15-12-2020 17:20:34: + PNP Bus 001 Device 009: added
15-12-2020 17:20:49: - PNP Bus 001 Device 006: removed

04b8..log:
15-12-2020 16:53:28:
15-12-2020 16:53:28:   ===============================
15-12-2020 16:53:28: + Bus 001 Device 006: added ET-4750 Series
15-12-2020 16:53:28:   Device info:
15-12-2020 16:53:28:     Ident:         04b8-1129-583443523037373398-EPSON-ET-4750-Series
15-12-2020 16:53:28:     Manufacturer:  EPSON
15-12-2020 16:53:28:     Product:       ET-4750 Series
15-12-2020 16:53:28:     SerialNumber:  583443523037373398
15-12-2020 16:53:28:     MfgAndProduct: EPSON ET-4750 Series
15-12-2020 16:53:28:
15-12-2020 16:53:28:   Device quirks:
15-12-2020 16:53:28:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
15-12-2020 16:53:28:       blacklist = false
15-12-2020 16:53:28:       http-connection = ""
15-12-2020 16:53:28:
15-12-2020 16:53:28:   USB stack parameters
15-12-2020 16:53:28:     authorized_default:          -1
15-12-2020 16:53:28:     autosuspend:                 2
15-12-2020 16:53:28:     blinkenlights:               N
15-12-2020 16:53:28:     initial_descriptor_timeout:  5000
15-12-2020 16:53:28:     nousb:                       N
15-12-2020 16:53:28:     old_scheme_first:            N
15-12-2020 16:53:28:     quirks:
15-12-2020 16:53:28:     usbfs_memory_mb:             16
15-12-2020 16:53:28:     usbfs_snoop:                 N
15-12-2020 16:53:28:     usbfs_snoop_max:             65536
15-12-2020 16:53:28:     use_both_schemes:            Y
15-12-2020 16:53:28:
15-12-2020 16:53:28:   USB interfaces:
15-12-2020 16:53:28:     Config Interface Alt Class SubClass Proto
15-12-2020 16:53:28:        1       0      0   255    255     255
15-12-2020 16:53:28:        1       1      0   7      1       2
15-12-2020 16:53:28: *      1       1      1   7      1       4
15-12-2020 16:53:28:        1       2      0   255    170     1
15-12-2020 16:53:28:        1       3      0   7      1       2
15-12-2020 16:53:28: *      1       3      1   7      1       4
15-12-2020 16:53:28:        1       4      0   255    1       0
15-12-2020 16:53:28: *      1       4      1   7      1       4
15-12-2020 16:53:28:        1       5      0   255    1       0
15-12-2020 16:53:28: *      1       5      1   7      1       4
15-12-2020 16:53:28:
15-12-2020 16:53:28:   USB[0]: open: Bus 001 Device 006 Interface 1 Alt 1
15-12-2020 16:53:28:   USB[1]: open: Bus 001 Device 006 Interface 3 Alt 1
15-12-2020 16:53:28:   USB[2]: open: Bus 001 Device 006 Interface 4 Alt 1
15-12-2020 16:53:29:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
15-12-2020 16:53:29: > IPP request:
15-12-2020 16:53:29: > {
15-12-2020 16:53:29: >     VERSION 2.0
15-12-2020 16:53:29: >     OPERATION Get-Printer-Attributes
15-12-2020 16:53:29: >
15-12-2020 16:53:29: >     GROUP operation-attributes-tag
15-12-2020 16:53:29: >     ATTR "attributes-charset" charset: utf-8
15-12-2020 16:53:29: >     ATTR "attributes-natural-language" naturalLanguage: en-US
15-12-2020 16:53:29: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
15-12-2020 16:53:29: >     ATTR "requested-attributes" keyword: all
15-12-2020 16:53:29: > }
[... 2032 lines cut ...]
15-12-2020 17:21:39: >   POST /ipp/print HTTP/1.1
15-12-2020 17:21:39: >   Host: localhost:60001
15-12-2020 17:21:39: >   User-Agent: CUPS/2.3.3op1 (Linux 5.9.0-4-amd64; x86_64) IPP/2.0
15-12-2020 17:21:39: >   Content-Length: 337
15-12-2020 17:21:39: >   Content-Type: application/ipp
15-12-2020 17:21:39: >   Date: Tue, 15 Dec 2020 16:21:39 GMT
15-12-2020 17:21:39: >
15-12-2020 17:21:39:   USB[3]: connection allocated, 1 in use: --- --- --- a--
15-12-2020 17:21:39:   HTTP[054]: connection 3 allocated
15-12-2020 17:21:39: > USB[3]: write: wanted 542 sent 542 total 542
15-12-2020 17:21:39: < USB[3]: read: wanted 4096 got 194 total 194
15-12-2020 17:21:39: < HTTP[054]: POST http://localhost:60001/ipp/print - 200 OK
15-12-2020 17:21:39: < HTTP[054]: HTTP response header:
15-12-2020 17:21:39: <   HTTP/1.1 200 OK
15-12-2020 17:21:39: <   Cache-Control: no-cache
15-12-2020 17:21:39: <   Connection: Keep-Alive
15-12-2020 17:21:39: <   Content-Length: 282
15-12-2020 17:21:39: <   Content-Type: application/ipp
15-12-2020 17:21:39: <   Keep-Alive: timeout=30
15-12-2020 17:21:39: <   Pragma: no-cache
15-12-2020 17:21:39: <   Server: Epson_IPP-Server/2.0.0
15-12-2020 17:21:39: <
15-12-2020 17:21:39: < USB[3]: read: wanted 4096 got 282 total 476
15-12-2020 17:21:39: < HTTP[054]: response body: got 282 bytes; EOF
15-12-2020 17:21:39:   USB[3]: connection released, 0 in use: --- --- --- ---

alexpevzner, 2020-12-15:
@heiner99,

please, try to rename `/usr/lib/udev/rules.d/71-ipp-usb.rules` into `/usr/lib/udev/rules.d/69-ipp-usb.rules`

It will cause `ipp-usb` to start before `/usr/lib/udev/rules.d/70-printers.rules` will be executed, which might help.

heiner99, 2020-12-16:
I don't have .../70-printer.rules on my PC,
maybe  because I purged system-printer-config

heiner99, 2020-12-16:
Yesterday I took another PC for testing and there everything was fine.
The difference was ipp-usb 0.9.13.

So I downloaded 0.9.13 from github and compiled it.
I got one compile error:
./usbio_libusb.go:64:43: cannot use int32(err) (type int32) as type _Ctype_int in argument to _Cfunc_libusb_strerror
So I replace line 64 with: return "dummy-error" // C.GoString(C.libusb_strerror(int32(err)))
Now I compiled it without errors.

Cups was able to find my printer and printing was ok:
I hope this will help you.

main.log:
16-12-2020 07:10:44:   ===============================
16-12-2020 07:10:44:   ipp-usb started in "debug" mode, pid=2263
16-12-2020 07:10:44: + PNP Bus 001 Device 006: added

048b...log:
16-12-2020 07:10:44:
16-12-2020 07:10:44:   ===============================
16-12-2020 07:10:44: + Bus 001 Device 006: added ET-4750 Series
16-12-2020 07:10:44:   Device info:
16-12-2020 07:10:44:     Ident:         04b8-1129-583443523037373398-EPSON-ET-4750-Series
16-12-2020 07:10:44:     Manufacturer:  EPSON
16-12-2020 07:10:44:     Product:       ET-4750 Series
16-12-2020 07:10:44:     MfgAndProduct: EPSON ET-4750 Series
16-12-2020 07:10:44:
16-12-2020 07:10:44:   Device quirks:
16-12-2020 07:10:44:     Connection:
16-12-2020 07:10:44:
16-12-2020 07:10:44:   USB stack parameters
16-12-2020 07:10:44:     authorized_default:          -1
16-12-2020 07:10:44:     autosuspend:                 2
16-12-2020 07:10:44:     blinkenlights:               N
16-12-2020 07:10:44:     initial_descriptor_timeout:  5000
16-12-2020 07:10:44:     nousb:                       N
16-12-2020 07:10:44:     old_scheme_first:            N
16-12-2020 07:10:44:     quirks:
16-12-2020 07:10:44:     usbfs_memory_mb:             16
16-12-2020 07:10:44:     usbfs_snoop:                 N
16-12-2020 07:10:44:     usbfs_snoop_max:             65536
16-12-2020 07:10:44:     use_both_schemes:            Y
16-12-2020 07:10:44:
16-12-2020 07:10:44:   USB interfaces:
16-12-2020 07:10:44:     Config Interface Alt Class SubClass Proto
16-12-2020 07:10:44:        1       0      0   255    255     255
16-12-2020 07:10:44:        1       1      0   7      1       2
16-12-2020 07:10:44: *      1       1      1   7      1       4
16-12-2020 07:10:44:        1       2      0   255    170     1
16-12-2020 07:10:44:        1       3      0   7      1       2
16-12-2020 07:10:44: *      1       3      1   7      1       4
16-12-2020 07:10:44:        1       4      0   255    1       0
16-12-2020 07:10:44: *      1       4      1   7      1       4
16-12-2020 07:10:44:        1       5      0   255    1       0
16-12-2020 07:10:44: *      1       5      1   7      1       4
16-12-2020 07:10:44:
16-12-2020 07:10:44:   USB[0]: open: Bus 001 Device 006 Interface 1 Alt 1
16-12-2020 07:10:44:   USB[1]: open: Bus 001 Device 006 Interface 3 Alt 1
16-12-2020 07:10:44:   USB[2]: open: Bus 001 Device 006 Interface 4 Alt 1
16-12-2020 07:10:44:   USB[3]: open: Bus 001 Device 006 Interface 5 Alt 1
16-12-2020 07:10:44: > IPP request:
16-12-2020 07:10:44: > {
16-12-2020 07:10:44: >     VERSION 2.0
16-12-2020 07:10:44: >     OPERATION Get-Printer-Attributes
16-12-2020 07:10:44: >
16-12-2020 07:10:44: >     GROUP operation-attributes-tag
16-12-2020 07:10:44: >     ATTR "attributes-charset" charset: utf-8
16-12-2020 07:10:44: >     ATTR "attributes-natural-language" naturalLanguage: en-US
16-12-2020 07:10:44: >     ATTR "printer-uri" uri: http://localhost:60000/ipp/print
16-12-2020 07:10:44: >     ATTR "requested-attributes" keyword: all
[... 26 lines cut ...]
16-12-2020 07:10:45: <
16-12-2020 07:10:45: < USB[0]: read: wanted 4096 got 4096 total 4291
16-12-2020 07:10:45: < USB[0]: read: wanted 4096 got 4096 total 8387
16-12-2020 07:10:45: < USB[0]: read: wanted 4096 got 450 total 8837
16-12-2020 07:10:45: < HTTP[000]: response body: got 8642 bytes; EOF
16-12-2020 07:10:45: < IPP response:
16-12-2020 07:10:45: < {
16-12-2020 07:10:45: <     VERSION 2.0
16-12-2020 07:10:45: <     STATUS successful-ok
16-12-2020 07:10:45: <
16-12-2020 07:10:45: <     GROUP operation-attributes-tag
16-12-2020 07:10:45: <     ATTR "attributes-charset" charset: utf-8
16-12-2020 07:10:45: <     ATTR "attributes-natural-language" naturalLanguage: en
16-12-2020 07:10:45: <
16-12-2020 07:10:45: <     GROUP printer-attributes-tag
16-12-2020 07:10:45: <     ATTR "copies-default" integer: 1
16-12-2020 07:10:45: <     ATTR "copies-supported" rangeOfInteger: 1-99
16-12-2020 07:10:45: <     ATTR "finishings-default" enum: 3
16-12-2020 07:10:45: <     ATTR "finishings-supported" enum: 3
16-12-2020 07:10:45: <     ATTR "sides-default" keyword: one-sided
16-12-2020 07:10:45: <     ATTR "sides-supported" keyword: one-sided two-sided-short-edge two-sided-long-edge
16-12-2020 07:10:45: <     ATTR "orientation-requested-default" enum: 3
16-12-2020 07:10:45: <     ATTR "orientation-requested-supported" enum: 3
16-12-2020 07:10:45: <     ATTR "media-default" keyword: na_letter_8.5x11in
etc...

heiner99, 2020-12-17:
Only for your information:

The first version which does not run OK on my PC is the version from 3.September:
ipp-usb-14bdcdbfb15ca3bcb0c1a9ea15c3b25d1f6635c5.zip

alexpevzner, 2020-12-17:
This is vary interesting observation, thank you!

May I ask you to comment line 533 in the `usbtransport.go` in the latest version:
```
err = conn.iface.SoftReset()
```
and retest.

heiner99, 2020-12-17:
Latest version from github with comment line 533 works fine on my PC.
Cups finds the printer and printing is ok.
Thanks you, Good work.

Only for your information:
Scanning with sane-airscan works.
Scanning with libsane1: failed to open device: escl:http://127.0.0.1:60000:out of memory

zdohnal, 2020-12-18:
@alexpevzner thank you for the info! Then the moving ipp-usb rule before system-config-printer-udev rule is the most sensible thing to do right now. If ipp-usb doesn't take care of the device, then s-c-p-udev can get a shot... it would be nice to have a aggregated application for USB printers, which will first try IPP-over-USB solution and then printer application if the former fails.

@heiner99 escl comes from sane-backends project, so if you are willing, please report the issue you see there.

alexpevzner, 2020-12-18:
@zdohnal,

@heiner99 made a very good observation, that this problem was introduced in the `ipp-usb` 0.99.14, while 0.99.13 works well with his device. It helped me to quickly identify the root cause of this problem, so expect 0.99.15 to be released soon with the  appropriate fixes.

Out of memory bug in the `sane-escl` is probably fixed (at least, I've already seen it reported), but logistics of `sane-backends` package into distros is very slow, so nobody knows when this fix will reach your distro; `sane-airscan` never had such a problem.

heiner99, 2020-12-19:
To be more precise:

1.
Latest version from github with comment line 533 works fine on my PC.
Cups finds the printer and printing is ok.
Scanning with sane-airscan works.
Scanning with libsane1: failed to open device: escl:http://127.0.0.1:60000:out of memory

2.
Version 0.9.13
Cups finds the printer and printing is ok.
Scanning with sane-airscan works.
Scanning with libsane1 works.

3.
Version 0.9.14 (and printer is switched off and on)
Cups finds the printer and printing is ok.
Scanning with sane-airscan works.
Scanning with libsane1 works.

alexpevzner, 2020-12-23:
Hi,

I've updated the sources to include the fix.

please, retest the latest github version, and if everything is OK, I'll release ot as 0.9.15

heiner99, 2020-12-23:
With latest version from github:

Cups finds the printer and printing is ok.
Scanning with sane-airscan works.
Scanning with libsane1: failed to open device: escl:http://127.0.0.1:60000:out of memory

Thanks you

alexpevzner, 2020-12-25:
Fixed in 0.99.15

Regarding out of memory error with libsabe1, this is `sane-escl` bug. As far as I remember (from reading sane-devel@alioth-lists.debian.net mail list), this bug already known and fixed, but logistics of sane-backends to most distros is very slow, so most likely the fix has not yet reached your distro,

https://www.mail-archive.com/sane-devel@alioth-lists.debian.net/msg02163.html

closed 2020-12-25 by alexpevzner