OpenPrinting/ipp-usb #18: FTBFS on 2020-12 Debian sid - cannot use int32(err) (type int32) as type _Ctype_int

OdyX, 2020-12-27:
A FTBFS bug was reported in Debian: https://bugs.debian.org/978348, with the following error: 

```
src/github.com/OpenPrinting/ipp-usb/usbio_libusb.go:64:43: cannot use int32(err) (type int32) as type _Ctype_int in argument to _Cfunc_libusb_strerror
```

alexpevzner, 2020-12-27:
Hi @OdyX,

I've hopefully fixed this bug, but I can't reproduce it on any system I build for, so please retest and confirm that the fix works for you, and after that I'll release `0.9.16`

heiner99, 2020-12-28:
I also get this error:

user@pc4:~/ipp16$ make
gotags -R . > tags
go build -ldflags "-s -w" -tags nethttpomithttp2
# github.com/OpenPrinting/ipp-usb
./usbio_libusb.go:64:57: cannot use int32(err) (type int32) as type _Ctype_int in argument to _Cfunc_libusb_strerror
make: *** [Makefile:13: all] Error 2
Press any key to continue...

user@pc4:~/ipp16$ go version
go version go1.15.6 linux/amd64
Press any key to continue...

heiner99, 2020-12-28:
With this change, it compiles ok:

`62 // String returns string representation of error code
 63 func (err UsbErrCode) String() string {
 64     return C.GoString(C.libusb_strerror(C.int(err)))
 65 }
`

OdyX, 2020-12-28:
@alexpevzner sadly, your patch https://github.com/OpenPrinting/ipp-usb/commit/f532e75cc58be1a83f09d1e6c44e7926c2f70510 doesn't build for me. But @heiner99's above does.

alexpevzner, 2020-12-29:
Unfortunately, @heiner99's variant doesn't build on 64-bit Fedora :-(

I've made second attempt to fix this problem. Please, test again.

heiner99, 2020-12-29:
Latest version compiles OK.

OdyX, 2020-12-29:
It builds fine indeed! Thanks for your work!

closed 2020-12-29 by OdyX

alexpevzner, 2020-12-29:
Released as 0.9.16. Thank you guys for your help!