OpenPrinting/ipp-usb #22: Trying to create invalid AVAHI records for certain devices

fabled, 2021-01-18:
Unable to publish AVAHI records for HP Color LaserJet Pro MFP M479fdn:
```
> HP Color LaserJet Pro M478f-9f [773DF1]: _printer._tcp TXT record:
> HP Color LaserJet Pro M478f-9f [773DF1]: _ipp._tcp TXT record:
    air=none
    mopria-certified=2.0
    rp=ipp/print
    priority=50
    kind=document
    PaperMax=legal-A4
    URF=CP1,MT1-2-8-9-10-11,PQ3-4-5,RS600,SRGB24,OB10,W8,DEVW8,DEVRGB24,ADOBERGB24,DM3,IS19-1-2,V1.4,FN3
    UUID=fe8dacd9-7af5-5bfd-fa31-844e8b9fcd77
    Color=T
    Duplex=T
    note=
    qtotal=1
    usb_MDL=HP Color LaserJet Pro M478f-9f
    usb_MFG=HP
    usb_CMD=PCL5c,PCLXL,POSTSCRIPT,NATIVEOFFICE,PDF,PJL,Automatic,JPEG,AppleRaster,PWGRaster,PCLM,NATIVEOFFICE,NATIVEOFFICE,NATIVEOFFICE,NATIVEOFFICE,NATIVEOFFICE,802.3,DESKJET,DYN
    ty=HP Color LaserJet Pro M478f-9f
    product=(HP Color LaserJet Pro M478f-9f)
    pdl=application/vnd.hp-PCL,application/vnd.hp-PCLXL,application/postscript,application/msword,application/pdf,image/jpeg,image/urf,image/pwg-raster,application/PCLm,application/vnd.openxmlformats-officedocument.wordprocessingml.document,application/vnd.ms-excel,application/vnd.openxmlformats-officedocument.spreadsheetml.sheet,application/vnd.ms-powerpoint,application/vnd.openxmlformats-officedocument.presentationml.presentation,application/octet-stream
    txtvers=1
    adminurl=http://localhost:60000/#hId-pgAirPrint
    Fax=T
    rfo=ipp/faxout
    Scan=T
> HP Color LaserJet Pro M478f-9f [773DF1]: _uscan._tcp TXT record:
    duplex=T
    is=platen,adf
    cs=binary,color,grayscale
    UUID=fe8dacd9-7af5-5bfd-fa31-844e8b9fcd77
    adminurl=http://HP842AFD773DF1.local.
    representation=images/printer.png
    pdl=application/octet-stream,application/pdf,image/jpeg
    ty=HP Color LaserJet Pro M478f-9f
    rs=eSCL
    vers=2.63
    txtvers=1
> HP Color LaserJet Pro M478f-9f [773DF1]: _http._tcp TXT record:
  DNS-SD: HP Color LaserJet Pro M478f-9f [773DF1] (USB): trying
  DNS-SD: FQDN: "vostro.local"
  DNS-SD: FQDN: "vostro.local"->"localhost"
- Bus 001 Device 011: resetting HP Color LaserJet Pro M478f-9f
  USB[0]: closed
  USB[1]: closed
  USB[2]: closed
- Bus 001 Device 011: removed HP Color LaserJet Pro M478f-9f
! PNP Bus 001 Device 011: AVAHI: Invalid record
```

fabled, 2021-01-18:
The printer's `pdl` line is 457 characters or so. Perhaps triggering the TXT record's 256 character "segment" limit.

fabled, 2021-01-18:
> The printer's `pdl` line is 457 characters or so. Perhaps triggering the TXT record's 256 character "segment" limit.

Can confirm this to be the issue, adding ugly hack to rewriting pdl to something shorter made the request go through.

alexpevzner, 2021-01-18:
Very impressive PDL list!

Yes, I confirm this issue. Very long TXT records may be split into multiple records with the same name. I will do it. RFC 4408 requires clients to concatenate these multiple records without spaces, but I bet there are broken implementations, so I'll split by commas with fallback to split in arbitrary places, if fragments between commas are too large. It will take a bit of time to implement.

closed 2021-01-20 by alexpevzner

alexpevzner, 2021-01-20:
It appeared to be not so simple to fix.

Unlike regular DNS, DNS-SD implies hard limit of 255 characters for each key=value pair in the TXT record (regular DNS allows to put multiple entries with the same key and split the value between them, DNS-SD explicitly disables multiple entries with the same key).

For now, I truncate PDL at the latest comma, that fits the limit, in a hope that firmware is smart enough to put common formats first, exotic formats last. If we'll see a printer that breaks this assumption, I will have to filter PDL list against a dictionary of "known" PDLs, which I really don't want to do: there is a risk that I'll forget to put something essential into this list.

commit 493f12c6 (2021-01-20): Huge PDL lists truncated, to fit DNS-SD limit of 255 bytes (fixes #22)

    Some printers return a really huge list of supported formats
    (PDL is Page Description Language, i.e., image format).
    DNS-SD limits us to fit into 255 bytes, so sometimes
    we need to truncate this list. Truncation needs to be
    implemented carefully, so resulting list remains syntactically
    correct and contains a reasonable subset of list, returned by device.

    One of such printers is HP Color LaserJet Pro MFP M479fdn, its PDL list
    is as long as 457 characters!

    Currently we truncate at the la [... cut]