OpenPrinting/ipp-usb #23: Is ipp-usb active or is HPLIP active for the next print?

Golddouble, 2021-02-28:
My system: MX Linux 19.3 (Debian 10 based, 64 bit); Xfce
My printer: HP printer.
Connection: via USB

My printer works also without HPLIP installed. (-> But bad quality.)
It also works with HPLIP installed. (-> But bad quality.)
It also works with ipp-usb (-> Better quality.)
(I use this ipp-usb-version for Debian 10: https://download.opensuse.org/repositories/home:/pzz/Debian_10/amd64/)

My question:
How can I find out, whether the next print will be made with ipp-usb or HPLIP, without first have to made a test print?

I have made several experiments:

**Some experiments** (all in a VM):

**Initial state:** 
HPLIP is installed and I have a ppd in "/etc/cups/ppd/" for my HP-printer. I also have in Xfce settings ->printer-settings a printer symbol, where I can change the settings of my printer.

**Experiment 1:**
I just leave all as it is an install https://download.opensuse.org/repositories/home:/pzz/Debian_10/amd64/ .
Then I start my VM new. 
Linux has NOT automatically created an additional printer symbol in Xfce-settings -> printer-settings .
I make a print.
The question is: How was this print made? With HPLIP or with ipp-usb? I do not know. But according to the quality of this print, it looks like the print was **made with HPLIP** and not with ipp-usb.

**Experiment 2:**
I first go into -> Xfce-settings -> printer-settings.
There I delete the symbol of my printer, where I can configure the printer settings.
Then I install https://download.opensuse.org/repositories/home:/pzz/Debian_10/amd64/
Then I start my VM new.
Linux has now automatically created a new printer-settings-symbol in Xfce-settings -> printer-settings .
I make a print.
The question is: How was this print made? With HPLIP or with ipp-usb? I do not know. But according to the quality of this print, it looks like the print now was **made with ipp-usb**.

So as you can see: In both examples HPLIP was installed, but ipp-usb was installed in a different way. 

**Questions:**
I would like to know, weather ipp-usb or HPLIP would be active for the next print, without first have to make a test print. 
How can I find this out? Can a run any command in a terminal to check this? 
Can I add a particular line to any config file to ensure that the printer will safely use the HPLIP driver the next time it prints? 
Can I add a particular line to any config file to ensure that the printer will safely use the ipp-usb non-driver the next time it prints?

Has the printer-symbol-file, we can find in -> Xfce settings -> printer-settings any information in it, weather it will be printed with HPLIP or ipp-usb? Why do I think so? As you can see in the experiments above: When these symbols are created new after the installation of ipp-usb (experiment 2), then ipp-usb is active. When they are not created new, then HPLIP seems to remain active.

alexpevzner, 2021-02-28:
What `lpstat -v` prints?

Golddouble, 2021-03-01:
Thank you.

For experiment 1:
```
$ lpstat -v
device for ENVY_5540: hp:/usb/ENVY_5540_series?serial=TH58Q1N1SQ0671
```
For experiment 2:
```
$ lpstat -v
device for ENVY-5540-series: usb://HP/ENVY%205540%20series?serial=TH58Q1N1SQ0671&interface=1
```

It looks like "hp:..." means that it will print with HPLIP and "usb:..." that it will print driver less.

But why do I have to delete all the symbols for printer-settings first and let it create a new one to bring driver less printing into work?

Is there a way I could keep my existing printer-settings and my printer-settings for my PDF-printer and am nevertheless able to print driver less?

Golddouble, 2021-03-01:
Further question:

On https://wiki.debian.org/CUPSDriverlessPrinting#summary I found this: 

Quote:

> ipp-usb is not provided for buster. However, its author distributes a package that functions provided the device is exposed to the local network by editing /etc/ipp-usb/ipp-usb.conf and having ...  "interface = all"

I do not understand for what the line ...
`interface = loopback # all | loopback`
... is good for. I don't notice any difference between 
 "interface = loopback" and "interface = all"

My system: Debian 10 based.
My printer: connected via usb

alexpevzner, 2021-03-01:
`interface = loopback` -- printer is exposed only to the host, the device is connected to

`interface = all` -- printer is exposed to the network.

Golddouble, 2021-03-01:
Ah I see. I have no network. Thank you.

closed 2021-03-02 by Golddouble