OpenPrinting/ipp-usb #24: Cross-compiling ipp-usb for OpenWRT

frleong, 2021-03-04:
Hi, I have interest in this project. I can successfully compile this for x64 Linux, but whenever I try to cross compile for MIPS architecture (or any other architecture other than amd64) to be run for OpenWRT router, I get the following errors:
_github.com/OpenPrinting/ipp-usb
./dnssd.go:128:12: undefined: dnssdSysdep
./dnssd.go:176:21: undefined: newDnssdSysdep
./dnssd.go:281:23: undefined: newDnssdSysdep
./logger.go:142:5: undefined: logIsAtty
./logger.go:143:15: undefined: logColorConsoleWrite
./main.go:158:15: undefined: UsbGetIppOverUsbDeviceDescs
./main.go:200:9: undefined: Daemon
./main.go:233:8: undefined: UsbInit
./usbtransport.go:32:16: undefined: UsbDevHandle
./usbtransport.go:549:13: undefined: UsbInterface
./main.go:233:8: too many errors_

Is it possible to port to other architectures, that are also Linux-based (e.g. OpenWRT)? Thanks!

alexpevzner, 2021-03-04:
Yet, it known to work on MIPS, but was build with native compilation on that platform, not with cross compilation. Unfortunately, I can't provide MIPS binaries, because OpenSUSE build services, that I use for binaries generation, doesn't build for MIPS.

Basically, cross-compilation should also work, assuming you have a working cross-gcc in path. `go build` command line should look like this:
```
CC=arm-linux-gnueabihf-gcc GOOS=linux GOARCH=arm CGO_ENABLED=1 go build
```

Replace `CC=arm-linux-gnueabihf-gcc` and `GOARCH=arm` with settings appropriate for your target.

frleong, 2021-03-05:
Thanks for your prompt reply. However, I do not know how to cross-compile libusb-1.0 and avahi. Do you have some hints for me?

frleong, 2021-03-05:
Thanks for your help! Finally, I succeeded in compiling ipp-usb for openwrt. I needed to use the toolchain from OpenWRT SDK and override CGO CFLAGS and CGO LDFLAGS to point to the OpenWRT Target include and lib directory. After that I could build a binary that could be run in the openwrt router.

frleong, 2021-03-05:
BTW, I've found that compressing the binary using upx reduces the filesize from 5.3MB to 2.1MB, at cost of a slight delay of around 0.5s before starting. Not bad, because routers have very limited space.

alexpevzner, 2021-03-05:
From another hand, compressed ipp-usb fully resides in RAM, while uncompressed loaded from disk page by page, on demand. And most of the pages will not be loaded, regardless of its executable file size, its resident set size (RSS) is relatively small.

frleong, 2021-03-11:
I see, but my OpenWRT router has plenty of RAM (256MB), but not too much flash storage, that's why upx works better in my case.

BTW, I am using HP LaserJet Pro M28a, and exposing the print/scan services via ipp-usb. I can access the web console, and perform scanning with Mopria Android client (eSCL), but I can't seem to find any suitable client for Windows 10. Any suggestions on how to make use of the scanner in Windows 10?

alexpevzner, 2023-02-25:
Looks like issue is fixed

closed 2023-02-25 by alexpevzner

kkraszy, 2023-09-02:
@frleong Could you describe in more detail the process of ipp-usb cross-compiling for openwrt. 
I've been trying to do this for a while but unfortunately with no results....
I understand you had to modify the Makefile, could you post it here?
I have currently managed to compile an openwrt build environment.