OpenPrinting/ipp-usb #25: ipp-usb only for systemd?

Golddouble, 2021-03-07:
Hello,

I was successful to use ipp-usb when I boot Linux with systemd, but not when I boot with sysVinit.
Does ipp-usb not support sysVinit?

Thank you.

alexpevzner, 2021-03-07:
Hi!

It should work with sysVinit, but somebody needs to write appropriate scripts.

`ipp-usb` has two modes of operations:

1) udev mode
```
ipp-usb udev
```
In this mode `ipp-usb` assumes that it will be started every time when compatible device is plugged into the system, and automatically exits when all compatible devices are unplugged.

2) standalone mode
```
ipp-usb standalone
```
In this mode, `ipp-usb` runs forever, and automatically handles PnP events by itself, serving all connected compatible devices.

Both of these modes can be used with `-bg` flag, which tells `ipp-usb` to deamonize itself.

Golddouble, 2021-03-08:
Thank you.

Do I understand right, that until now both the "udev mode" and the "standalone mode" do not work with sysVinit, because appropriate scripts do not exist until now? 
Or is there a combination of "mode" and "flag" that works without "the missing appropriate scripts"? If yes, which combination works without "the missing appropriate scripts"?

Thank you.

Golddouble, 2021-03-29:
Other question:
When I have run the following command:
`sudo ipp-usb standalone -bg`

Then driverless printing is active and it works.

Problem: When I then start my VM I have no access to this printer.
Question: How can I then deactivate my printer in the host? I am looking for something like the opposite of the command "sudo ipp-usb standalone -bg".

I have not found anything for this in 
`man ipp-usb`

There I have only found options to activate it, but not how to inactivate it.

Thank you.

Edit: If there is another way to free up the printer for the VM, then I am also open for that solution.