OpenPrinting/ipp-usb #27: ipp-usb and CUPS?

bitwombat, 2021-04-04:
Can you please clarify how ipp-usb and CUPS play together? CUPS has a USB backend, and speaks IPP, so does a person use ipp-usb  _instead of_ CUPS?

zdohnal, 2021-04-07:
Hi,

I hope Alex (the ipp-usb author) or you doesn't mind if I answer :) . ipp-usb registers a printer which is capable of IPP-over-USB standard and it is connected to your PC via USB as a DNS-SD service on localhost. Then CUPS can automatically find the service on localhost and create a temporary queue for it.

So ipp-usb is meant to provide DNS-SD service of USB printer, which CUPS can easily process. The biggest pros are you don't need a classic driver for your printer, because CUPS is able to generate 'driver' by itself based on IPP communication with device, and that you don't need to install a printer on your machine at all - the printer will appear when you are about to print (in application's print dialog) and it will disappear after successful printing, so it doesn't block your resources.

USB backend in CUPS is for devices which don't support IPP-over-USB standard and you need a classic driver for printer to work, if you use USB backend.

Alex, if I missed something, please correct me.

Thanks!

alexpevzner, 2021-04-07:
Hi guys,

sorry for delay with answer: I'm very busy with another project and feel very tired sometimes.

If `ipp-usb` is installed and printer supports IPP over USB protocol, most likely `ipp-usb` takes control, and CUPS works this way: `CUPS->ipp-usb->printer`. At this case, for CUPS the printer appears as a network printer, supporting IPP protocol and connected to the `localhost`, while `ipp-usb` translates network communication with CUPS into the USB-level communication with the device.

Usually, when `ipp-usb` takes control, it blocks all another USB-level drivers (printing, scanning, proprietary drivers) that may speak with the device in the same time, but there are some exceptions. Actually, it depends on device (how USB functions wired to the USB interfaces), but in most cases `ipp-usb` is not compatible with legacy USB drivers.

It is actually hard to tell, if the particular device supports IPP over USB protocol. For the convenience, `ipp-usb check` command prints a list of detected compatible devices:
```
# ipp-usb check
IPP over USB devices:
 Num  Device              Vndr:Prod  Model
   1. Bus 003 Device 003  0482:069d  "Kyocera ECOSYS M2040dn"

```

This is also possible to look to the `avahi-browse` output:
```
# avahi-browse -rt _ipp._tcp
+     lo IPv4 Kyocera ECOSYS M2040dn (USB)                  _ipp._tcp            local
=     lo IPv4 Kyocera ECOSYS M2040dn (USB)                  _ipp._tcp            local
   hostname = [tosha.local]
   address = [127.0.0.1]
   port = [60000]
   txt = ["air=none" "mopria-certified=1.3" "rp=ipp/print" "priority=50" "kind=document,envelope" "PaperMax=legal-A4" "URF=CP255,DM4,IFU0,IS19-20,OB1-10,PQ4,RS600,V1.4,W8" "UUID=4509a320-00a0-008f-00b6-002507510eca" "Color=F" "Duplex=T" "note=" "qtotal=1" "usb_MDL=ECOSYS M2040dn" "usb_MFG=Kyocera" "usb_CMD=PCLXL,PostScript Emulation,PCL5E,PJL" "ty=ECOSYS M2040dn" "product=(ECOSYS M2040dn)" "pdl=application/octet-stream,application/pdf,image/tiff,image/jpeg,image/urf,application/postscript,application/vnd.hp-PCL,application/vnd.hp-PCLXL,application/vnd.xpsdocument,image/pwg-raster" "txtvers=1" "adminurl=http://localhost:60000/airprint" "Fax=F" "Scan=T"]
```

If you, among other things, see a device at the `127.0.0.1:60000` address, this is `ipp-usb`.

bitwombat, 2021-04-07:
Thanks for all the details @alexpevzner and @zdohnal. Very clear. So many things to learn when CUPS doesn't work :)

closed 2021-04-07 by bitwombat