OpenPrinting/ipp-usb #40: Samsung ProXpress M4020ND

mikhailnov, 2022-02-08:
Hello

I have a printer Samsung ProXpress M4020ND under hands at the moment (and I will probably have access to it again later). It works via IPP via network.

But ipp-usb (from current git master) does not work.

Firstly, udev rule does not trigger ipp-usb-service. It work If change `ENV{ID_USB_INTERFACES}=="*:070104:*"` to `ENV{ID_USB_INTERFACES}=="*:0701??:*"`, then it does trigger it, but it does not find any devices and exits.

```
user@mn-rosa-pc /mnt/dev/sources/wpa_supplicant $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 006: ID 046d:c34b Logitech, Inc. USB Keyboard
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 010: ID 04e8:331d Samsung Electronics Co., Ltd M332x 382x 402x Series
Bus 001 Device 002: ID 1462:b926 Micro Star International MYSTIC LIGHT 
Bus 001 Device 005: ID 8087:0032 Intel Corp. AX210 Bluetooth
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
user@mn-rosa-pc /mnt/dev/sources/wpa_supplicant $ sudo ipp-usb
  ===============================
  ipp-usb started in "debug" mode, pid=26939

...and nothing happens. If I plug and play any USB device it becomes printed here.
```

More logs here:
https://linux-hardware.org/?probe=0600816ba9

strace of running ipp-usb and replugging the printer into USB
[st.log](https://github.com/OpenPrinting/ipp-usb/files/8023100/st.log)

The problem is that it is not clear how to debug it...

zdohnal, 2022-02-22:
Hi Mikhail,
 
I'm not an ipp-usb project owner, but I check the issues, so I might be of some help.

> I have a printer Samsung ProXpress M4020ND under hands at the moment (and I will probably have access to it again later). It works via IPP via network.
> 
> But ipp-usb (from current git master) does not work.
> 

ipp-usb depends on IPP 2.0+ and IPP-over-USB support so the fact the printer works via IPP via network doesn't necessarily mean it works with ipp-usb.

In case of IPP 2.0+ via network the best test is to create IPP Everywhere queue:

```
$ lpadmin -p <name> -v ipp://<IP/hostname>/ipp/print -m everywhere -E
```

If this works, the printer supports IPP Everywhere, which has IPP 2.0+ as an requirement. If it doesn't work, ipp-usb won't work either.

> Firstly, udev rule does not trigger ipp-usb-service. It work If change `ENV{ID_USB_INTERFACES}=="*:070104:*"` to `ENV{ID_USB_INTERFACES}=="*:0701??:*"`, then it does trigger it, but it does not find any devices and exits.
> 

This change proves your printer doesn't support IPP-over-USB, so it cannot be supported by ipp-usb. Devices with IPP-over-USB must support interface protocol 4. So unfortunately it looks like your printer won't work with ipp-usb.

mikhailnov, 2022-02-22:
@zdohnal Thanks a lot!

alexpevzner, 2022-02-22:
Hi @mikhailnov,

looking to the `lsusb` output, I guess your device doesn't support IPP over USB.

IPP over USB is:
```
      bInterfaceClass         7 Printer
      bInterfaceSubClass      1 Printer
      bInterfaceProtocol      4
```
Your device doesn't offer such an interface. This is why `ipp-usb` is not started for your device by `udev` rule, and this is why `ipp-usb`, being started manually or via modified `udev` rule still doesn't see your device.

Even if your device supports IPP over network and printing over USB, it doesn't automatically imply IPP over USB support.

closed 2022-02-22 by alexpevzner

mikhailnov, 2022-02-22:
Thanks! I think it should be documented in README.