OpenPrinting/ipp-usb #42: ipp-usb removes device node /dev/usb/lp0

VK2BEA, 2022-03-08:
It seems that ipp-usb removes the device node for my USB printer (/dev/usb/lp0)

Is there some way to disable this so that it leave the node?

alexpevzner, 2022-03-09:
ipp-usb, when active, temporary detaches kernel driver from the device. So the answer is "no", it is not possible.

VK2BEA, 2022-03-09:
> ipp-usb, when active, temporary detaches kernel driver from the device. So the answer is "no", it is not possible.

Can I blacklist the printer from ipp-usb in /etc/ipp-usb.conf? (I see there is /usr/share/ipp-usb/blacklist.conf but this would be clobbered on an update). (I'm using the Fedora distribution)

alexpevzner, 2022-03-09:
You can uninstall ipp-usb, if you don't need it.

But what is wrong with printing via ipp-usb? Some device features are missed?

VK2BEA, 2022-03-09:
> You can uninstall ipp-usb, if you don't need it.
> 
> But what is wrong with printing via ipp-usb? Some device features are missed?

Yes, I have uninstalled it but that is rather brute-force 8-) 
I'm using the printer from an application in a rather specialized manner (sending a long Postscript program & fonts that remain resident and then sending a small amount of data to print each page). Its not something that a normal user would do. 
I don't see the need for ipp-usb to kill the device node and this behavior must be a recent development (I think usb-ipp has been in Fedora for some time but the device node only disappeared this week).

zdohnal, 2022-03-10:
I've (as a Fedora CUPS maintainer) made ```ipp-usb``` a weak dependency for CUPS in the last update, so normal users can use driverless printing via USB if their devices are capable of working with it. Otherwise users would have needed to proactively find out about ```ipp-usb``` and install it.
As a weak dependency user can uninstall ```ipp-usb``` anytime without harming CUPS itself. However, @VK2BEA please note that if you use CUPS for printing to the USB port, it won't be possible to print to USB device via CUPS without ```ipp-usb``` or a printer application (an application which can either support old devices as an IPP Everywhere device, or can enhance their features beyond options/features of IPP Everywhere) in the future. Since you are sending PS to your printer, I guess the already implemented ```ps-printer-app``` by @tillkamppeter can be used here - it is available as SNAP for now, once cups-filters-2.0 is out it will be packaged as RPM.
Probably you can send the data to the port without CUPS as well - users of network printers can use ```ncat```, so maybe USB printers can be reached by simple ```cat``` to the port? I have never tried this myself. The known downsides of this is the document must be already in format accepted by printer (no filtering happens) and if you want to set any options, they must be made defaults on the printer itself.

tillkamppeter, 2022-03-10:
@alexpevzner could you make ipp-usb read a second blacklist file which is somewhere under `/etc`, so that admins can manually black-list a printer for specialty tasks like @VK2BEA's one?

ipp-usb should first read the `/usr/share/ipp-usb/blacklist.conf` and after that the second blacklist in `/etc`. If there are contradicting entries in the 2 files, the one in `/etc` should be used, so that the admin can do corrections.

You can choose whether you create a new file in `/etc` for that a create a section in `/etc/ipp-usb.conf`.

Please also introduce a way to white-list printers, so that the admin could also correct a printer wrongly included in `/usr/share/ipp-usb/blacklist.conf`.

alexpevzner, 2022-03-10:
Hi @tillkamppeter,

yes, I will.

alexpevzner, 2022-03-14:
Hi @tillkamppeter,

@zdohnal committed this improvement as #44. Tnanks, @zdohnal!