OpenPrinting/ipp-usb #48: ipp-usb is not ready for this device

BELLECalain, 2022-04-08:
don't know what it means  .
the debian bug tracker  system told me to post here .
i'm lost 
it appears on sid 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1008997

alain@sid:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux bookworm/sid
Release:	unstable
Codename:	sid

alain@sid:~$ uname -a
Linux sid 5.16.0-6-amd64 #1 SMP PREEMPT Debian 5.16.18-1 (2022-03-29) x86_64 GNU/Linux

alain@sid:~$ apt policy cups ipp-usb hplip 
cups:
  Installé : 2.4.1op1-2
  Candidat : 2.4.1op1-2
 Table de version :
 *** 2.4.1op1-2 500
        100 http://deb.debian.org/debian testing/main amd64 Packages
        500 http://deb.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status
     2.3.3op2-3+deb11u1 100
        100 http://deb.debian.org/debian stable/main amd64 Packages
ipp-usb:
  Installé : 0.9.20-1
  Candidat : 0.9.20-1
 Table de version :
 *** 0.9.20-1 500
        100 http://deb.debian.org/debian testing/main amd64 Packages
        500 http://deb.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status
     0.9.17-3+b4 100
        100 http://deb.debian.org/debian stable/main amd64 Packages
hplip:
  Installé : (aucun)
  Candidat : 3.22.2+dfsg0-1
 Table de version :
     3.22.2+dfsg0-1 500
        500 http://deb.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status
     3.21.12+dfsg0-1 100
        100 http://deb.debian.org/debian testing/main amd64 Packages
     3.21.2+dfsg1-2 100
        100 http://deb.debian.org/debian stable/main amd64 Packages

alain@sid:~$ lpstat -t
scheduler is running
system default destination: envy
matériel pour envy : ipp://localhost:60000/ipp/print
matériel pour HP-ENVY-5530-series : hp:/usb/ENVY_5530_series?serial=CN595530XD06BX
envy accepte des requêtes depuis mer. 06 avril 2022 11:16:16
HP-ENVY-5530-series accepte des requêtes depuis mar. 05 avril 2022 16:05:38
printer envy is idle.  enabled since mer. 06 avril 2022 11:16:16
printer HP-ENVY-5530-series disabled since mar. 05 avril 2022 16:05:38 -
	File "/usr/lib/cups/filter/hpcups" not available: No such file or directory

alain@sid:~$ scanimage -L
device `hpaio:/usb/ENVY_5530_series?serial=CN595530XD06BX' is a Hewlett-Packard ENVY_5530_series all-in-one

BELLECalain, 2022-04-08:
found a  solution 

on sid  
Linux sid 5.16.0-6-amd64 #1 SMP PREEMPT Debian 5.16.18-1 (2022-03-29) x86_64 GNU/Linux

alain@sid:~$ apt-cache madison ipp-usb
   ipp-usb |   0.9.20-1 | http://deb.debian.org/debian testing/main amd64 Packages
   ipp-usb |   0.9.20-1 | http://deb.debian.org/debian unstable/main amd64 Packages
   ipp-usb | 0.9.17-3+b4 | http://deb.debian.org/debian stable/main amd64 Packages

sudo aptitude install ipp-usb= 0.9.17-3+b4

sudo apt-mark hold ipp-usb .

pour verifier  

apt-mark showhold

sudo dpkg -l ipp-usb 

pour mettre a jour 

sudo apt-mark unhold ipp-usb 

sudo apt  install  --reinstall ipp-usb 

pour verifier 
sudo dpkg -l ipp-usb

alteholz, 2022-04-13:
Hi,

I can confirm this issue with a different printer with version 0.9.20.
@alexpevzner,  is there a reason why you check in usbcommon.go:IsIppOverUsb() for Proto==4? According to [1] this is a reserved value. At least my old Samsung ML-1520 delivers a '2' and thus is not recognized as printer.

  Thorsten

[1] https://www.usb.org/sites/default/files/usbprint11a021811.pdf

BELLECalain, 2022-04-14:
I like your post. 
at least i see that i'm not the only one to have this problem.

in order to help dev's, can you explain me how to debug the ipp-usb package and provide useful information for its rewriting or troubleshooting ?

I don't know anything about debugging or programming.

if you prefer that I leave this to others, tell me.
it's probably very difficult and I think I don't have the level but I'm willing to learn the little I can.

friendly,
respectfully,
alain.

Translated with www.DeepL.com/Translator (free version)

tangyanli, 2022-04-14:
> Hi,
>
> I can confirm this issue with a different printer with version 0.9.20. @alexpevzner, is there a reason why you check in usbcommon.go:IsIppOverUsb() for Proto==4? According to [1] this is a [...]

"Proto==4" is IPP interface, if the "Proto==2", means the device doesn't support IPP over USB specification.
More details you can refer the IPP over USB specification.

BELLECalain, 2022-04-23:
i've just tested ipp-usb with 5.17 kernel .
do not work anymore .
i made some errors ? i forgot something ? 
could you help me  for tests and debugging ?

thanks ,
friendly ,
respectfully .
alain .

tillkamppeter, 2022-04-24:
@BELLECalain do all versions of ipp-usb not work with kernel 5.17? Or is it only the new 0.9.20 version?

Thanks, @tangyanli USB printing (7/1/x) has currently 4 different protocols: x = 1 is mon-directional streaming, x = 2 is bi-directional streaming, x = 3 is IEEE-1284 packaet format (AFAIK only used by HPLIP currently), x = 4 is IPP-over-USB.

IPP-over-USB devices (7/1/4) ALWAYS print and scan driverless. For printing this means at least one of AirPrint, IPP Everywhere, Mopria, or Wi-Fi Direct Print is supported (or at least IPP 2.0 plus understanding at least one page description language of PDF, Apple Raster/URF, PWG Raster, PCLm), for scanning this means to understand at least one of eSCL and WSD/WS Scan.

alexpevzner, 2022-04-24:
Hi @BELLECalain,

may I ask you to collect a whole set of log files (`/var/log/ipp-usb/*`) when running on the 5.17.x kernel? It will be better if before testing you will erase old log files, so there will be less mess between old and new logs.

debiantriage, 2022-04-25:
Hello @alexpevzner @BELLECalain  provided four logs to the Debian bug report at https://lists.debian.org/debian-printing/2022/04/msg00071.html.

I have attached the fourth one.

[ipp-usb.log.txt](https://github.com/OpenPrinting/ipp-usb/files/8553569/ipp-usb.log.txt)

alexpevzner, 2022-04-25:
Hi @debiantriage, nice to hear from you and thanks!

> libusb_bulk_transfer: Operation timed out

Looks like `libusb` is broken on 5.17.x this kernel (or on this distro). It would be nice to retry wth apparmor/selinux disabled (depending on a distro).

BELLECalain, 2022-04-25:
***@***.***:~$ sudo aptitude search ~b

***@***.***:~$ sudo apt-get check
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait

it seems there is no broken packages

alexpevzner, 2022-04-25:
I don't mean the package itself (i.e., files the package consist of) is broken. I mean, the package's functionality is broken.

One of possible reasons is the apparmor/selinux settings. It makes sense to temporary disable apparmor/selinux and to retry. If it will help, it will seriously localise the source of the problem.

What exactly your Linux use, depends on a distro. Fedora and similar love to use selinix. Ubuntu loves apparmor. I don't know about other distros.

debiantriage, 2022-04-25:
@BELLECalain A quick way to do as @alexpevzner  suggests is to purge apparmor and reboot. It is easily reinstalled.

As a separate test, would you try a different USB port?

BELLECalain, 2022-04-25:
uninstalled apparmor and reboot .

system-config-printer  do not work better and do not recognize my printer .

with the 0.9.20-1 ipp-usb package , it seems not to work better than 
with the stable one

(ok for stable version) .

saw this :

https://wiki.debian.org/AppArmor/HowToUse#Enable_AppArmor

and did this before uninstall :

echo 'GRUB_CMDLINE_LINUX_DEFAULT="$GRUB_CMDLINE_LINUX_DEFAULT apparmor=0"' \
   | sudo tee /etc/default/grub.d/apparmor.cfg

now deleted file .cfg

i am reinstalling  apparmor .

do i have to do this  : ?

echo 'GRUB_CMDLINE_LINUX_DEFAULT="$GRUB_CMDLINE_LINUX_DEFAULT apparmor=1 security=apparmor"' \
   | sudo tee /etc/default/grub.d/apparmor.cfg

thanks .

friendly ,

respectfully ;

alain .

BELLECalain, 2022-04-25:
i don't know if using another usb port will be useless .

like using the printer with the stable version of ipp-usb (instead of 
the sid version) is ok .

debiantriage, 2022-04-25:
> libusb_bulk_transfer: Operation timed out

Bearing in mind that my understanding of ipp-usb internals is very tenuous, may I inquire whether the following (from #32) is worth
consisideration?

> Regarding DevInitTimeout, I can add device-specific quirk to selectiverly increase this timeout on devices that require it, but I don't want to increase it for all devices, because if initialization stuck, it prevents ipp-usb from normal shutdown, and long timeout here is really annying.

My thoughts have turned to seeing issue like this one and #33 as having firmware bugs involved, although I am still perplexed by @BELLECalain regaining printing and scanning  by downgrading ipp-usb.

alexpevzner, 2022-04-26:
Hi All,

I've just tested `ipp-usb` 0.9.20 with 5.17.4 kernel at Fedora 35 with Kyocera ECOSYS M2040dn. It works pretty well. So seems the problem is not in the kernel version.

alexpevzner, 2022-04-27:
Hi!

I'm trying to understand the situation with this issue, and I must confess, I'm completely confused. Too many information, sometimes controverting, split into small pieces around entire Internet.

Now I'm trying to bring all parts together.

First questions:
1. Is there any configuration (combination of kernel version and ipp-usb version), that **definitely works**?
2. Is there any configuration, that **definitely doesn't work**?
3. When it works, does it work reliable or time to time?
4. When it doesn't work, doesn't it work always or time to time?

alexpevzner, 2022-04-29:
Ping

alexpevzner, 2022-04-29:
What is unstable kernel and unstable ipp-usb?

debiantriage, 2022-05-01:
@alexpevzner On the present Debian unstable ipp-usb  is 0.9.20 and @BELLECalain was using kernel version 5.16.0-6-amd64 when the bug was reported. He also reports that downgrading to ipp-usb 0.9.17 brings back printing and scanning.

alexpevzner, 2022-05-02:
Hi,

I've reviewed all changes between `0.9.17` and `0.9.20`. The whole list is following:
1. Some changes in documentation. Should not be relevant
2. Snap port. Should not be relevant
3. Loading quirks from the alternative location, `/etc/ipp-usb/quirks/*.conf`, Should not be relevant
4. Some tweaks around DNS-SD advertising. Should not be relevant.
5. Added `usb-max-interfaces` quirk and applied to the `HP OfficeJet Pro 8730` device. Should not be relevant.
6. Fax is not probed by sending `Get-Printer-Attributes` request to the `/ipp/faxout` if device basic capabilities, obtained from the USB side, doesn't include `FAX` bit (see #32). Should not be relevant.
7. While requesting printer attributes with the `Get-Printer-Attributes` request, instead of requesting `"all"`, only really needed attributes is requested, because some enterprise-class HP devices return so huge amount of information, so retrieving it takes too much time (see #32).
8. Fixed USB 3.0 support by using I/O buffer size which is 1024\*N (was 512\*N) while reading from USB. Should not be relevant.

There are literally no other changes between `0.9.17` and `0.9.20`.

None of these changes look suspicious. Only 6) and 7) are visible from the USB side.

If somebody has a problematic device (`HP-ENVY-5530-series`), able to build the `ipp-usb` locally (this is really easy) and willing to help, we can try to debug this issue by temporary disabling effect of these changes directly in the `ipp-usb` code. It it will help, I can add a corresponding quirk

BELLECalain, 2022-05-02:
tried ro compile ipp-usb git version .
i didn't succed .
I have very big problems with the prerequisites.
I don't know what to install and how 
to compile the sources. 

https://github.com/OpenPrinting/ipp-usb

if someone can help me .

respecrfully ,
friendly ,
alain .

BELLECalain, 2022-05-03:
tried the ipp-usb/testing,unstable 0.9.20-1+b1 amd64 

do not seems to work any more .

in case of , could you explain me the procedure to do the tests ?
don't think i made an error but i wonder .

BELLECalain, 2022-05-03:
alain@sid:~/Téléchargements/ipp-usb$ sudo ./ipp-usb
[sudo] Mot de passe de alain : 
  ===============================
  ipp-usb started in "debug" mode, pid=30140
+ HOTPLUG: added Bus 005 Device 007
+ PNP Bus 005 Device 007: added

  ===============================
+ Bus 005 Device 007: added ENVY 5530 series
  Device info:
    Ident:         03f0-c311-CN595530XD06BX-HP-ENVY-5530-series
    Manufacturer:  HP
    Product:       ENVY 5530 series
    SerialNumber:  CN595530XD06BX
    MfgAndProduct: HP ENVY 5530 series
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      http-connection = ""

  USB stack parameters
    authorized_default:          -1
    autosuspend:                 2
    blinkenlights:               N
    initial_descriptor_timeout:  5000
    nousb:                       N
    old_scheme_first:            N
    quirks:
    usbfs_memory_mb:             16
    usbfs_snoop:                 N
    usbfs_snoop_max:             65536
    use_both_schemes:            Y

  USB interfaces:
    Config Interface Alt Class SubClass Proto
       1       0      0   255    204     0
*      1       0      1   7      1       4
       1       1      0   7      1       2
*      1       1      1   7      1       4
       1       2      0   8      6       80
       1       3      0   255    4       1
*      1       3      1   7      1       4

  USB[0]: open: Bus 005 Device 007 Interface 0 Alt 1
  USB[1]: open: Bus 005 Device 007 Interface 1 Alt 1
  USB[2]: open: Bus 005 Device 007 Interface 3 Alt 1
> HTTP[000]: POST http://localhost:60000/ipp/print
> HTTP[000]: request body: got 462 bytes; closed
> HTTP[000]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- --- ---
  HTTP[000]: connection 0 allocated
< HTTP[000]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[000]: response body: got 3283 bytes; EOF
  USB[0]: connection released, 0 in use: --- --- ---
> HTTP[001]: POST http://localhost:60000/ipp/faxout
> HTTP[001]: request body: got 463 bytes; closed
> HTTP[001]: body is small (463 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a-- ---
  HTTP[001]: connection 1 allocated
! USB[1]: zero-size read
! USB[1]: zero-size read
! USB[1]: zero-size read
! USB[1]: zero-size read
! HTTP[001]: Device initialization timed out
  USB[1]: connection released, 0 in use: --- --- ---
! IPP FaxOut probe failed: HTTP: Post "http://localhost:60000/ipp/faxout": Device initialization timed out
[... 169 lines cut ...]
       1       1      0   7      1       2
*      1       1      1   7      1       4
       1       2      0   8      6       80
       1       3      0   255    4       1
*      1       3      1   7      1       4

  USB[0]: open: Bus 005 Device 007 Interface 0 Alt 1
  USB[1]: open: Bus 005 Device 007 Interface 1 Alt 1
  USB[2]: open: Bus 005 Device 007 Interface 3 Alt 1
> HTTP[005]: POST http://localhost:60000/ipp/print
> HTTP[005]: request body: got 462 bytes; closed
> HTTP[005]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- --- ---
  HTTP[005]: connection 0 allocated
! USB[0]: send: libusb_bulk_transfer: Operation timed out
! HTTP[005]: libusb_bulk_transfer: Operation timed out
  USB[0]: connection released, 0 in use: --- --- ---
! IPP: HTTP: Post "http://localhost:60000/ipp/print": libusb_bulk_transfer: Operation timed out
> HTTP[006]: GET http://localhost:60000/eSCL/ScannerCapabilities
  USB[1]: connection allocated, 1 in use: --- a-- ---
  HTTP[006]: connection 1 allocated

printer is not recognized  in system-config-printer  and do not work at all .
i can  not shut it down .
and i have to manually unplug it .

alexpevzner, 2022-05-03:
Looks that device hangs during probing for fax at first attempt of the `ipp-usb` to start and doesn't respond at all during second attempts.

Did you unplug the device between these two attempts?

Does the device has faxing facilities?

Do you still need the building instructions?

BELLECalain, 2022-05-04:
what i did : 

by default, I left the debian stable version of ipp-usb in place and froze it (apt-mark hold)

then, I compiled the git sources.
and finally, I launched the git version of ipp-usb.
(what you see) 
I made an error(s) ?

n.b.: when I copied this I did not touch the printer.
I pressed its on/off button: I couldn't turn it off.

alexpevzner, 2022-05-04:
Hi @BELLECalain,

I have a hypothesis, that problems with your device is caused by attempt to probe faxing facilities. If you have successfully compiled a git version of the `ipp-usb`, probing for fax can be disabled using quirks.

Please, add the following lines to the ` /usr/share/ipp-usb/quirks/HP.conf` file:
```
[HP ENVY 5530 series]
  disable-fax = true
```

If it helps, please, let me know: I will add this quirk to the distribution.

BTW, does your device **actually** have a fax unit?

BELLECalain, 2022-05-04:
i looked at the printer : i didn't saw any fax option .

alain@sid:~$ cat /usr/share/ipp-usb/quirks/HP.conf
# ipp-usb quirks file -- quirks for HP devices

[HP LaserJet MFP M28-M31]
  http-connection = keep-alive

[HP OfficeJet Pro 8730]
  http-connection = close

[HP ENVY 5530 series]
  disable-fax = true

i power on my printer and i test it : 

alain@sid:~/Téléchargements/ipp-usb$ sudo ./ipp-usb 
[sudo] Mot de passe de alain : 
ipp-usb already running

system-config-printer recognize my printer 
and i can print a test page .

alexpevzner, 2022-05-04:
> ipp-usb already running

You probably test with the `ipp-usb`, installed as a package. Can you temporary kill it (`killall ipp-usb`), or even temporary uninstall, and run `ipp-usb` manually from the directory where you have build it?

Or alternatively you can temporary replace `/usr/sbin/ipp-usb` with the `ipp-usb` file that you have build from sources.

BELLECalain, 2022-05-04:
printer powered on , ipp-usb (debian stable version) frozen : 
alain@sid:~/Téléchargements/ipp-usb$ killall ipp-usb
ipp-usb(121764): Opération non permise
ipp-usb: aucun processus trouvé
alain@sid:~/Téléchargements/ipp-usb$ sudo killall ipp-usb

i launch manually the git version : 
alain@sid:~/Téléchargements/ipp-usb$ sudo ./ipp-usb 
  ===============================
  ipp-usb started in "debug" mode, pid=122300
+ PNP Bus 005 Device 006: added

  ===============================
+ Bus 005 Device 006: added ENVY 5530 series
  Device info:
    Ident:         03f0-c311-CN595530XD06BX-HP-ENVY-5530-series
    Manufacturer:  HP
    Product:       ENVY 5530 series
    SerialNumber:  CN595530XD06BX
    MfgAndProduct: HP ENVY 5530 series
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      http-connection = ""

  USB stack parameters
    authorized_default:          -1
    autosuspend:                 2
    blinkenlights:               N
    initial_descriptor_timeout:  5000
    nousb:                       N
    old_scheme_first:            N
    quirks:
    usbfs_memory_mb:             16
    usbfs_snoop:                 N
    usbfs_snoop_max:             65536
    use_both_schemes:            Y

  USB interfaces:
    Config Interface Alt Class SubClass Proto
       1       0      0   255    204     0
*      1       0      1   7      1       4
       1       1      0   7      1       2
*      1       1      1   7      1       4
       1       2      0   8      6       80
       1       3      0   255    4       1
*      1       3      1   7      1       4

  USB[0]: open: Bus 005 Device 006 Interface 0 Alt 1
  USB[1]: open: Bus 005 Device 006 Interface 1 Alt 1
  USB[2]: open: Bus 005 Device 006 Interface 3 Alt 1
> HTTP[000]: POST http://localhost:60000/ipp/print
> HTTP[000]: request body: got 462 bytes; closed
> HTTP[000]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- --- ---
  HTTP[000]: connection 0 allocated
< HTTP[000]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[000]: response body: got 3289 bytes; EOF
  USB[0]: connection released, 0 in use: --- --- ---
> HTTP[001]: POST http://localhost:60000/ipp/faxout
> HTTP[001]: request body: got 463 bytes; closed
> HTTP[001]: body is small (463 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a-- ---
  HTTP[001]: connection 1 allocated
! USB[1]: zero-size read
! USB[1]: zero-size read
[... 116 lines cut ...]
       1       0      0   255    204     0
*      1       0      1   7      1       4
       1       1      0   7      1       2
*      1       1      1   7      1       4
       1       2      0   8      6       80
       1       3      0   255    4       1
*      1       3      1   7      1       4

  USB[0]: open: Bus 005 Device 006 Interface 0 Alt 1
  USB[1]: open: Bus 005 Device 006 Interface 1 Alt 1
  USB[2]: open: Bus 005 Device 006 Interface 3 Alt 1
> HTTP[004]: POST http://localhost:60000/ipp/print
> HTTP[004]: request body: got 462 bytes; closed
> HTTP[004]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- --- ---
  HTTP[004]: connection 0 allocated
! USB[0]: send: libusb_bulk_transfer: Operation timed out
! HTTP[004]: libusb_bulk_transfer: Operation timed out
  USB[0]: connection released, 0 in use: --- --- ---
! IPP: HTTP: Post "http://localhost:60000/ipp/print": libusb_bulk_transfer: Operation timed out
> HTTP[005]: GET http://localhost:60000/eSCL/ScannerCapabilities
  USB[1]: connection allocated, 1 in use: --- a-- ---
  HTTP[005]: connection 1 allocated

system-config-printer do not recognize my printer .

alexpevzner, 2022-05-04:
> Device quirks:
>      disable-fax = false

This is wrong, must be **true**. Probably, I've guessed device name incorrectly.

What `ipp-usb check` prints?

BELLECalain, 2022-05-04:
i noticed that uninstalling ipp-usb also remove the quirks .
removing  ipp-usb and reinstalling quirks .
noticed : removing ipp-usb also remove .../ipp-usb/quirks in /usr/share/
don't know what to do .

alexpevzner, 2022-05-04:
Just run `ipp-usb` from the directory where you've built it, and use quirks located at the `ipp-usb-quirks` subdirectory of the source tree directory -- `ipp-usb` loads quirks out of there as well.

BELLECalain, 2022-05-04:
updated ipp-usb-quirks  in the git repository on my pc .
# ipp-usb quirks file -- quirks for HP devices

[HP LaserJet MFP M28-M31]
  http-connection = keep-alive

[HP OfficeJet Pro 8730]
  http-connection = close

# eSCL requests hangs on this device, if both USB interfaces are
# in use. Limiting number of interfaces to 1 makes scanning
# reliable in a cost of making scan cancellation impossible,
# as there is no second interface to send cancel request.
# (ADF scans still can be canceled between retrieval of
# subsequent pages).
[HP ScanJet Pro 4500 fn1]
  usb-max-interfaces = 1

# HP Photosmart 6520 series doesn't implement true faxing,
# but instead implements internet-based eFax,
# which makes no sense when connected via USB
# so can be safely disabled for this kind of devices.
[HP Photosmart 6520 series]
  disable-fax = true

[HP ENVY 5530 series]
  disable-fax = true

unfreezing ipp-usb and uninstalling it .
alain@sid:~/Téléchargements/ipp-usb/ipp-usb-quirks$ apt policy ipp-usb
ipp-usb:
  Installé : (aucun)
  Candidat : 0.9.20-1+b1
 Table de version :
     0.9.20-1+b1 500
        100 http://deb.debian.org/debian testing/main amd64 Packages
        500 http://deb.debian.org/debian unstable/main amd64 Packages
     0.9.17-3+b4 100
        100 http://deb.debian.org/debian stable/main amd64 Packages
        100 /var/lib/dpkg/status

running  sudo ./ipp-usb 

alain@sid:~/Téléchargements/ipp-usb$ sudo ./ipp-usb 
  ===============================
  ipp-usb started in "debug" mode, pid=32396
+ HOTPLUG: added Bus 005 Device 008
+ PNP Bus 005 Device 008: added

  ===============================
+ Bus 005 Device 008: added ENVY 5530 series
  Device info:
    Ident:         03f0-c311-CN595530XD06BX-HP-ENVY-5530-series
    Manufacturer:  HP
    Product:       ENVY 5530 series
    SerialNumber:  CN595530XD06BX
    MfgAndProduct: HP ENVY 5530 series
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [HP ENVY 5530 series] (/home/alain/Téléchargements/ipp-usb/ipp-usb-quirks/HP.conf:25)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = true
    from [*] (/home/alain/Téléchargements/ipp-usb/ipp-usb-quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      http-connection = ""

[... 621 lines cut ...]
< HTTP[059]: POST http://localhost:60000/ipp/print - 200 OK
! USB[1]: zero-size read
! USB[1]: zero-size read
! USB[1]: zero-size read
< HTTP[059]: response body: got 1495 bytes; EOF
  USB[1]: connection released, 0 in use: --- --- ---

all worked well . 
like you see , i was able to print a test page .
the quirk you told me to add seems to be very usefull .

what to do now ? to my side ? and to your side ? 
trying to make install from the git sources : 

 alain@sid:~/Téléchargements/ipp-usb$ sudo make install
[sudo] Mot de passe de alain : 
gotags -R . > tags
/bin/sh: 1: gotags: not found
make: [Makefile:12 : all] Erreur 127 (ignorée)
go build -ldflags "-s -w" -tags nethttpomithttp2
error obtaining VCS status: exit status 128
	Use -buildvcs=false to disable VCS stamping.
make: *** [Makefile:13 : all] Erreur 1

don't know what to do ?

alexpevzner, 2022-05-04:
I think, the simplest way for you will be:
1. Install latest "official" `ipp-usb`
2. Replace `/usr/sbin/ipp-usb` with the binary you've built
3. Replace `/usr/share/ipp-usb/quirks/HP.conf` with your updated version
4. Unplug the printer and plug it back, so `ipp-usb` will restart

Next release of the `ipp-usb` will include this quirk. I plan to create this release in few days (waiting for #9 in case it will bring one more useful quirk), and then it will take some time to propagate into the distro you use

BELLECalain, 2022-05-04:
updated ipp-usb-quirks  in the git repository on my pc .
# ipp-usb quirks file -- quirks for HP devices

[HP LaserJet MFP M28-M31]
  http-connection = keep-alive

[HP OfficeJet Pro 8730]
  http-connection = close

# eSCL requests hangs on this device, if both USB interfaces are
# in use. Limiting number of interfaces to 1 makes scanning
# reliable in a cost of making scan cancellation impossible,
# as there is no second interface to send cancel request.
# (ADF scans still can be canceled between retrieval of
# subsequent pages).
[HP ScanJet Pro 4500 fn1]
  usb-max-interfaces = 1

# HP Photosmart 6520 series doesn't implement true faxing,
# but instead implements internet-based eFax,
# which makes no sense when connected via USB
# so can be safely disabled for this kind of devices.
[HP Photosmart 6520 series]
  disable-fax = true

[HP ENVY 5530 series]
  disable-fax = true

unfreezing ipp-usb and uninstalling it .
alain@sid:~/Téléchargements/ipp-usb/ipp-usb-quirks$ apt policy ipp-usb
ipp-usb:
  Installé : (aucun)
  Candidat : 0.9.20-1+b1
 Table de version :
     0.9.20-1+b1 500
        100 http://deb.debian.org/debian testing/main amd64 Packages
        500 http://deb.debian.org/debian unstable/main amd64 Packages
     0.9.17-3+b4 100
        100 http://deb.debian.org/debian stable/main amd64 Packages
        100 /var/lib/dpkg/status

running  sudo ./ipp-usb 

alain@sid:~/Téléchargements/ipp-usb$ sudo ./ipp-usb 
  ===============================
  ipp-usb started in "debug" mode, pid=32396
+ HOTPLUG: added Bus 005 Device 008
+ PNP Bus 005 Device 008: added

  ===============================
+ Bus 005 Device 008: added ENVY 5530 series
  Device info:
    Ident:         03f0-c311-CN595530XD06BX-HP-ENVY-5530-series
    Manufacturer:  HP
    Product:       ENVY 5530 series
    SerialNumber:  CN595530XD06BX
    MfgAndProduct: HP ENVY 5530 series
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [HP ENVY 5530 series] (/home/alain/Téléchargements/ipp-usb/ipp-usb-quirks/HP.conf:25)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = true
    from [*] (/home/alain/Téléchargements/ipp-usb/ipp-usb-quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      http-connection = ""

[... 637 lines cut ...]
gotags -R . > tags
/bin/sh: 1: gotags: not found
make: [Makefile:12 : all] Erreur 127 (ignorée)
go build -ldflags "-s -w" -tags nethttpomithttp2
error obtaining VCS status: exit status 128
	Use -buildvcs=false to disable VCS stamping.
make: *** [Makefile:13 : all] Erreur 1

don't know what to do ?

Install latest "official" ipp-usb (done, version 0.9.20-1+b1 installed)
    Replace /usr/sbin/ipp-usb with the binary you've built (done) 
    Replace /usr/share/ipp-usb/quirks/HP.conf with your updated version (done) 
    Unplug the printer and plug it back, so ipp-usb will restart (done and now , testing) : all is ok .

what was the problem ? what was the error ? how did you solve it ?

Next release of the ipp-usb will include this quirk. I plan to create this release in few days (waiting for https://github.com/OpenPrinting/ipp-usb/issues/9 in case it will bring one more useful quirk), and then it will take some time to propagate into the distro you use

thanks a lot .
and again , sorry for the flood .

friendly ,
respectfully ,
alain .

alexpevzner, 2022-05-04:
> what was the problem ? what was the error ? how did you solve it ?

Your printer reports, via USB device capabilities, that it has fax unit, but sometimes hangs when `ipp-usb` attempts to load fax parameters.

I have disabled probing for fax, using the quirk, and the problem dissipated (well, I hope. Actually, we have not tested for enough time to be sure).

BTW, does your printer actually have a fax unit?

Please use this version for few days. If everything still will be OK, I will consider that the problem is finally solved.

BELLECalain, 2022-05-05:
Does my printer have a fax unit?

no, not at all. or else, I don't know where it is.
I have never used such a device. 
I never use it.
I know it exists, but my printer is a "consumer" printer and I don't know if it's made for it . 
I'm not sure. Not sure at all.
 
I don't have a fax unit.
On the other hand, it is well marked: "print / scan / copy / photo / web".
I suspect what this means but I don't know more.
It is a usb / wifi printer.
but the wifi works very badly (it's its weak point), so it's more like a classic usb printer.
 
Anyway, thank you for the support and the troubleshooting.
I hope I haven't been too much trouble.
and sorry for the flood .

Translated with www.DeepL.com/Translator (free version)

tillkamppeter, 2022-05-05:
To determine whether a printer has a fax unit, check the connectors on the device's back. For fax a phone connector is needed, this is an RJ11 socket, it looks similar to an Ethernet socket but is smaller (6 instead of 8 contacts, often only the center 2 of them actually used). Usually there are 2 such sockets so that in countries where the wall socket for the phone only has a single slot, the phone is connected to the second RJ11 of the fax device. Also in the box you find a phone cable, RJ11 on one end and a phone plug specific to your country on the other end.

But it is possible that HP (or perhaps alo other manufacturers) implement a kind of fax-over-internet. Then the device's firmware reports viua DNS-SD and IPP that the device supports fax, and also accepts appropriate jobs. Internally it sends these jobs to a fax service on the internet which passes the fax on into the phone network as needed (like VoIP telephony).

So also check the front panel menus of your device, the web admin interface, and also the DNS-SD TXT record which the device reports (use `avahi-discover` to find out what your device shouts into the network).

BELLECalain, 2022-05-05:
first , i ran avahi-discover alone without power on the printer .
then , i powered on the printer while avahi-discover was running .
and , at last , i shut down the printer with avahi-discover still running .
now , i post you the results .
n.b.: with the last works about ipp-usb . 
i modified nothing since the solution was found .
didn't saw any fax support on my printer .
there is only a main plug and a usb plug . that's all .
and , in the "front panel" , i didn't  saw any fax option .
 
alain@sid:~$ avahi-discover
Browsing domain 'local' on -1.-1 ...
Browsing domain 'lan' on -1.-1 ...
Browsing for services of type '_androidtvremote2._tcp' in domain 'local' on 2.1 ...
Browsing for services of type '_androidtvremote2._tcp' in domain 'local' on 2.0 ...
Browsing for services of type '_spotify-connect._tcp' in domain 'local' on 2.0 ...
Found service 'Bbox Miami' of type '_androidtvremote2._tcp' in domain 'local' on 2.1.
Found service 'Bbox Miami' of type '_androidtvremote2._tcp' in domain 'local' on 2.0.
Found service '77d179db4b4e5ede' of type '_spotify-connect._tcp' in domain 'local' on 2.0.
Browsing for services of type '_googlecast._tcp' in domain 'local' on 2.0 ...
Browsing for services of type '_http._tcp' in domain 'local' on 2.0 ...
Browsing for services of type '_https._tcp' in domain 'local' on 2.0 ...
Browsing for services of type '_ipp._tcp' in domain 'local' on 2.0 ...
Found service 'bbox Ippos Printer' of type '_ipp._tcp' in domain 'local' on 2.0.
Found service 'BouygtelTV-14f2366f3bcb811011524b436e537c2a' of type '_googlecast._tcp' in domain 'local' on 2.0.
Found service 'bbox Setup' of type '_http._tcp' in domain 'local' on 2.0.
Found service 'bbox Secure Setup' of type '_https._tcp' in domain 'local' on 2.0.
Browsing for services of type '_printer._tcp' in domain 'local' on 1.0 ...
Browsing for services of type '_ipp._tcp' in domain 'local' on 1.0 ...
Browsing for services of type '_uscan._tcp' in domain 'local' on 1.0 ...
Browsing for services of type '_http._tcp' in domain 'local' on 1.0 ...
Browsing for services of type '_ipp-usb._tcp' in domain 'local' on 1.0 ...
Found service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_printer._tcp' in domain 'local' on 1.0.
Found service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_ipp._tcp' in domain 'local' on 1.0.
Found service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_uscan._tcp' in domain 'local' on 1.0.
Found service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_http._tcp' in domain 'local' on 1.0.
Found service '0501' of type '_ipp-usb._tcp' in domain 'local' on 1.0.
Service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_printer._tcp' in domain 'local' on 1.0 disappeared.
Service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_ipp._tcp' in domain 'local' on 1.0 disappeared.
Service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_uscan._tcp' in domain 'local' on 1.0 disappeared.
Service 'HP ENVY 5530 series [FE04D6] (USB)' of type '_http._tcp' in domain 'local' on 1.0 disappeared.
Service '0501' of type '_ipp-usb._tcp' in domain 'local' on 1.0 disappeared.

tillkamppeter, 2022-05-05:
Thanks. Then it has really no fax ...

alexpevzner, 2022-05-05:
Hi Till,

it's the second HP device that claims fax support but have problems in attempt to query its parameters. Previous was `HP Photosmart 6520 series`; it implements faxing over Internet, and probably hangs when not connected to the Internet.

BELLECalain, 2022-05-06:
today , scan test : 

alain@sid:~$ scanimage -L
device `escl:http://localhost:60000' is a HP ENVY 5530 series [FE04D6] (USB) platen scanner
device `airscan:e0:HP ENVY 5530 series [FE04D6] (USB)' is a eSCL HP ENVY 5530 series [FE04D6] (USB) ip=127.0.0.1

simple-scan works perfectly .

xsane gives images in black and white while it is set in color.

alexpevzner, 2022-05-06:
> but i wonder : is my printer too old ?
> do i have to change it ?

If your printer does its job for you, why worry?

> or i keep it still a few years ?

Printers of this class rarely survive for more that few years. But I don't see any reasons why not to use it until it appeared to be physically broken

alexpevzner, 2022-05-06:
FIXED

closed 2022-05-06 by alexpevzner

debiantriage, 2022-05-06:
May I thank @alexpevzner for the professional, competent and successful treatment of the issue in this long thread. Kudos to @BELLECalain for hanging in there. Our printing system would be very much the poorer without the support of maintainer and users.

alexpevzner, 2022-05-06:
Thank you very much, Brian, for your warm words!

tillkamppeter, 2022-05-06:
I also want to thank you two, @alexpevzner and @BELLECalain for your patience and effort to hunt down this bug and finally find the solving quirk rule. I have observed the long thread and helped a little bit by fixing the formatting of @BELLECalain's messages (please put a line with only three backticks before and after a block of code lines, to prevent the code from being markdown-formatted). This is the way how reporting, investigating, and solving issues should work.

tillkamppeter, 2022-05-06:
FYI, it is fixed in commit 3bcc48ce0

alexpevzner, 2022-05-06:
Thank you, Till!

I've tagged the current version as `0.9.21` release, so it's time to start propagation it to various distros. Among with fixing `HP-ENVY-5530-series`, it also fixes `HP Photosmart 6520 series`, which has the similar problem.

tillkamppeter, 2022-05-07:
Debian unstable is now updated to ipp-usb 0.9.21 and Ubuntu 22.10 Kinetic Kudu will follow in a few hours via auto-syncing.

alexpevzner, 2022-05-07:
Hi @zdohnal,

please, push to Fedora/RHEL too...

zdohnal, 2022-05-09:
Hi @alexpevzner ,

on my way, I always try to push the newest versions into Fedora. 

RHEL/CentOS Stream is a different story, but ipp-usb hasn't been there just yet - unfortunately it won't be there until there is a solution for coexisting with queues created with classic drivers (either by quirks in ipp-usb, driver packages and sane-backends or change of ipp-usb design to don't block the USB port the whole time ipp-usb is running - P.S. I haven't forgot our email communication :) ) while classic drivers are still supported, or there is a migration path for devices which are capable of using ipp-usb but they are now installed with classic driver...

alexpevzner, 2022-05-09:
Hi @zdohnal,

> on my way, I always try to push the newest versions into Fedora.

I just wanted to notify you that this newest version already exists :-)

> unfortunately it won't be there until there is a solution for coexisting with queues created with classic drivers

Yes, I understand. BTW, does `ipp-usb` exist for RHEL/CentOS at least as optional package, not installed by default?

The more I thinking about this problem, the more I come to conclusion that for the existent users switch from legacy to IPP-USB should be **explicit**, while for the new users, `ipp-usb` should be installed by default. But I have no idea which mechanisms do we have to implement it.

This problem needs to be solved somehow, because otherwise it creates a bad reputation for the driverless printing among Linux users.

Note, there are already a mechanism that prevents CUPS from attempt to use legacy drivers with devices already handled by the `ipp-usb`. See #28 for discussion and details, and https://github.com/OpenPrinting/cups/issues/186 too

zdohnal, 2022-05-09:
> Hi @zdohnal,
> 
> > on my way, I always try to push the newest versions into Fedora.
> 
> I just wanted to notify you that this newest version already exists :-)

No problem, thanks for the heads-up :) - I have set [upstream release monitoring](https://docs.fedoraproject.org/en-US/package-maintainers/Upstream_Release_Monitoring/) for ```ipp-usb``` so I get a bug in Fedora bugzilla once there is a new release, but an additional heads-up is welcome :) . 

I wanted to propose a new thread, but then I've found out you'd already created a one :D - so I'll move there.

zdohnal, 2022-05-09:
@alexpevzner built and submitted updates for F37, F36, F35 and F34.

Dany-Wilde, 2022-10-11:
> Hi , Alexander Pevzner and thanks .
>
>
> don't understood .
>
> could you explain me what to do , in details ?
>
> or i keep the downgraded package if it is  simpliest .
>
> friendly ,
>
> respectfully ,
>
> alain .