OpenPrinting/ipp-usb #51: Blacklist HP Inc. HP Laser 107a

alexpevzner, 2022-05-09:
I create this issue only for records. The actual discussion located here:

&nbsp;&nbsp;&nbsp;&nbsp;https://bugs.launchpad.net/ubuntu/+source/ipp-usb/+bug/1970055

The `HP Inc. HP Laser 107a` announces 7/1/4 interfaces, but actually responds `HTTP 404 Not found` to all the IPP over USB requests.

Note, `HP Inc. HP Laser MFP 135a` has the same problem and already blacklisted:

&nbsp;&nbsp;&nbsp;&nbsp;https:https://github.com/alexpevzner/sane-airscan/issues/68

closed 2022-05-09 by alexpevzner

alexpevzner, 2022-06-20:
Version `0.9.22` should work out of box, doesn't it?

paviladtudk, 2026-10-14:
Data point from macOS (26.7.1): the HP Laser 107a's 7/1/4 interface causes the same problem there. macOS's built-in ippusbd bridge claims it and switches interface 0 to alt 1, after which classic print jobs on alt 0 are silently dropped (printer shows "offline", every other job lost, replugging "fixes" it).

USB descriptor: iface 0 alt 0 = 7/1/2, iface 0 alt 1 = 7/1/4 (shared EP 0x02 OUT / 0x81 IN), iface 1 alt 1 = 7/1/4.

Workaround for Mac users (disable the per-device bridge, force the classic alt setting, print via SpliX): https://github.com/paviladtudk/hp-laser-107a-macos

commit 16742059 (2022-05-09): HP Inc. HP Laser 107a blacklisted (fixes #51)