OpenPrinting/ipp-usb #53: How to simulate a USB printer that gets recognized by ipp-usb?

Xerxekyran, 2022-06-13:
Hi there,

thanks for the great work!

I am using IPP-USB but I also do want to programatically test its functionality. I have a linux system and am testing normal/non-ipp usb printers using the USB printer gadget (https://www.kernel.org/doc/Documentation/usb/gadget_printer.rst). Is there a way of simulating a usb printer that than will be found by IPP-USB?

Regards
Lars

alexpevzner, 2022-06-13:
Hi @Xerxekyran,

most likely, yes. All details are hard to explain on one short message. Basically, the virtual device should have some 7/1/4 interfaces (class = 7, subclass = 1, protocol = 4). Data, sent to these interfaces, should be interpreted as HTTP requests, and data, that could be read from these interfaces, should be HTTP responses to these requests.

It should also return a correct Device Info Descriptor (see IPP-USB specification, 4.3 and function `(devhandle *UsbDevHandle) usbIppBasicCaps()` for details). And also, vendor number, product number, device serial number must be reasonable, similar to the "real" USB devices. Manufacturer and product name strings must also be valid.

The IPP-USB protocol specification can be loaded from here: https://www.usb.org/document-library/ipp-protocol-10

Xerxekyran, 2022-06-14:
Thank you for your answer. I guess there does not yet exists any software which handles this type of "mocking" already?

alexpevzner, 2022-06-14:
> I guess there does not yet exists any software which handles this type of "mocking" already?

No. And I'm not sure about usefulness of such a software. The real problems are mostly about firmware bugs, not about `ipp-usb` bugs.

zdohnal, 2022-06-15:
Just my 2 cents - IMHO any functionality for testing is useful. Currently I have to go to our minilab, where I have IPP-over-USB device, and manually test whether the device is seen and prints via ipp-usb from a print dialog.

Although most real problems are in firmware, but some tests verifying the functionality of ipp-usb and checks whether there are no regressions are always good. I was looking for a mocking USB library myself, but no luck so far.

yetamrra, 2022-07-19:
Drive-by comment here: ChromeOS has a (very basic) virtual MFP that can simulate an IPP-USB printer.  It uses usbip to show up as a device with IPP-USB printer interfaces and implements just enough to submit an IPP print job or an eSCL scan.  Like the comment above mentioned, it's probably not adequate to verify firmware quirks, but it's useful for verifying end-to-end connectivity from your client through ipp-usb (or ippusb_bridge).

You can find the source at https://chromium.googlesource.com/chromiumos/third_party/virtual-usb-printer .  Building the source code depends on various other libraries from ChromeOS, so it might be tricky to build separately.  I don't think it has any runtime dependencies that wouldn't be available on a typical Linux system.