OpenPrinting/ipp-usb #55: ipp-usb: cannot print on HP m28a on ArchLinux

rilian-la-te, 2022-07-10:
from /var/log/ipp-usb/device.log

```
10-07-2022 21:04:57:   USB[0]: connection allocated, 1 in use: a-- --- ---
10-07-2022 21:04:57:   HTTP[074]: connection 0 allocated
10-07-2022 21:04:57: > USB[0]: write: wanted 976 sent 976 total 976
10-07-2022 21:04:57: < USB[0]: read: wanted 4096 got 0 total 0
10-07-2022 21:04:57: ! USB[0]: zero-size read
10-07-2022 21:04:57: < USB[0]: read: wanted 4096 got 1024 total 1024
10-07-2022 21:04:57: < HTTP[074]: POST http://localhost:60000/ipp/print - 200 OK
10-07-2022 21:04:57: < HTTP[074]: HTTP response header:
10-07-2022 21:04:57: <   HTTP/1.1 200 OK
10-07-2022 21:04:57: <   Content-Length: 981
10-07-2022 21:04:57: <   Content-Type: application/ipp
10-07-2022 21:04:57: <
10-07-2022 21:04:57: < USB[0]: read: wanted 4096 got 0 total 1024
10-07-2022 21:04:57: ! USB[0]: zero-size read
10-07-2022 21:04:57: < USB[0]: read: wanted 4096 got 0 total 1024
10-07-2022 21:04:57: ! USB[0]: zero-size read
10-07-2022 21:04:57: < USB[0]: read: wanted 4096 got 0 total 1024
10-07-2022 21:04:57: ! USB[0]: zero-size read
10-07-2022 21:04:57: < USB[0]: read: wanted 4096 got 28 total 1052
10-07-2022 21:04:57: < HTTP[074]: response body: got 981 bytes; EOF
10-07-2022 21:04:57:   USB[0]: connection released, 0 in use: --- --- ---

```

keep-alive quirk is set for M28a.

alexpevzner, 2022-07-11:
Hi,

the log fragment that you've published doesn't contain any problem by itself. zero-size reads are not problems, but sometimes correlates with real problems, though there are a lot of devices that have zero-size reads though works absolutely reliable.

First thing to do, is to isolate the problem to the particular component (it may be on CUPS side, for example).

alexpevzner, 2022-07-11:
Btw, I see you are from Moscow, like me. Where are you located here?

alexpevzner, 2022-07-11:
> @alexpevzner in Butovo

Very near to me. I'm in Yasenevo.

> Scanning works. Web-browser page too. But printing is not - KDE reports than "job is cancelled by printer" for any page, even test one. HPLIP works.

`ipp-usb` is only a transport. Job can be cancelled due to many reasons. If after failed printing job all IPP-USB interfaces are idle:
```
10-07-2022 21:04:57:   USB[0]: connection released, 0 in use: --- --- ---
```
this is very unlikely that the problem is in the `ipp-usb`.

My Kyocera also doesn't work out of box with the open source stack. For me, it helps to extract the `.PPD` file from the proprietary driver bundle and to manually install it on `CUPS`.

@tillkamppeter, can you help us, please? How to solve problems like these?

tillkamppeter, 2022-07-11:
@rilian-la-te here we would need some more information:
- If there is a way to connect the printer by Ethernet or Wi-Fi instead of USB, it should also do driverless printing and scanning. Is it working completely and correctly this way or do you get the same printing problem?
- To investigate the printing problem itself you should put CUPS into debug logging mode (`cupsctl --debug-logging`) and print another (failing) job. After that get the `/var/log/cups/error_log` file and attach it here. Note that GitHub only accepts it if you rename it to have the extension `.txt`.
- Also attach the PPD file of your CUPS queue (in `/etc/cups/ppd/`, also rename it to have a `.txt` extension for attaching it).
- Get further info from your printer with the command `ipptool -tv ipp://localhost:60000/ipp/print get-printer-attributes.txt > attrs.txt` Attach the file `attrs.txt` to this bug report.

Please attach the files one by one, do not compress them and do not package them together. Thanks.

rilian-la-te, 2022-07-11:
@tillkamppeter will do. 

Printer do not support ethernet. 

I will do attaching PPD and other files.

rilian-la-te, 2022-07-11:
@tillkamppeter 

Logs:
[error_log.txt](https://github.com/OpenPrinting/ipp-usb/files/9087589/error_log.txt)
[attrs.txt](https://github.com/OpenPrinting/ipp-usb/files/9087602/attrs.txt)
[printer.ppd.txt](https://github.com/OpenPrinting/ipp-usb/files/9087603/printer.ppd.txt)

tillkamppeter, 2022-07-13:
@rilian-la-te please also do the following:
```
sudo systemctl stop cups
```
Then edit the PPD file of your printer in `/etc/cups/ppd/`. It has the name of your print queue. Remove the line
`*cupsFilter2: "image/pwg-raster image/pwg-raster 10 -"`. Save the file.
The start CUPS again:
```
sudo systemctl start cups
```
Try to print again. Does this work now?

rilian-la-te, 2022-07-21:
@tillkamppeter nope, same symptoms. Removing cups-filte is not working.

rilian-la-te, 2022-07-21:
@debiantriage everywhere mode is also does not work.

rilian-la-te, 2022-07-21:
New log:
[error_log.txt](https://github.com/OpenPrinting/ipp-usb/files/9162265/error_log.txt)

debiantriage, 2022-07-22:
Jobs 35, 36 and 38 are printed on testq. For nsswitch.conf we see

> I [21/Jul/2022:22:38:22 +0300] [Job 35] File of type text/plain queued by "rilian".
D [21/Jul/2022:22:38:22 +0300] [Job 35] Sending job to queue tagged as raw...

Sending as raw should not happen with a correctly set up  everywhere queue. No filtering takes place; nsswitch.conf is sent without any filtering and the M28a cannot possibly print it.

@rilian-la-te 

Thanks for the input but, unfortunately, it does not seem to provide any definite useful information about the operation of ipp-usb. The lpadmin command should produce a PPD in /etv/cups/ppd after querying the printer. Does it?

rilian-la-te, 2022-07-22:
@debiantriage

[testq.ppd.txt](https://github.com/OpenPrinting/ipp-usb/files/9172130/testq.ppd.txt)
 
With second attempt, nsswitch.conf is printed using Everywhere generator.

rilian-la-te, 2022-07-22:
New log:

[error_log.txt](https://github.com/OpenPrinting/ipp-usb/files/9172155/error_log.txt)

Driverless also was able to print nsswitch.conf.. 

But KDE test page is still cannot be printed.

debiantriage, 2022-07-23:
- nsswitch.conf prints on testq and hp-home with everywhere and driverless. I would expect that printing from (say) Firefox would work for either queue.
- The test page does not print on testq and hp-home with either everywhere or driverless. We see _Job canceled at printer_. I do not knw the reason for that. Maybe a cups-filter issue?

rilian-la-te, 2022-07-26:
@debiantriage printing from Firefox works.

As nsswitch.conf too. But I want to print some other files.

rilian-la-te, 2022-07-26:
@debiantriage @tillkamppeter from Okular pdf printing works too, so, only printing non-pdf files is not working.

Is there any way to force jobs as pdf?

rilian-la-te, 2022-07-26:
Printing from firefox using GTK dialog does not work, which is strange.

rilian-la-te, 2022-07-26:
PNG files is also printed using lp -d

rilian-la-te, 2022-07-26:
@tillkamppeter @debiantriage looked to /var/spool/cups and see than all jobs with format 
`document-format-detectedapplication/vnd.cups-pdf-bannerIdocument-formatapplication/vnd.cups-pdf-banner`
is not printed, but with
`ocument-format-detectedapplication/pdfIdocument-formatapplication/pdf` 
is printed. Is I missed some conversions from banner to pdf?

rilian-la-te, 2022-07-26:
Even `lp -d hp-home /usr/share/cups/data/testprint` works from command line.

But printing test page does not work from Qt and Gtk print dialogs. It is same file, I do not know who it should not work
[error_log.txt](https://github.com/OpenPrinting/ipp-usb/files/9193698/error_log.txt)

Look to jobs 61 and 62. It is effectively same file, one from KDE print dialog, and one from lp -d. First succeed, but second failed.

rilian-la-te, 2022-07-26:
Here is /var/spool/cups output

```
[5x5linux-manjaro cups]# cat d00061-001
#PDF-BANNER
Template default-testpage.pdf
Show printer-name printer-info printer-location printer-make-and-model printer-driver-name printer-driver-version paper-size imageable-area job-id options time-at-creation time-at-processing

[5x5linux-manjaro cups]# cat d00062-001
#PDF-BANNER
Template default-testpage.pdf
Show printer-name printer-info printer-location printer-make-and-model printer-driver-name printer-driver-version paper-size imageable-area job-id options time-at-creation time-at-processing
```

```
[5x5linux-manjaro cups]# cat c00061
Gattributes-charsetutf-8Httributes-natural-languageru-ruE
                                                         printer-uri ipp://localhost/printers/hp-homeBjob-originating-user-namerilianjob-name   testprint!copies#
finishings!job-cancel-after*0Djob-hold-untilno-hold!
                                                    job-priority2B
job-sheetsnoneBnone!    number-upDprint-color-mode
monochromejob-uuid-urn:uuid:65751982-1235-3d1d-78ae-29d4af8fa746Bjob-originating-host-name      localhost1date-time-at-completed
                                                                                                                                �▒▒%+1date-time-at-creation
                                                                                                                                                           �▒▒+1date-time-at-processing
                                                                                                                                                                                       �▒▒+!time-at-completedb�j!time-at-creationb�j!time-at-processingb�j!job-id=# job-state       Djob-state-reasons▒processing-to-stop-point!job-impressions-completed!▒job-media-sheets-completedEjob-printer-uri+ipp://5x5linux-manjaro:631/printers/hp-home!
                                                                                                                                                                                                      job-k-octetsI▒document-format-supplied▒application/octet-streamI▒document-format-detectedapplication/vnd.cups-pdf-bannerIdocument-formatapplication/vnd.cups-pdf-bannerBdocument-name-supplied        testprintAjob-printer-state-messageDjob-printer-state-reasonsmedia-empty-warning
```

```
[5x5linux-manjaro cups]# cat c00062

Gattributes-charsetutf-8Httributes-natural-languageru-ruBjob-originating-user-namerilianE
                                                                                         printer-uri+ipp://rilian@localhost:631/printers/hp-homejob-nameПробная страницаI▒document-format-detectedapplication/vnd.cups-pdf-bannerIdocument-formatapplication/vnd.cups-pdf-banner!
                                     job-priority2job-uuid-urn:uuid:b612bc0b-d334-3686-76b2-42f442792db8Bjob-originating-host-name      localhost1date-time-at-completed
                                                                                                                                                                        �▒+1date-time-at-creation
                                                                                                                                                                                                 �▒
                                                                                                                                                                                                   +1date-time-at-processing
                                                                                                                                                                                                                            �▒
                                                                                                                                                                                                                              +!time-at-completedb�j�!time-at-creationb�j�!time-at-processingb�j�!job-id>#  job-stateDjob-state-reasons▒processing-to-stop-point!job-impressions-completed!▒job-media-sheets-completedEjob-printer-uri+ipp://5x5linux-manjaro:631/printers/hp-home!
                                                                                                                                                                                                                                       job-k-octetsDjob-hold-untilno-holdB
job-sheetsnoneBnoneAjob-printer-state-message6Задание отменено на принтере.Djob-printer-state-reasonsmedia-empty-warning
```

rilian-la-te, 2022-07-26:
[error_log.txt](https://github.com/OpenPrinting/ipp-usb/files/9193736/error_log.txt)
Did 63 and 64 with debug logging (also same file, 63 is sucessful, 64 is failed)

rilian-la-te, 2022-07-26:
Tried to print both methods into file devices, and see this:
[jobs.tar.gz](https://github.com/OpenPrinting/ipp-usb/files/9193806/jobs.tar.gz)

Maybe this bug caused by a wrong colorspace?

```
[rilian@5x5linux-manjaro ~]$ file failed-job 
failed-job: Cups Raster version 2, Big Endian, 300x300 dpi, 2480x3508 pixels 1 bits/color 1 bits/pixel ColorOrder=Chunky ColorSpace=black
[rilian@5x5linux-manjaro ~]$ file success-job 
success-job: Cups Raster version 2, Big Endian, 300x300 dpi, 2480x3508 pixels 8 bits/color 8 bits/pixel ColorOrder=Chunky ColorSpace=sGray
[rilian@5x5linux-manjaro ~]$ 

```

rilian-la-te, 2022-07-27:
@tillkamppeter fixed the issue by editing PPD (reference earlier).

Correct PPD: 
[printer.ppd.txt](https://github.com/OpenPrinting/ipp-usb/files/9203800/printer.ppd.txt)

There was an issue than PPD was set on RGB color model, and RGB color model was not defined in PPD. So, if printer supports only monochrome, we should not define RGB in PPD, or allow saving default settings (see https://github.com/OpenPrinting/cups/issues/415). So, when creating a driverless, we should check printer attributes and set default colormodel only to supported values.

And printing from lp works, I guess, because we have https://github.com/OpenPrinting/cups/issues/421

rilian-la-te, 2023-02-14:
@alexpevzner is the bug fixed?

alexpevzner, 2023-02-14:
@tillkamppeter?