OpenPrinting/ipp-usb #59: Unable to add printer Ricoh with IPP-over-USB

bfg01, 2022-10-24:
Printer is Ricoh SP 377SFNwx. Using openSUSE Leap 15.4, installed from scratch (no upgrade).

As network printer, with URI ipp:// and driver "everywhere", seemingly no problems until now.

Wanted to try with IPP-over-USB.
So connected printer via USB, and tried "lpinfo -v". There was this found URI "direct usb://UNKNOWN/Printer[...]".
Did internet search and found that an additional daemon was needed: https://github.com/OpenPrinting/ipp-usb
So manually downloaded the RPM package and installed.
Now shown URI was "direct usb://RICOH/SP[...]" (cannot remember it exactly)

So tried adding it, and got:
```
# lpadmin -p <a_name> -Ev usb://RICOH/SP[...] -m everywhere; echo $?
lpadmin: Unable to connect to "RICOH:0": Name or service not known
1
```

Tried several things:
---Double-checking that firewalld was -temporarily- disabled (otherwise "lpinfo -v" wouldn't even have worked)
---Trying URI ipp://RICOH/SP[...] instead
---Trying the "lsusb" advice here https://wiki.debian.org/CUPSDriverlessPrinting#IPP-over-USB:_The_Basics ; and I did get the right output like there says, so printer does have support for this

Could someone help please?
Thanks for your attention.

zdohnal, 2022-10-24:
@bfg01 I've passed the issue to ipp-usb project for further investigation.

bfg01, 2022-10-24:
Thanks.

I'm sorry, I made a mistake:
command "driverless [list]" shows no URIs at all, and instead fails with exit status 1.

bfg01, 2022-10-26:
By chance, any findings?

BTW, legacy "usb://..." with model's PPD from OpenPrinting website doesn't work neither...
"lpadmin" does succeed in adding the queue with no errors, but when trying to print anything either with any GUI app or "lp" command, the print job stays hung infinitely, with no activity from printer nor any console output. I always have to use the "cancel" command to remove the print job.

bfg01, 2022-10-26:
Regarding ipp-usb here, since log was composed **mostly** by just repeated lines, I shortened it and I think it was not necessary to post it as attachment. Log:
```
21-10-2022 18:09:47:   ===============================
21-10-2022 18:09:47:   ipp-usb started in "udev" mode, pid=11057
21-10-2022 18:09:47:   ctrlsock: listening at "/var/ipp-usb/ctrl"
21-10-2022 18:09:47: + PNP Bus 001 Device 013: added
21-10-2022 18:09:47: ! PNP Bus 001 Device 013: libusb_claim_interface: Invalid parameter
21-10-2022 18:09:49: + PNP Bus 001 Device 013: retry
21-10-2022 18:09:49: ! PNP Bus 001 Device 013: libusb_claim_interface: Invalid parameter
21-10-2022 18:09:52: + PNP Bus 001 Device 013: retry
21-10-2022 18:09:52: ! PNP Bus 001 Device 013: libusb_claim_interface: Invalid parameter
21-10-2022 18:09:55: + PNP Bus 001 Device 013: retry
[...]
21-10-2022 18:20:07: ! PNP Bus 001 Device 013: libusb_claim_interface: Invalid parameter
21-10-2022 18:20:09: - HOTPLUG: removed Bus 001 Device 013
21-10-2022 18:20:09: - PNP Bus 001 Device 013: removed
21-10-2022 18:20:09:   No IPP-over-USB devices present, exiting
21-10-2022 18:20:09:   ctrlsock: shutdown
21-10-2022 18:20:09:   ipp-usb finished
24-10-2022 09:50:01:   ===============================
24-10-2022 09:50:01:   ipp-usb started in "udev" mode, pid=4294
24-10-2022 09:50:01:   ctrlsock: listening at "/var/ipp-usb/ctrl"
24-10-2022 09:50:01: + PNP Bus 001 Device 006: added
24-10-2022 09:50:01: ! PNP Bus 001 Device 006: libusb_claim_interface: Invalid parameter
24-10-2022 09:50:03: + PNP Bus 001 Device 006: retry
24-10-2022 09:50:04: ! PNP Bus 001 Device 006: libusb_claim_interface: Invalid parameter
24-10-2022 09:50:06: + PNP Bus 001 Device 006: retry
[...]
24-10-2022 10:31:29: ! PNP Bus 001 Device 007: libusb_open: No such device (it may have been disconnected)
24-10-2022 10:31:29: - HOTPLUG: removed Bus 001 Device 007
24-10-2022 10:31:29: - PNP Bus 001 Device 007: removed
24-10-2022 10:31:36:   interrupt signal received, exiting
24-10-2022 10:31:36:   ctrlsock: shutdown
24-10-2022 10:31:36:   ipp-usb finished
24-10-2022 10:31:41:   ===============================
24-10-2022 10:31:41:   ipp-usb started in "udev" mode, pid=4251
24-10-2022 10:31:41:   ctrlsock: listening at "/var/ipp-usb/ctrl"
24-10-2022 10:31:41: + PNP Bus 001 Device 008: added
24-10-2022 10:31:41: ! PNP Bus 001 Device 008: libusb_claim_interface: Invalid parameter
24-10-2022 10:31:43: + PNP Bus 001 Device 008: retry
24-10-2022 10:31:44: ! PNP Bus 001 Device 008: libusb_claim_interface: Invalid parameter
24-10-2022 10:31:46: + PNP Bus 001 Device 008: retry
[...]
25-10-2022 12:31:06: ! PNP Bus 001 Device 007: libusb_claim_interface: Invalid parameter
25-10-2022 12:31:07: - HOTPLUG: removed Bus 001 Device 007
25-10-2022 12:31:07: - PNP Bus 001 Device 007: removed
25-10-2022 12:31:07:   No IPP-over-USB devices present, exiting
25-10-2022 12:31:07:   ctrlsock: shutdown
25-10-2022 12:31:07:   ipp-usb finished
```

Does it say something?
Thanks again.

alexpevzner, 2022-11-02:
Hi @bfg01,

sorry for delay. May I ask you to show me output of the `lsusb -v` command for this device?

bfg01, 2022-11-12:
@alexpevzner ,
Just attached here.

Brief offtopic:
How does the '@' pinging work here? I've been "pinged" like this before, however no notification at all has ever appeared anywhere in the Github web interface...
[lsusbOut.txt](https://github.com/OpenPrinting/ipp-usb/files/9995607/lsusbOut.txt)

alexpevzner, 2022-11-12:
Thanks, @bfg01,

Regarding "pinging", I will receive notifications from this topic in any case, because I'm the member of this discussion (and project owner), so I receive notifications in any case.

alexpevzner, 2022-11-12:
Hi @bfg01,

I hope I've fixed this bug, but cant't test by myself due to a lack of the appropriate device.

Can you build `ipp-usb` and test it by yourself?

bfg01, 2022-11-14:
@alexpevzner ,
Sorry sir, still exact same errors...

Posting all logs, using Debian 11:
When building:
```
$ make; echo $?
gotags -R . > tags
/bin/sh: 1: gotags: not found
make: [Makefile:12: all] Error 127 (ignored)
go build -ldflags "-s -w" -tags nethttpomithttp2
go: downloading github.com/OpenPrinting/goipp v1.0.0
0
$
```

Ipp-usb log:
```
$ sudo Downloads/ipp-usb/ipp-usb; echo $?
  ===============================
  ipp-usb started in "debug" mode, pid=26356
  ctrlsock: listening at "/var/ipp-usb/ctrl"
+ HOTPLUG: added Bus 001 Device 020
+ PNP Bus 001 Device 020: added

  ===============================
+ Bus 001 Device 020: added RICOH SP 377SFNwX
  Device info:
    Ident:         05ca-0459-RH11117C-RICOH-SP-377SFNwX
    Manufacturer:  RICOH
    Product:       RICOH SP 377SFNwX
    SerialNumber:  RH11117C
    MfgAndProduct: RICOH SP 377SFNwX
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [*] (/home/user/Downloads/ipp-usb/ipp-usb-quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      init-delay = 0s
      request-delay = 0s
      http-connection = ""

  USB stack parameters
    authorized_default:          -1
    autosuspend:                 2
    blinkenlights:               N
    initial_descriptor_timeout:  5000
    nousb:                       N
    old_scheme_first:            N
    usbfs_memory_mb:             16
    usbfs_snoop:                 N
    usbfs_snoop_max:             65536
    use_both_schemes:            Y

  USB interfaces:
    Config Interface Alt Class SubClass Proto
       1       0      0   7      1       2
       1       1      0   255    255     255
*      1       3      0   7      1       4
*      1       4      0   7      1       4
*      1       5      0   7      1       4
*      1       6      0   7      1       4

  USB[0]: open: Bus 001 Device 020 Interface 3 Alt 0
  USB[1]: open: Bus 001 Device 020 Interface 4 Alt 0
  USB[2]: open: Bus 001 Device 020 Interface 5 Alt 0
  USB[3]: open: Bus 001 Device 020 Interface 6 Alt 0
> HTTP[000]: POST http://localhost:60000/ipp/print
> HTTP[000]: request body: got 462 bytes; closed
> HTTP[000]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- --- --- ---
  HTTP[000]: connection 0 allocated
< HTTP[000]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[000]: response body: got 1236 bytes; EOF
  USB[0]: connection released, 0 in use: --- --- --- ---
> HTTP[001]: POST http://localhost:60000/ipp/faxout
> HTTP[001]: request body: got 463 bytes; closed
> HTTP[001]: body is small (463 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a-- --- ---
  HTTP[001]: connection 1 allocated
< HTTP[001]: POST http://localhost:60000/ipp/faxout - 200 OK
< HTTP[001]: response body: got 1236 bytes; EOF
  USB[1]: connection released, 0 in use: --- --- --- ---
  IPP FaxOut service detected
> HTTP[002]: GET http://localhost:60000/eSCL/ScannerCapabilities
  USB[2]: connection allocated, 1 in use: --- --- a-- ---
  HTTP[002]: connection 2 allocated
< HTTP[002]: GET http://localhost:60000/eSCL/ScannerCapabilities - 200 OK
< HTTP[002]: response body: got 6854 bytes; EOF
  USB[2]: connection released, 0 in use: --- --- --- ---
> RICOH SP 377SFNwX (d01913): _printer._tcp TXT record:
> RICOH SP 377SFNwX (d01913): _ipp._tcp TXT record:
    air=none
    mopria-certified=1.3
    rp=ipp/print
    priority=50
    kind=document
    PaperMax=legal-A4
    URF=V1.4,W8,DM3,IS1,CP99,PQ3,OB10,RS600,RS200
    UUID=6861d9b0-a100-11e0-8264-002673d01913
    Color=F
    Duplex=T
    note=
    qtotal=1
    usb_MDL=SP 377SFNwX
    usb_MFG=RICOH
    usb_CMD=:JBGRD,URF
    ty=RICOH SP 377SFNwX
    product=(RICOH SP 377SFNwX)
    pdl=image/urf,image/pwg-raster,application/octet-stream
    txtvers=1
    adminurl=http://localhost:60000/index.html
    Fax=T
    rfo=ipp/faxout
    Scan=T
> RICOH SP 377SFNwX (d01913): _uscan._tcp TXT record:
    duplex=T
    is=platen,adf
    cs=binary,color,grayscale
    UUID=6861d9b0-a100-11e0-8264-002673d01913
    adminurl=http://377SFNwX-D01913.local:80/index.html
    representation=http://377SFNwX-D01913.local:80/RemyMF2_128.png
    pdl=application/pdf,image/jpeg
    ty=RICOH SP 377SFNwX
    rs=eSCL
    vers=2.62
    txtvers=1
> RICOH SP 377SFNwX (d01913): _http._tcp TXT record:
> RICOH SP 377SFNwX (d01913): _ipp-usb._tcp TXT record:
  DNS-SD: RICOH SP 377SFNwX (d01913) (USB): trying
  DNS-SD: FQDN: "host1.local"
  DNS-SD: FQDN: "host1.local"->"localhost"
  DNS-SD: +subtype: "_universal._sub._ipp._tcp"
+ DNS-SD: RICOH SP 377SFNwX (d01913) (USB): publishing requested
  DNS-SD: RICOH SP 377SFNwX (d01913) (USB): AVAHI_ENTRY_GROUP_REGISTERING
  DNS-SD: RICOH SP 377SFNwX (d01913) (USB): AVAHI_ENTRY_GROUP_ESTABLISHED
  DNS-SD: RICOH SP 377SFNwX (d01913) (USB): published
- HOTPLUG: removed Bus 001 Device 020
- PNP Bus 001 Device 020: removed
- DNS-SD: RICOH SP 377SFNwX (d01913) (USB): removed
  USB[0]: closed
  USB[1]: closed
  USB[2]: closed
  USB[3]: closed
- Bus 001 Device 020: removed RICOH SP 377SFNwX
^C  interrupt signal received, exiting
  ctrlsock: shutdown
  ipp-usb finished
0
$
```

CUPS version here is 2.3.3. Here I had to manually start avahi-daemon. Not sure if this matters...

Thanks again.

alexpevzner, 2022-11-14:
@bfg01,

I see no errors in this log. The log looks just perfect. What actually doesn't work at this step?

alexpevzner, 2022-11-14:
While `ipp-usb` is running, please show me output of the following 3 commands:
```
avahi-browse -rt _ipp._tcp
avahi-browse -rt _uscan._tcp
scanimage -L
```

bfg01, 2022-11-14:
@alexpevzner
Command "driverless [list]" just exits with status '1', and "lpinfo -v" show no URIs, except the legacy "usb://..."

alexpevzner, 2022-11-14:
`avahi-daemon` ,ust be running for this stuff to work...

bfg01, 2022-11-14:
@alexpevzner ,
Apologies sir, it turns that it was actually a broken USB port, which most of the time no longer makes contact...
Tried again, and this time it worked and was finally able to print, at least the CUPS test page.

Just one last -a bit fool- doubt, couldn't this fix affect printers using *continuous* USB interface listing?

Thanks very much for all your help.

alexpevzner, 2022-11-14:
@bfg01,

regardless of your broken USB port, there was a real bug that was fixed, thanks to your report.

> Just one last -a bit fool- doubt, couldn't this fix affect printers using continuous USB interface listing?

libusb returns me, roughly speaking, an array of interface descriptors. Each structure in this array contains, among other things, interface index within a device, used to open the interface. Usually these indices correspond 1:1 to the structure position in the array, but it is not guaranteed. By mistake, I've used structure position in the array instead of the interface index contained within each structure of that array. This mistake was not noticed before your report, so thanks for that!

closed 2022-11-14 by alexpevzner

alexpevzner, 2022-11-14:
Hi @zdohnal,

JFYI, I've just released the `0.9.23` version which includes this fix...

zdohnal, 2022-11-21:
Hi @alexpevzner ,

thank you for the ping! I'll update the package in Fedora today.

I'm glad you are okay with all things going on in the world,

Zdenek

alexpevzner, 2022-11-21:
Hi @zdohnal,

Glad to hear from you.

I'm more or less fine these hard days. Found a full time job in a big enterprise (or, to be  precise, that job has found me). Believe or not, this is my first time in the such a big enterprise, quite interesting experience. The only drawback, I have almost no spare time for working on my open source projects.