OpenPrinting/ipp-usb #62: Samsung Xpress M2826ND PCL Error

sasikumar-mhc, 2023-01-23:
[image]

There werent any logs available at /var/log/ ???

Thanks.

alexpevzner, 2023-01-23:
Hi,

the problem seems to be at the CUPS side. @michaelrsweet, @tillkamppeter, could you please have a look?

zdohnal, 2023-01-24:
@sasikumar-mhc have you printed to the printer created from ipp-usb?

@alexpevzner IIRC we send PDF or Apple/PWG raster to the printer when using driverless, not PCL - it sounds  the old print queue was used, or the printer has IPP-over-USB interface, but accepts only PCL.

sasikumar-mhc, 2023-01-24:
@zdohnal using ubuntu mate 22.04.1 which comes preinstalled with ipp-usb and printer auto installed after plugging it in? We have a variety of printers and i have to remove ipp-usb for the printers which are having issues and install their oem drivers. Have already reported the issues with certain printers. Hope I was able to answer your question.

tillkamppeter, 2023-01-24:
Could you do the following test:
```
sudo snap install ipp-usb
driverless
driverless --std-ipp-uris
ippfind
```
and post the screen output of all these commands here.
Then do
```
ipptool -tv ipp://localhost:60000/ipp/print get-printer-attributes.test > attrs.txt
```
and attach `attrs.txt` to this bug report.
Now run
```
lpadmin -p testipp -E -v ipp://localhost:60000/ipp/print -m everywhere
```
Do you get any error messages? Are you able to print on the printer `testipp`?
Run also
```
lpadmin -p testipp2 -E -v ipp://localhost:60000/ipp/print -m driverless:ipp://localhost:60000/ipp/print
```
Do you get any other messages than just a warning that printer drivers are deprecated? Are you able to print on the printer `testipp2`?
Please post also the output of these `lpadmin` commands here.
If you could not print on both of these test queues, uninstall ipp-usb again, so that you can print on your original queue.
```
sudo snap remove ipp-usb
lpadmin -x testipp
lpadmin -x testipp2
```

sasikumar-mhc, 2023-01-25:
@alexpevzner  @zdohnal  @tillkamppeter
I would like to explain the scenario of the error. Its actually printing the contents of the document under certain scenarios.

1. Single Page Print ( No Issues )
2. Multiple Pages, Single Side Print (No Issues)
3. Multiple Pages, Duplex Print (No Issues)
4. Single Page Multiple Copies (No Issues)
5. Multiple Page Duplex Print Multiple Copies - I Tried 2 Pages - 2 Copies (Received PCL Error Page instead of Second Copy)

[image]

@tillkamppeter Will update You in a while..

Thanks.

sasikumar-mhc, 2023-01-25:
sudo snap install --edge ipp-usb
driverless
driverless --std-ipp-uris
ippfind
snap "ipp-usb" is already installed, see 'snap help refresh'

$ ipptool -tv ipp://localhost:60000/ipp/print get-printer-attributes.test > attrs.txt
ipptool: Unable to connect to "localhost" on port 60000 - Host is down

lpadmin -p testipp -E -v ipp://localhost:60000/ipp/print -m everywhere

> (No Action or Output)

$ lpadmin -p testipp2 -E -v ipp://localhost:60000/ipp/print -m driverless:ipp://localhost:60000/ipp/print
lpadmin: Unable to open PPD "/tmp/038b063d7078a": Missing PPD-Adobe-4.x header on line 0.

lpadmin -x testipp

> (No Action or Output)

lpadmin -x testipp2
lpadmin: The printer or class does not exist.

I have noticed the issue now exists in OEM driver also..
Maybe issue is with the printer??
We have another printer of the same model and the issue was present in it with ipp-usb installed and after install oem driver.. now seems okay.. but it is still under observation..

Thanks.