OpenPrinting/ipp-usb #63: HP Multifunction Printers - Scanning doesnt work

sasikumar-mhc, 2023-01-23:
Hi,
HP Printer Model : LaserJetPro MFP M329dw and 1 more HP MFT Printer(Will share the model number soon).
I have tried using document scanner in ubuntu mate. It always shows as failed to connect to scanner.

[ipp-usb-hp- laserjet pro MFP M329dw.zip](https://github.com/OpenPrinting/ipp-usb/files/10479104/ipp-usb-hp-.laserjet.pro.MFP.M329dw.zip)

Had to remove ipp-usb and installed hplip and hp-plugin for getting scanner to work.

Attached whatever logs were available..

Thanks.

alexpevzner, 2023-01-23:
Hi,

with `ipp-usb` proprietary drivers are not expected to work, but `sane-airscan` and `sane-escl` should work. Did you try some of them together with the `ipp-usb`?

debiantriage, 2023-01-23:
-  Disconnect from USB.
- Install ipp-usb.
- Connect to USB
- Give outputs from
`avahi-browse -rt _ipp._tcp`
`avahi-browse -rt _uscan._tcp`

alexpevzner, 2023-01-23:
Hi @debiantriage,

from the `ipp-usb` log I see no attempt to scan using eSCL. But scanning part of the printer detected and published to Avahi. Looks like sane-airscan/sane-escl not installed or disabled.

debiantriage, 2023-01-23:
Thank you @alexpevzner  I meant to, but completely forgot to look at the file attached by @sasikumar-mhc ! Maybe @sasikumar-mhc would say which OS is being used?

sasikumar-mhc, 2023-01-24:
@debiantriage  Im using ubuntu mate 22.04.1

sasikumar-mhc, 2023-01-24:
@alexpevzner ubuntu mate comes preinstalled with ipp-usb and im hoping that they also included sane-airscan/sane-escl ??
or should it be installed manually by me?  

@debiantriage 
[ipp_tcp.txt](https://github.com/OpenPrinting/ipp-usb/files/10486687/ipp_tcp.txt)
[ipp-usb.zip](https://github.com/OpenPrinting/ipp-usb/files/10486689/ipp-usb.zip)
[timeout.txt](https://github.com/OpenPrinting/ipp-usb/files/10486690/timeout.txt)
[uscan_tcp.txt](https://github.com/OpenPrinting/ipp-usb/files/10486691/uscan_tcp.txt)

https://drive.google.com/file/d/1mWyLhPrQjzNWEmOFzqurmdaTCniCDhtp/view?usp=share_link

Thanks.

alexpevzner, 2023-01-24:
Hi @sasikumar-mhc,

looks like for now the problems are are actually at the `sane-airscan` side. I need its protocol trace logs.

Please, in the `/etc/sane.d/airscan.conf` file uncomment the following line:
```
[debug]
trace = ~/airscan/trace
```

Logs will be written to the `~/airscan/trace` directory, which will be created automatically. For now I only need `*.log` files, `*.tar` files are not needed.

sasikumar-mhc, 2023-01-24:
@alexpevzner 
[simple-scan-zeroconf.log](https://github.com/OpenPrinting/ipp-usb/files/10488533/simple-scan-zeroconf.log)
[simple-scan-HP-LaserJet-Pro-M329-[8F773A]-(USB).log](https://github.com/OpenPrinting/ipp-usb/files/10488535/simple-scan-HP-LaserJet-Pro-M329-.8F773A.-.USB.log)

Thanks.

alexpevzner, 2023-01-24:
@sasikumar-mhc,

From the printer's response to the `ScanJobs` command, looks like scanning is administratively prohibited:
```
<Details>
    <ElementXPath>/ScanSettings/ContextID</ElementXPath>
    <Description>Scan State is disabled, context_id missing!</Description>
 </Details>
```

Manual (http://h10032.www1.hp.com/ctg/Manual/c06184015.pdf) on a page 78 says the following:

> IMPORTANT: For security purposes, scanning remotely is disabled by default. To enable this feature, in the Embedded Web Server, open the **Settings** tab, click **Security**, click **Administrator Settings**, select the **Enable** checkbox beside **Scan from a Computer or Mobile Device**, and then click **Apply**

This is very likely, that from the administrative point of view, the device doesn't distinguish between eSCL scanning from network and from USB

So I suggest to try to tweak this setting.

sasikumar-mhc, 2023-01-25:
@alexpevzner 
It worked. Thank you.

alexpevzner, 2023-02-25:
Fixed by proper printer configuration

closed 2023-02-25 by alexpevzner