OpenPrinting/ipp-usb #64: HP Color LaserJet Enterprise MFP M578 detection during printer start up

fabled, 2023-03-06:
Somewhat related to #32, I detected additional issues in this HP printer.

When the printer is starting up, it behaves as follows during early boot:

- `/ipp/print` requests return `503 Service not ready`, and sadly it is missing the last `\n` so `ipp-usb` does not see it as full request
- `/eSCL/ScannerCapabilities` requests return `404 Not Found`, and also miss the last `\n`

This results `ipp-usb` to reset and retry device initialization. This happens because the `EsclService` init times out, and the error condition in https://github.com/OpenPrinting/ipp-usb/blob/master/device.go#L120 is triggered.

Later during the startup sequence, the scanner side starts first responding with success, but IPP printer still gives 503. During this condition, `ipp-usb` is left in hung state. The printer is not detected, because the attribute reading aborts early in https://github.com/OpenPrinting/ipp-usb/blob/master/ipp.go#L155 when error is received. Therefor no timeout happens, and https://github.com/OpenPrinting/ipp-usb/blob/master/usbtransport.go#L550 triggers. This tries to indefinitely drain the response which never comes and the channels is left stuck. (Recovery can be done with restart ipp-usb, or disconnect/connect USB cable.)

Perhaps the device init logic could be adjusted to automatically always do timeout retry if ipp return 503 error (either directly, or via quirk).

I wonder also if the response draining could be also also adjusted to cope better with printer bugs such as this one.

alexpevzner, 2023-03-17:
Hi Timo,

> Perhaps the device init logic could be adjusted to automatically always do timeout retry if ipp return 503 error (either directly, or via quirk).

Looks like it is not enough, because regardless of status, if `ipp-usb` can't fetch the entire response, because trailing `\n` is missed, it will stuck anyway.

May be, for this kind of device, it will be better to simply delay initialization after device was connected, based on quirk?

alexpevzner, 2023-03-17:
BTW, `ipp-usb` already has `init-delay` quirk, so the idea to delay initialization is simple to test...

fabled, 2023-03-18:
> Looks like it is not enough, because regardless of status, if `ipp-usb` can't fetch the entire response, because trailing `\n` is missed, it will stuck anyway.

As workaround, I'm using now:
``` diff
--- a/device.go
+++ b/device.go
@@ -84,6 +84,10 @@ func NewDevice(desc UsbDeviceDesc) (*Device, error) {
 
        if err != nil {
                dev.Log.Error('!', "IPP: %s", err)
+               if strings.StartsWith(err, "HTTP: 503") {
+                       err = ErrInitTimedOut
+                       goto ERROR
+               }
        }
 
        log.Flush()
```

This works because the `503` triggers a device init retry doing a reset. Not perhaps the most elegant, but works.

In any case, not receiving the trailing `\n` should be eventually timed out. Seems there's too many buggy printers like that out there. Perhaps receiving few zero-length USB URBs should be sufficient for that.

> May be, for this kind of device, it will be better to simply delay initialization after device was connected, based on quirk?

This is not a real option. When the device is connected via cable, and you hit power on. It takes about 1-2 minutes for the USB interface to come up, and then it takes 4-6 minutes for the IPP layers to be up. For those 4-6 minutes I see the `503` error.

On the other hand, if the printer is online, and I connect the cable, everything works immediately.

Adding a 6 minute `init-delay` that is needed only if the device is cold starting, does not sound feasible to me.

alexpevzner, 2023-03-18:
> Adding a 6 minute init-delay that is needed only if the device is cold starting, does not sound feasible to me.

Yes, 6 minutes looks a little bit too much to wait.

If we suppress the following logic in the `device.go`, line 144, won't it be enough for this device:

```
        if len(dnssdServices) == 0 {
                err = ErrUnusable
                goto ERROR
        }
```

fabled, 2023-03-19:
> If we suppress the following logic in the `device.go`, line 144, won't it be enough for this device:

No. See first comment. During cold boot, there in the end will be a state that the printer query will return 503, but scanner will return 200. During this stage it will permanently detect the MFP as having scanner only and printing will not be available.

alexpevzner, 2024-12-10:
Hi @fabled,

I've recently added the following quirks to resolve issues explained here as well as under #83 and #32:
* `init-timeout = DELAY` allows to tweak initialization timeout
* `init-retry-partial = true | false`. If enabled, the `ipp-usb` will retry initialization, if some parts of device (printer or scanner) didn't initialize properly with some erroneous HTTP status code.
* `zlp-recv-hack = true | false`. If enabled, that at the initialization time zero-length packet reception followed by receive timeout will be considered as EOF, so allowing reception of responses with truncated body.

Please, test on your devices. Waiting for your feedback. If everything is OK, I will need a list of affected devices, to update quirks distributed with the `ipp-usb` (you can write it in a comment or send me a PR)

Hope we will be able to resolve these issues now.

alexpevzner, 2024-12-17:
Hi @fabled,

may I ask you to retest your problematic HP printers with the newly added quirks?

fabled, 2024-12-18:
> may I ask you to retest your problematic HP printers with the newly added quirks?

Sure. But it'll probably by early next year as I am currently on vacation and without access to the devices.

I think these quirks should be applied at least to all HP "enterprise" printers. Would it be possible to match the HTTP response `Server` field in quirks? Alternatively, I can see if there's a wildcarded usb device name match that works.

alexpevzner, 2024-12-27:
Hi Timo,

Thank you for your response.

OK, I will wait until early next year. I want the next release to include these quirks, tested, and proven quirks for devices you have access to.

The `zlp-recv-hack` is risky, it can cause loss of synchronization if used with the "wrong" device, so I don't want to blindly apply it to the wide class of hardware without explicit testing of each of affected devices. So lets be specific in choosing affected devices.

The mechanism to apply the quirk based on a `Server` field doesn't exist for now.

So wish you a nice vacation.

Marry Christmas and happy New Year! :)

fabled, 2025-01-20:
Finally managed to do some testing. The devices at hand match the `MfgAndProduct` as follows:

- HP HP Color LaserJet Flow X677
- Hewlett-Packard HP Color LaserJet FlowMFP M578

The broken scenario is that computer is on with ipp-usb running and printer is connected but power off. When powering on it takes a long time to initializer fully.

The results:

- Only `zlp-recv-hack`. When the printer USB hardware gets initialized, all fails: IPP (http 503) and eSCL (http 404) returns error. ipp-usb marks device initialized and logs "Device doesn't implement print or scan service" and device becomes permanently unavailable.
- Both `zlp-recv-hack` and `init-retry-partial`: Properly waits for IPP to become available, after which the eSCL side fails with `unexpected EOF`. This seems to be side effect of `zlp-recv-hack` which is needed, but now it prematurely fails the eSCL capabilities request. The initial eSCL capabilities request often takes several seconds. The result was working printing and broken scanning.
- Both above + `init-timeout 20s`: Seems to work consistently.

So with current code, the working quirk could be:
```
[* HP Color LaserJet Flow*]
init-timeout = 20s
init-retry-partial = true
zlp-recv-hack = true
```
Though, I believe this is needed for every modern HP Enterprise series printer. There is no simple wildcard match, because there are also LaserJet Pro series printers which might not need this. The HTTP Server response these give out is `HP_Compact_Server` but understadably can be difficult to use for quirks selection. Also, there are likely non-color devices and potentially other devices without Flow or something in between. So it'll likely take time determine good matching for this.

Though, since the IPP and eSCL sides seem to behave somewhat different and have different response latency, I would recommend that the `zlp-recv-hack` triggers only if an http error status is received. Though, I am not sure if this is possible easily, and might get a bit ugly if the lower layer needs to parse higher layer stuff. It currently seems that the error really is only when getting back http error during the device boot up sequence. I *think* the error response after this point are better. But in general there also seems to be HTTP requests where the timeout to give success/error is long, and the ZLP recv hack should not trigger unless at least some amount of data is received (perhaps calculate that at least few lines are received, or the initial header block).

Seems that the retry / reset happens now every two seconds. Which IMHO, is a bit too often for the device. Perhaps this could be made configurable?

alexpevzner, 2025-02-25:
Hi!

Thanks for response and sorry for delay from my side.

> Properly waits for IPP to become available, after which the eSCL side fails with unexpected EOF

This should be fixed by now. **ipp-usb** should keep trying if EOF occurs during the eSCL initialization.

I've also included quirks suggested by you into the set of quirks that come with the **ipp-usb**.

Please, retest. If everything is OK, I', probably ready to issue the next release of the **ipp-usb**.

alexpevzner, 2025-02-26:
I've removed a **println** line, leaked by mistake.

Is there any other problems with the current codebase?

fabled, 2025-02-28:
> Is there any other problems with the current codebase?

I did basic smoke testing, and things look mostly good.

One issue is with the quirks handling still. It seems that HP products depending on the model return as `Manufacturer` either *HP* or *Hewlett-Packard*. Additionally the `Product` seems to always start with *HP* regardless of the `Manufacturer` field.

I did not realize that the `MfgAndProduct` is only conditionally prepended the `Manufacturer` if its not a prefix of the existing `Product`. So the quirk `[* HP Color LaserJet Flow*]` should really be `[*HP Color LaserJet Flow*]` to match all the devices I have.

Alternatively, it might be an idea to rewrite `Manufacturer` *Hewlett-Packard* to *HP*. This is actually what Linux kernel and CUPS does in various places. E.g. https://github.com/OpenPrinting/cups/blob/master/backend/usb-libusb.c#L1343-L1346
If this is done, the quirk match could be just `[HP Color LaserJet Flow*]` because then the manufacturer would not be prepended.

alexpevzner, 2025-02-28:
> So the quirk [* HP Color LaserJet Flow*] should really be [*HP Color LaserJet Flow*] 

Fixed.

> Alternatively, it might be an idea to rewrite Manufacturer Hewlett-Packard to HP.

It is hard to test against the representative population of printers.

Looks like match engine needs to be more flexible. I will think about it.

There enough changes since the latest (0.9.28) release, so I want to put a new release label. Basically, I'm waiting only for this issue to be closed to do so. So, the next release depends solely on your blessing :-)

So I ask you, are we good enough for the next release? :)

fabled, 2025-02-28:
> So I ask you, are we good enough for the next release? :)

I think so. Its a lot better already now. I'm still hunting an issue causing missed pages when printing large amounts of jobs/papers. But that might be also cups or printer hardware issue. But to debug it, I'd greatly appreciate logging to syslog. Created #95  for this. But its not a release blocker for me. I can always apply commits if needed to test stuff.

alexpevzner, 2025-02-28:
So I create the 0.9.29 release and close this issue as completed

closed 2025-02-28 by alexpevzner