OpenPrinting/ipp-usb #65: Race condition in UsbTransport.Shutdown

fabled, 2023-03-17:
I got the following debug output:
```
Bus 001 Device 040: resetting HP Color LaserJet FlowMFP M577
USB[0]: recv: libusb_bulk_transfer: No such device...
USB[0]: connection released, 0 in use: --- --- --- ---
Bus 001 Device 040: shutdown: 1 connections still in use
```
After which things just hang.

What apparently happens is:

Thread A: executes func (conn *usbConn) put()
Thread B: func (transport *UsbTransport) Shutdown(ctx context.Context) error
Thread A: logs "connection released, 0 in use"
Thread B: transport.connInUse returns 1
Thread A: returns connection to pool + async signals the channel
Thread B: starts listening signal and missed it

As an immediate workaround I just added Shutdown to execute the polling for loop every 500ms if no channel message is received. Which is ugly and does not fix the root cause, but workarounds the symptoms.

alexpevzner, 2023-03-17:
Hi @fabled,

I can't reproduce it so I've blindly approached it based on theoretical analysis. Please, test.

fabled, 2023-03-18:
I can reproduce it easily, and will test Monday. But looking at the commit, I think it should work just fine.

alexpevzner, 2023-07-03:
Hi @fabled,

Can I close this issue as resolved?

closed 2023-07-04 by fabled