OpenPrinting/ipp-usb #68: Add support for FreeBSD

cwfoo, 2023-04-25:
I tried to build ipp-usb on FreeBSD 13.1-RELEASE using the following steps:

* `pkg install avahi-app gmake go pkgconf`
    (The avahi-app package provides libavahi-common.so and libavahi-client.so).
* `gmake`

However, this error appears:

```
$ gmake
gotags -R . > tags
/bin/sh: gotags: not found
gmake: [Makefile:12: all] Error 127 (ignored)
go build -ldflags "-s -w" -tags nethttpomithttp2
# github.com/OpenPrinting/ipp-usb
./dnssd.go:130:12: undefined: dnssdSysdep
./dnssd.go:178:21: undefined: newDnssdSysdep
./dnssd.go:283:23: undefined: newDnssdSysdep
gmake: *** [Makefile:13: all] Error 1
```

The build succeeds after this line is removed:

https://github.com/OpenPrinting/ipp-usb/blob/5d0719d595a2438dfc0e56a62b90a1bde82f6577/dnssd_avahi.go#L1

I'm happy to report that I managed to print and scan via ipp-usb on FreeBSD.

Another issue: `gmake PREFIX=/usr/local/ install` will fail because there is no `-t` option in FreeBSD's `install`.

https://github.com/OpenPrinting/ipp-usb/blob/5d0719d595a2438dfc0e56a62b90a1bde82f6577/Makefile#L20-L27

Are there any plans to add official support for FreeBSD? Or is this project supposed to be Linux-only?

alexpevzner, 2023-04-25:
Hi @cwfoo,

I happy to hear that `ipp-usb` is about to become UNIX-wide, not Linux-only solution.

> The build succeeds after this line is removed:

The correct line, which includes FreeBSD would probably look like:
```
// +build freebsd linux
```

If you test it and it works, you probably could create a PR, I will accept it.

> Another issue: gmake PREFIX=/usr/local/ install will fail because there is no -t option in FreeBSD's install

I would appreciate if you could try to find a solution that works equally well on both Linux and BSD (I can help with testing on Linux)

cwfoo, 2023-05-23:
ipp-usb is now in the FreeBSD ports tree.
https://www.freshports.org/print/ipp-usb/

commit 9434a441 (2023-07-03): Added announce of the FreeBSD port (fixes #68)
closed 2023-07-03 by alexpevzner