OpenPrinting/ipp-usb #70: HPLIP, HPLIP.conf and ipp-usb

debiantriage, 2023-06-29:
This does not concern an issue with ipp-usb. It may be of interest to users and developers but I wouldn't be surprised should it be closed as needing no action.

A user [required help](https://answers.launchpad.net/hplip/+question/706974) to print with an HP Smart Tank 5100 Series. He had installed upstream's HPLIP 3.23.5. I do not do HPLIP with a modern device, so asked for outputs from avahi-browse and driverless. These outputs were empty. I was taken aback.

The user provided

> scott@all-in-one:/var/log/cups$ sudo ipp-usb check
[sudo] password for scott:
IPP over USB devices:
Num Device Vndr:Prod Model
> 1. Bus 003 Device 004 03f0:5054 "HP Smart Tank 5100 series"

ipp-usb knows the printer exists but avaahi-browse doesn't. Puzzling.

The use looked at the ipp-usb logs:

> In "main.log", I see that every few seconds it spits out 2 lines very much like the following:
28-06-2023 12:01:58: + PNP Bus 003 Device 004: retry
28-06-2023 12:01:58: ! PNP Bus 003 Device 004: Device is blacklisted

and

> 28-06-2023 12:08:28: Device quirks:
28-06-2023 12:08:28: from [HP*] (/usr/share/ipp-usb/quirks/HPLIP.conf:1)
28-06-2023 12:08:28: blacklist = true

HPLIP.conf is a file proved by HPLIP. It effectively kills ipp-usb by blacklisting **all** HP debices. How smart is that?

alexpevzner, 2023-06-29:
Wow! We are being recognized by the whole Hewlett-Packard company!

Well, we probably could add a whitelist option which overrides blacklist...

zdohnal, 2023-06-30:
Well, that's why I highly recommend that if you turn to OS forums/portals, use the packages from your OS if they are available.
I guess every distro removes that config file (I do it at least during packaging and I didn't find it in Ubuntu either, but I didn't look too deep), so that's the best we can do. I would not recommend whitelisting a blacklist in upstream code (if it wasn't sarcasm :) ).

To sum it up, if you install project from a different source than OS repos, you have to know what are you doing and if something doesn't work, ask that source for help.