OpenPrinting/ipp-usb #72: Printing impossible after disconnect at wrong time

Xerxekyran, 2023-10-31:
Hi there,

I am facing an issue where printing is not possible anymore until I restart the printer. When I try to print something and disconnect the USB-connection during this process (when I see the "! USB[1]: zero-size read" log entries in the IPP-USB printer log) then sometimes the printer initialization of IPP-USB gets stuck in an endless initilization loop. Re-connecting the printer or even restarting IPP-USB does not help. Only shutting down the printer and starting it up again. I know that this is not just a problem of this one printer (HP OfficeJet 200 Mobile Series), but also for example a Brother printer acted like this.
Is there a way of getting more information about what the actual problem is? Here are my logs for this situation:

```
// printing process:
31-10-2023 08:38:13: ! USB[1]: zero-size read
31-10-2023 08:38:14: ! USB[0]: zero-size read
31-10-2023 08:38:14: ! USB[1]: zero-size read
31-10-2023 08:38:14: ! USB[0]: zero-size read
31-10-2023 08:38:14: ! USB[1]: zero-size read
31-10-2023 08:38:15: ! USB[1]: zero-size read
31-10-2023 08:38:15: ! USB[1]: zero-size read

// disconnecting USB (the printer still has power but is not connected anymore)
31-10-2023 08:38:16: - DNS-SD: HP OfficeJet 200 Mobile Series [17EF3C] (USB): removed
31-10-2023 08:38:16: ! HTTP[155]: read tcp 127.0.0.1:60000->127.0.0.1:44058: use of closed network connection
31-10-2023 08:38:16: ! HTTP[155]: read tcp 127.0.0.1:60000->127.0.0.1:44058: use of closed network connection
31-10-2023 08:38:16: - Bus 003 Device 011: removed OfficeJet 200 Mobile Series
31-10-2023 08:38:29: + Bus 003 Device 012: added OfficeJet 200 Mobile Series

// and the loop begins
31-10-2023 08:38:35: ! USB[0]: send: libusb_bulk_transfer: Operation timed out
31-10-2023 08:38:35: ! HTTP[158]: libusb_bulk_transfer: Operation timed out
31-10-2023 08:38:35: ! IPP: HTTP: Post "http://localhost:60000/ipp/print": libusb_bulk_transfer: Operation timed out
31-10-2023 08:38:35: - Bus 003 Device 012: resetting OfficeJet 200 Mobile Series
31-10-2023 08:38:35: - Bus 003 Device 012: removed OfficeJet 200 Mobile Series
31-10-2023 08:38:37: + Bus 003 Device 012: added OfficeJet 200 Mobile Series
31-10-2023 08:38:38: ! HTTP[160]: ipp-usb is not ready for this device
31-10-2023 08:38:38: ! HTTP[161]: ipp-usb is not ready for this device
31-10-2023 08:38:39: ! HTTP[162]: ipp-usb is not ready for this device
31-10-2023 08:38:39: ! HTTP[163]: ipp-usb is not ready for this device
31-10-2023 08:38:40: ! HTTP[164]: ipp-usb is not ready for this device
31-10-2023 08:38:40: ! HTTP[165]: ipp-usb is not ready for this device
31-10-2023 08:38:42: ! HTTP[166]: ipp-usb is not ready for this device
31-10-2023 08:38:42: ! HTTP[167]: ipp-usb is not ready for this device
31-10-2023 08:38:42: ! USB[0]: send: libusb_bulk_transfer: Operation timed out
31-10-2023 08:38:42: ! HTTP[159]: libusb_bulk_transfer: Operation timed out
31-10-2023 08:38:42: ! IPP: HTTP: Post "http://localhost:60000/ipp/print": libusb_bulk_transfer: Operation timed out
31-10-2023 08:38:42: - Bus 003 Device 012: resetting OfficeJet 200 Mobile Series
31-10-2023 08:38:43: - Bus 003 Device 012: removed OfficeJet 200 Mobile Series
31-10-2023 08:38:45: + Bus 003 Device 012: added OfficeJet 200 Mobile Series
```

Thank you for any help on this topic.

Regards
Lars

Xerxekyran, 2023-10-31:
I set the IPP-USB logs to debug to get some more information here:

```
// printing file 
31-10-2023 09:50:10: ! USB[1]: zero-size read
31-10-2023 09:50:10: < HTTP[014]: POST http://localhost:60000/ipp/print - 200 OK
31-10-2023 09:50:10: < HTTP[014]: response body: got 1440 bytes; EOF
31-10-2023 09:50:10:   USB[1]: connection released, 1 in use: a-- ---
31-10-2023 09:50:10: > HTTP[015]: POST http://localhost:60000/ipp/print
31-10-2023 09:50:10: > HTTP[015]: request body: got 334 bytes; EOF
31-10-2023 09:50:10: > HTTP[015]: body is small (334 bytes), prefetched before sending
31-10-2023 09:50:10:   USB[1]: connection allocated, 2 in use: a-- a--
31-10-2023 09:50:10:   HTTP[015]: connection 1 allocated
31-10-2023 09:50:10: ! USB[1]: zero-size read
31-10-2023 09:50:10: < HTTP[015]: POST http://localhost:60000/ipp/print - 200 OK
31-10-2023 09:50:10: < HTTP[015]: response body: got 285 bytes; EOF
31-10-2023 09:50:10:   USB[1]: connection released, 1 in use: a-- ---
31-10-2023 09:50:11: > HTTP[016]: POST http://localhost:60000/ipp/print
31-10-2023 09:50:11: > HTTP[016]: request body: got 740 bytes; EOF
31-10-2023 09:50:11: > HTTP[016]: body is small (740 bytes), prefetched before sending
31-10-2023 09:50:11:   USB[1]: connection allocated, 2 in use: a-- a--
31-10-2023 09:50:11:   HTTP[016]: connection 1 allocated
31-10-2023 09:50:11: ! USB[1]: zero-size read

// printer cable disconnected
31-10-2023 09:50:11: ! USB[1]: recv: libusb_bulk_transfer: Input/Output Error
31-10-2023 09:50:11: ! HTTP[016]: libusb_bulk_transfer: Input/Output Error
31-10-2023 09:50:11:   USB[1]: connection released, 1 in use: a-- ---
31-10-2023 09:50:11: > HTTP[016]: POST http://localhost:60000/ipp/print
31-10-2023 09:50:11: ! HTTP[016]: libusb_bulk_transfer: Input/Output Error
31-10-2023 09:50:11: > HTTP[017]: POST http://localhost:60000/ipp/print
31-10-2023 09:50:11: > HTTP[017]: request body: got 334 bytes; EOF
31-10-2023 09:50:11: > HTTP[017]: body is small (334 bytes), prefetched before sending
31-10-2023 09:50:11:   USB[1]: connection allocated, 2 in use: a-- a--
31-10-2023 09:50:11:   HTTP[017]: connection 1 allocated
31-10-2023 09:50:11: ! USB[1]: send: libusb_bulk_transfer: Input/Output Error
31-10-2023 09:50:11: ! HTTP[017]: libusb_bulk_transfer: Input/Output Error
31-10-2023 09:50:11:   USB[1]: connection released, 1 in use: a-- ---
31-10-2023 09:50:11: > HTTP[017]: POST http://localhost:60000/ipp/print
31-10-2023 09:50:11: ! HTTP[017]: libusb_bulk_transfer: Input/Output Error
31-10-2023 09:50:11: - DNS-SD: HP OfficeJet 200 Mobile Series [17EF3C] (USB): removed
31-10-2023 09:50:11: > HTTP[013]: request body: got 238 bytes; read tcp 127.0.0.1:60000->127.0.0.1:54250: use of closed network connection
31-10-2023 09:50:11: ! HTTP[013]: read tcp 127.0.0.1:60000->127.0.0.1:54250: use of closed network connection
31-10-2023 09:50:11:   USB[0]: connection released, 0 in use: --- ---
31-10-2023 09:50:11: > HTTP[013]: POST http://localhost:60000/ipp/print
31-10-2023 09:50:11: ! HTTP[013]: read tcp 127.0.0.1:60000->127.0.0.1:54250: use of closed network connection
31-10-2023 09:50:11:   USB[0]: closed
31-10-2023 09:50:11:   USB[1]: closed
31-10-2023 09:50:11: - Bus 003 Device 003: removed OfficeJet 200 Mobile Series

// printer connected again
31-10-2023 09:50:21:
31-10-2023 09:50:21:   ===============================
31-10-2023 09:50:21: + Bus 003 Device 004: added OfficeJet 200 Mobile Series
31-10-2023 09:50:21:   Device info:
31-10-2023 09:50:21:     Ident:         03f0-e711-TH9BQ850DD0689-HP-OfficeJet-200-Mobile-Series
31-10-2023 09:50:21:     Manufacturer:  HP
31-10-2023 09:50:21:     Product:       OfficeJet 200 Mobile Series
31-10-2023 09:50:21:     SerialNumber:  TH9BQ850DD0689
31-10-2023 09:50:21:     MfgAndProduct: HP OfficeJet 200 Mobile Series
31-10-2023 09:50:21:     BasicCaps:     print,scan,fax,http
31-10-2023 09:50:21:
31-10-2023 09:50:21:   Device quirks:
31-10-2023 09:50:21:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
31-10-2023 09:50:21:       blacklist = false
31-10-2023 09:50:21:       usb-max-interfaces = 0
31-10-2023 09:50:21:       disable-fax = false
31-10-2023 09:50:21:       init-delay = 0s
31-10-2023 09:50:21:       request-delay = 0s
31-10-2023 09:50:21:       http-connection = ""
[... 138 lines cut ...]
31-10-2023 09:50:36:        1       1      0   8      6       80
31-10-2023 09:50:36:        1       2      0   255    4       1
31-10-2023 09:50:36:        1       3      0   255    4       1
31-10-2023 09:50:36: *      1       3      1   7      1       4
31-10-2023 09:50:36:
31-10-2023 09:50:36:   USB[0]: open: Bus 003 Device 004 Interface 0 Alt 1
31-10-2023 09:50:36:   USB[1]: open: Bus 003 Device 004 Interface 3 Alt 1
31-10-2023 09:50:36: > HTTP[026]: POST http://localhost:60000/ipp/print
31-10-2023 09:50:36: > HTTP[026]: request body: got 462 bytes; closed
31-10-2023 09:50:36: > HTTP[026]: body is small (462 bytes), prefetched before sending
31-10-2023 09:50:36:   USB[0]: connection allocated, 1 in use: a-- ---
31-10-2023 09:50:36:   HTTP[026]: connection 0 allocated
31-10-2023 09:50:39: > HTTP[027]: POST /ipp/print
31-10-2023 09:50:39: ! HTTP[027]: ipp-usb is not ready for this device
31-10-2023 09:50:39: > HTTP[028]: POST /ipp/print
31-10-2023 09:50:39: ! HTTP[028]: ipp-usb is not ready for this device
31-10-2023 09:50:41: ! USB[0]: send: libusb_bulk_transfer: Operation timed out
31-10-2023 09:50:41: ! HTTP[026]: libusb_bulk_transfer: Operation timed out
31-10-2023 09:50:41:   USB[0]: connection released, 0 in use: --- ---
31-10-2023 09:50:41: ! IPP: HTTP: Post "http://localhost:60000/ipp/print": libusb_bulk_transfer: Operation timed out
31-10-2023 09:50:41: - Bus 003 Device 004: resetting OfficeJet 200 Mobile Series
31-10-2023 09:50:41:   USB[0]: closed
31-10-2023 09:50:41:   USB[1]: closed
31-10-2023 09:50:41: - Bus 003 Device 004: removed OfficeJet 200 Mobile Series
```

alexpevzner, 2023-10-31:
Hi @Xerxekyran!

First of all, seems your `ipp-usb` is not of the latest version. The latest is `0.9.23` for now.

If upgrade will not help, could you please play a bit with `init-delay` and `init-reset` quirks? For simplicity, you can edit direcly the `/usr/share/ipp-usb/quirks/default.conf` file (for all devices); if it will help, I'll tell later how to make these settings more specific.

Xerxekyran, 2023-10-31:
Hi @alexpevzner thank you for your quick answer. I am currently playing around with different settings for init-delay, init-reset, disable-fax and http-connection in the default.conf but could not yet find a setting where this problem is solved. 
I will try upgrading next.

Xerxekyran, 2023-10-31:
> First of all, seems your `ipp-usb` is not of the latest version. The latest is `0.9.23` for now.

I checked my build scripts and there it states that I use version 0.9.23 already. What made you think I am using an older version?

Xerxekyran, 2023-10-31:
Even when using relatively high values it seems not to recover from the "failure state of the printer":

```
31-10-2023 12:37:11:   Device quirks:
31-10-2023 12:37:11:     from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
31-10-2023 12:37:11:       blacklist = false
31-10-2023 12:37:11:       usb-max-interfaces = 0
31-10-2023 12:37:11:       disable-fax = true
31-10-2023 12:37:11:       init-delay = 5s
31-10-2023 12:37:11:       request-delay = 5s
31-10-2023 12:37:11:       init-reset = hard
31-10-2023 12:37:11:       http-connection = ""
```

I tried hard, soft and none init-reset, high and low values for the delay and with or without the fax feature. Is there anything else I can test?

alexpevzner, 2023-10-31:
> > First of all, seems your `ipp-usb` is not of the latest version. The latest is `0.9.23` for now.
> 
> I checked my build scripts and there it states that I use version 0.9.23 already. What made you think I am using an older version?

Absence of the `init-reset` quirk parameter in your early log. This parameter was added in 0.9.21

alexpevzner, 2023-10-31:
> I know that this is not just a problem of this one printer (HP OfficeJet 200 Mobile Series), but also for example a Brother printer acted like this.

Do you know the model name of this Brother?

alexpevzner, 2023-11-01:
BTW, #66 looks similar. Does tweaking `usb-max-interfaces` help in your case (0 means "unlimited", your device has only 2 interfaces, so the only reasonable number to try is 1)?

alexpevzner, 2023-11-01:
BTW, my colleague suggested a reasonable idea. Have you tried to replace an USB cable?

Xerxekyran, 2023-11-03:
> Do you know the model name of this Brother?

Brother HL-L2370DW

> BTW, my colleague suggested a reasonable idea. Have you tried to replace an USB cable?

This problem was reported to me from a person at a different location, so there was a different cable in use. 

Further testing showed me that even printing directly to the printer via USB does not work (shutting IPP-USB off) as long as it is in this failure state. I needed to restart the printer completely by powering it off to recover it. After that everything works fine again.

Xerxekyran, 2023-11-03:
> BTW, #66 looks similar. Does tweaking `usb-max-interfaces` help in your case (0 means "unlimited", your device has only 2 interfaces, so the only reasonable number to try is 1)?

Setting this to 2 or 1 does not "heal" the problematic state. I will check if it prevents getting to the state as soon as I can get my hands on the printer again in a couple of days.

I appreciate your help, thank you.

alexpevzner, 2023-12-02:
Hi @Xerxekyran,

> What do you think, do you see an IPP-USB issue here at all?

'ipp-usb` is the communication daemon only. The fact that when problematic job is removed, printing recovers, tells that this is unlikely to be the `ipp-usb` issue.

However, it may be hard for CUPS guys to deal with this problem at their side (and there are not too much guys, after all).

I'll try to find one of devices you've mentioned locally and if I will succeed, it will be best of all. Otherwise, we will try to debug remotely, by e-mail :)

volfyd, 2025-01-31:
I found this issue because I have a Brother HL-L2350DW and ipp-usb would show

> USB[1]: zero-size read

I set

> usb-max-interfaces = 1

because of @alexpevzner  's suggestion and that worked and the connection now seems stable. I can browse port 60000 without every other request causing the zero-size read.