OpenPrinting/ipp-usb #73: Fail to add Canon Selphy CP1500 printer

bwolleb, 2023-11-06:
Hi,

I am trying to setup and use a Canon Selphy CP1500 on a Raspberry Pi 4B. Currently, it works well when it is added using network/dnssd but doesn't when connecting it in usb. Manually running the `ipp-usb` when connecting the printer indeed shows that something fails (log below).

I am using the latest Raspberry Pi OS / Debian 12 bookworm with cups 2.4.2 and ipp-usb 0.9.23 from the repository, and the printer firmware is up to date (version 1.0.2.0). I also tried compiling the current repo HEAD, and it gives the same result.

Do you have an idea of the problem?

```
+ HOTPLUG: added Bus 001 Device 026
+ PNP Bus 001 Device 026: added

  ===============================
+ Bus 001 Device 026: added SELPHY CP1500
  Device info:
    Ident:         04a9-3302-B122081513308562-Canon-SELPHY-CP1500
    Manufacturer:  Canon
    Product:       SELPHY CP1500
    SerialNumber:  B122081513308562
    MfgAndProduct: Canon SELPHY CP1500
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      init-delay = 0s
      request-delay = 0s
      http-connection = ""

  USB stack parameters
    authorized_default:          -1
    autosuspend:                 2
    blinkenlights:               N
    initial_descriptor_timeout:  5000
    nousb:                       N
    old_scheme_first:            N
    quirks:
    usbfs_memory_mb:             16
    usbfs_snoop:                 N
    usbfs_snoop_max:             65536
    use_both_schemes:            Y

  USB interfaces:
    Config Interface Alt Class SubClass Proto
       1       0      0   7      1       2
*      1       0      1   7      1       4
       1       1      0   255    0       0
*      1       1      1   7      1       4

  USB[0]: open: Bus 001 Device 026 Interface 0 Alt 1
  USB[1]: open: Bus 001 Device 026 Interface 1 Alt 1
> HTTP[091]: POST http://localhost:60000/ipp/print
> HTTP[091]: request body: got 462 bytes; closed
> HTTP[091]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[091]: connection 0 allocated
< HTTP[091]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[091]: response body: got 1129 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
! IPP: IPP: server-error-internal-error
> HTTP[092]: GET http://localhost:60000/eSCL/ScannerCapabilities
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[092]: connection 1 allocated
< HTTP[092]: GET http://localhost:60000/eSCL/ScannerCapabilities - 404 Not Found
< HTTP[092]: client has gone; draining response from USB
! ESCL: eSCL: HTTP status: 404 Not Found
- Bus 001 Device 026: resetting SELPHY CP1500
! USB[1]: recv: libusb_bulk_transfer: No such device (it may have been disconnected)
- Bus 001 Device 026: shutdown: 1 connections still in use
! USB[1]: recv: libusb_bulk_transfer: Input/Output Error
  USB[1]: connection released, 0 in use: --- ---
  USB[0]: closed
  USB[1]: closed
- Bus 001 Device 026: removed SELPHY CP1500
! PNP Bus 001 Device 026: Device doesn't implement print or scan service
```

alexpevzner, 2023-11-06:
Hi @bwolleb,

can I see logs from `/var/log/ipp-usb`?

bwolleb, 2023-11-06:
Of course, here is the full log, it seems there are all my unsuccessful attempts there. (This is from my laptop, not the raspberry pi, although I think it doesn't change anything since it is the same ipp-usb version).

[04a9-3302-B122081513308562-Canon-SELPHY-CP1500.log](https://github.com/OpenPrinting/ipp-usb/files/13272484/04a9-3302-B122081513308562-Canon-SELPHY-CP1500.log)

alexpevzner, 2023-11-06:
Can I ask you to find the following lines in the `ipp.go`, line 178:
```
        // Check response status
        if msg.Code >= 100 {
                err = fmt.Errorf("IPP: %s", goipp.Status(msg.Code))
                return
        }

```

Comment it out, rebuild and retry?

bwolleb, 2023-11-06:
I will, first thing I'll do tomorrow morning, thanks for your help!

alexpevzner, 2023-11-06:
OK, no problem.

And even if we will get `ipp-usb` happy with your printer, it is not guaranteed that CUPS will be happy as well...

bwolleb, 2023-11-07:
Ok nice, commenting those lines allowed ipp-usb to properly add the device, and CUPS was happy as well, I successfully printed two borderless photos. Here is the output of ipp-usb when the printer was detected:

```
  ===============================
  ipp-usb started in "debug" mode, pid=1592
  ctrlsock: listening at "/var/ipp-usb/ctrl"
+ HOTPLUG: added Bus 001 Device 008
+ PNP Bus 001 Device 008: added

  ===============================
+ Bus 001 Device 008: added SELPHY CP1500
  Device info:
    Ident:         04a9-3302-B122081513308562-Canon-SELPHY-CP1500
    Manufacturer:  Canon
    Product:       SELPHY CP1500
    SerialNumber:  B122081513308562
    MfgAndProduct: Canon SELPHY CP1500
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      init-delay = 0s
      request-delay = 0s
      http-connection = ""

  USB stack parameters
    authorized_default:          -1
    autosuspend:                 2
    blinkenlights:               N
    initial_descriptor_timeout:  5000
    nousb:                       N
    old_scheme_first:            N
    quirks:
    usbfs_memory_mb:             16
    usbfs_snoop:                 N
    usbfs_snoop_max:             65536
    use_both_schemes:            Y

  USB interfaces:
    Config Interface Alt Class SubClass Proto
       1       0      0   7      1       2
*      1       0      1   7      1       4
       1       1      0   255    0       0
*      1       1      1   7      1       4

  USB[0]: open: Bus 001 Device 008 Interface 0 Alt 1
  USB[1]: open: Bus 001 Device 008 Interface 1 Alt 1
> HTTP[000]: POST http://localhost:60000/ipp/print
> HTTP[000]: request body: got 462 bytes; closed
> HTTP[000]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[000]: connection 0 allocated
< HTTP[000]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[000]: response body: got 1129 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
> HTTP[001]: POST http://localhost:60000/ipp/faxout
> HTTP[001]: request body: got 463 bytes; closed
> HTTP[001]: body is small (463 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[001]: connection 1 allocated
< HTTP[001]: POST http://localhost:60000/ipp/faxout - 404 Not Found
< HTTP[001]: client has gone; draining response from USB
! IPP FaxOut probe failed: HTTP: 404 Not Found
> HTTP[002]: GET http://localhost:60000/eSCL/ScannerCapabilities
> HTTP[002]: body is chunked, sending as is
  USB[0]: connection allocated, 2 in use: a-- ar-
  HTTP[002]: connection 0 allocated
< HTTP[002]: GET http://localhost:60000/eSCL/ScannerCapabilities - 404 Not Found
< HTTP[002]: client has gone; draining response from USB
! ESCL: eSCL: HTTP status: 404 Not Found
> Canon SELPHY CP1500: _printer._tcp TXT record:
> Canon SELPHY CP1500: _ipp._tcp TXT record:
    air=none
    mopria-certified=2.1
    rp=ipp/print
    priority=50
    kind=photo
    PaperMax=<legal-A4
    URF=W8,SRGB24,V1.5,RS300,IS7,MT11,PQ4,OB9,IFU0,OFU0,CP99
    UUID=332f6000-1947-447e-b4d8-5c625af96216
    Color=T
    Duplex=F
    note=
    qtotal=1
    usb_MDL=SELPHY CP1500
    usb_MFG=Canon
    usb_CMD=URF
    ty=Canon SELPHY CP1500 HTTP
    product=(Canon SELPHY CP1500 HTTP)
    pdl=image/urf,application/octet-stream,image/jpeg,image/pwg-raster
    txtvers=1
    Fax=F
    Scan=F
> Canon SELPHY CP1500: _http._tcp TXT record:
> Canon SELPHY CP1500: _ipp-usb._tcp TXT record:
  DNS-SD: Canon SELPHY CP1500 (USB): trying
  USB[1]: connection released, 1 in use: ar- ---
  DNS-SD: FQDN: "msbproto.local"
  DNS-SD: FQDN: "msbproto.local"->"localhost"
  DNS-SD: +subtype: "_universal._sub._ipp._tcp"
+ DNS-SD: Canon SELPHY CP1500 (USB): publishing requested
  DNS-SD: Canon SELPHY CP1500 (USB): AVAHI_ENTRY_GROUP_REGISTERING
  USB[0]: connection released, 0 in use: --- ---
  DNS-SD: Canon SELPHY CP1500 (USB): AVAHI_ENTRY_GROUP_ESTABLISHED
  DNS-SD: Canon SELPHY CP1500 (USB): published
```

alexpevzner, 2023-11-10:
Hi @bwolleb,

could you please retest with the latest github version. Please, run it direcly from the build directory (`./ipp-usb`), so it will use quirks files from the source directory, not from the system installation.

If you confirm that the problem is resolved, the fix will be included into the next release

bwolleb, 2023-11-13:
Hi @alexpevzner, yes it still works using your latest commit on master. For the record, I still have to manually generate the ppd with the `driverless` command for borderless printing, but I guess it is another issue (maybe related to [this one](https://github.com/OpenPrinting/cups-filters/issues/492)). Thanks for your work!

alexpevzner, 2023-12-09:
Fixed in 0.9.24

closed 2023-12-09 by alexpevzner

switzel, 2024-02-21:
I'm glad I found this issue because I'm having the same issue with a CP-1300. However my printer did work reliably in the past, so I located the commit that introduced the code: it is 7d21f23 from Februrary 2020, which is present already in release 0.9.0 and there are no earlier releases on GitHub. Does somebody have an explanation, how it would have worked before (it was certainly via ipp, not Gutenprint or something else)?

alexpevzner, 2024-02-21:
Hi @switzel,

if problem in that commit, you can try `disable-fax = true` quirk (see `man ipp-usb` for description of quirk files).

If it help, let me know, I will add it to the known quirks, so with the next release it will be available for everybody

andi34, 2024-07-16:
@bwolleb could you tell how you've done the setup including the packages needed to be installed?
I've tried 
```sh
#!/bin/bash

if [ "$UID" != 0 ]; then
    echo "ERROR: Only root is allowed to execute the installer. Forgot sudo?"
    exit 1
fi

apt-get install autoconf build-essential libavahi-client-dev \
     libavahi-common-dev golang-go \
     libusb-1.0-0-dev -y

git clone https://github.com/OpenPrinting/ipp-usb.git
cd ipp-usb
make
make install
```
But doesn't compile and can't really fix it... Seems like missing package for go. Tried `go get -u github.com/jstemmer/gotags` but still complaining...

bwolleb, 2024-07-16:
@andi34 you should be good, I compile and install it automatically on the latest raspberry pi OS with the following script:
```
apt-get install -y git golang libusb-1.0-0-dev libavahi-client-dev
cd /tmp
git clone -b 0.9.24 https://github.com/OpenPrinting/ipp-usb.git
cd ipp-usb
go get -u github.com/jstemmer/gotags
make
make install
```
And it works really well. I also get the `gotags` error, but it is not fatal it seems, the make process continues normally and the compiled package works as expected.

andi34, 2024-07-17:
Great! Thanks a lot @bwolleb . Can I contact you if I need further instructions on setting the printer up? From the Ubuntu documentation I don't really get if the printer only needs to be plugged in and ipp-usb installed to set it up inside cups or if there are any further steps needed.

Garionhk, 2024-07-17:
@bwolleb Thanks for you script to install the new ipp-usb. 

I have that installed ipp-usb and also cups has reconized the printer.  However, it cannot print out and has an error message: 

Cannot print incompatible images or memory card not readable.

the ipp-usb log seems not sending data to printer.

! ESCL: eSCL: HTTP status: 404 Not Found
> Canon SELPHY CP1500: _printer._tcp TXT record:
> Canon SELPHY CP1500: _ipp._tcp TXT record:
    air=none
    mopria-certified=2.1
    rp=ipp/print
    priority=50
    kind=photo
    PaperMax=<legal-A4
    URF=W8,SRGB24,V1.5,RS300,IS7,MT11,PQ4,OB9,IFU0,OFU0,CP99
    UUID=d09aadb0-3e94-403b-8583-40f8dfececb8
    Color=T
    Duplex=F
    note=
    qtotal=1
    usb_MDL=SELPHY CP1500
    usb_MFG=Canon
    usb_CMD=URF
    ty=Canon SELPHY CP1500 HTTP
    product=(Canon SELPHY CP1500 HTTP)
    pdl=image/urf,application/octet-stream,image/jpeg,image/pwg-raster
    txtvers=1
    Fax=F
    Scan=F
> Canon SELPHY CP1500: _http._tcp TXT record:
> Canon SELPHY CP1500: _ipp-usb._tcp TXT record:
  USB[1]: connection released, 1 in use: ar- ---
  DNS-SD: Canon SELPHY CP1500 (USB): trying
  DNS-SD: FQDN: "photobooth.local"
  DNS-SD: FQDN: "photobooth.local"->"localhost"
  DNS-SD: +subtype: "_universal._sub._ipp._tcp"
  USB[0]: connection released, 0 in use: --- ---
+ DNS-SD: Canon SELPHY CP1500 (USB): publishing requested
  DNS-SD: Canon SELPHY CP1500 (USB): AVAHI_ENTRY_GROUP_REGISTERING
  DNS-SD: Canon SELPHY CP1500 (USB): AVAHI_ENTRY_GROUP_ESTABLISHED
  DNS-SD: Canon SELPHY CP1500 (USB): published
> HTTP[045]: POST http://localhost:60000/ipp/print
> HTTP[045]: request body: got 173 bytes; EOF
> HTTP[045]: body is small (173 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[045]: connection 1 allocated
< HTTP[045]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[045]: response body: got 8788 bytes; EOF
  USB[1]: connection released, 0 in use: --- ---
> HTTP[046]: POST http://127.0.0.1:60000/
> HTTP[046]: request body: got 614 bytes; EOF
> HTTP[046]: body is small (614 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[046]: connection 0 allocated
< HTTP[046]: POST http://127.0.0.1:60000/ - 200 OK
< HTTP[046]: response body: got 642 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
> HTTP[047]: POST http://localhost:60000/ipp/print
> HTTP[047]: request body: got 207 bytes; EOF
> HTTP[047]: body is small (207 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[047]: connection 1 allocated
< HTTP[047]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[047]: response body: got 165 bytes; EOF
  USB[1]: connection released, 0 in use: --- ---
> HTTP[048]: POST http://localhost:60000/ipp/print
> HTTP[048]: request body: got 714 bytes; EOF
[... 87 lines cut ...]
  HTTP[059]: connection 1 allocated
< HTTP[059]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[058]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[058]: response body: got 1130 bytes; EOF
  USB[0]: connection released, 1 in use: --- ar-
> HTTP[060]: POST http://localhost:60000/ipp/print
> HTTP[060]: request body: got 339 bytes; EOF
> HTTP[060]: body is small (339 bytes), prefetched before sending
  USB[0]: connection allocated, 2 in use: a-- ar-
  HTTP[060]: connection 0 allocated
< HTTP[059]: response body: got 338 bytes; EOF
  USB[1]: connection released, 1 in use: a-w ---
< HTTP[060]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[060]: response body: got 338 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
> HTTP[061]: POST http://localhost:60000/ipp/print
> HTTP[061]: request body: got 745 bytes; EOF
> HTTP[061]: body is small (745 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[061]: connection 1 allocated
< HTTP[061]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[061]: response body: got 1137 bytes; EOF
  USB[1]: connection released, 0 in use: --- ---

can you please help e about this matter?  Thank you

andi34, 2024-07-19:
@bwolleb thanks, got it working.
 Edges are off, any idea how to solve this?

[image]

bwolleb, 2024-07-19:
Nice photo @andi34 ! What driver are you using? The only configuration that worked for me for borderless printing was by using the `driverless` driver stack (generator from cups-filters) and the `PageSize: Postcard.Borderless` option.

andi34, 2024-07-22:
Random image from Google 🙈

Wrote everything together here with the adjusted PPD

https://gist.github.com/andi34/fbf0e8f4dba939c38a473a23c453eb10

Garionhk, 2024-11-29:
> @Garionhk <https://github.com/Garionhk> , I'm currently having the same
> issue as you. Did you ever solve It?

No. I didn't.

At the end, I just bought a second hand cp1300. 😅

andi34, 2024-11-29:
Check if updating the firmware of the printer helps.

Garionhk, 2024-11-29:
I returned the canon cp1500. Before that, I did installed new firmware.

In my case, I use raspberry pi and for some reason I need to use old
raspbian to allow my camera live feed to work.

I makes a Cantonese (Chinese) video explaining how this works and made a
ISO image.  It will be nice if there an ISO image directly to download to
save all installing hassles because it is an amazing project and really
want people to know about this.

Thank you for helping the community.

Eddy

7b7bruno, 2024-11-29:
I reflashed the newest firmware a few times while trying different things and for some reason It works pretty reliably for me now, but only when I print from the image viewer. When I try printing from CLI with lp, It throws the same `"Cannot print incompatible images or memory card not readable."`, so It seems It's a problem with CUPS. Does anyone know what the Raspberry PI OS image viewer does differently compared to the lp command, when printing images?

NullYing, 2025-03-04:
This PR can fix the issue.
https://github.com/OpenPrinting/ipp-usb/pull/96

This quirk is no longer needed.
https://github.com/OpenPrinting/ipp-usb/blob/master/ipp-usb-quirks/Canon.conf

[image]