OpenPrinting/ipp-usb #75: cant discover older HP OfficeJets at printer boot

Matze1224, 2024-02-12:
As ipp-usb got the Ubuntu default, we now utilise it instead of hplip. There quite some HP OfficeJet printers around with different series, where of the following I could reproduce the error:
HP OfficeJet Pro 8710
HP OfficeJet Pro 8730

There are no more models I can test but maybe there are more models poping up in the wild.
Newer models like the HP OfficeJet Pro 9010 are unaffected by this (at least I tested this model without the following problem).

The error occurs if the printer was powered off quite a time (think at least a day, tested it today after the weekend) and gets started after Linux is already up. After the printer successfully started, it wouldnt be detected by the system. After a printer restart, replugging of the usb cable or a simple restart the ipp-usb daemon, it will work. Trying the same scenario again, it works at the first printer start.

I could solve the problem by adding the following quirk:
```ini
[HP OfficeJet Pro 8710]
  init-reset = soft
```

**Logs:**

_error was the first start, success the second which worked ..._

[03f0-7a12-HP-OfficeJet-Pro-8730_error.log](https://github.com/OpenPrinting/ipp-usb/files/14236414/03f0-7a12-HP-OfficeJet-Pro-8730_error.log)
[03f0-7a12-HP-OfficeJet-Pro-8730_success.log](https://github.com/OpenPrinting/ipp-usb/files/14236415/03f0-7a12-HP-OfficeJet-Pro-8730_success.log)
[03f0-7c12-HP-OfficeJet-Pro-8710_error.log](https://github.com/OpenPrinting/ipp-usb/files/14236416/03f0-7c12-HP-OfficeJet-Pro-8710_error.log)
[03f0-7c12-HP-OfficeJet-Pro-8710_success.log](https://github.com/OpenPrinting/ipp-usb/files/14236417/03f0-7c12-HP-OfficeJet-Pro-8710_success.log)

_tried it with the second printer of the same model we have to test with the quirk applied_

[03f0-7c12-HP-OfficeJet-Pro-8710_fixed.log](https://github.com/OpenPrinting/ipp-usb/files/14236480/03f0-7c12-HP-OfficeJet-Pro-8710_fixed.log)

Looks like the problem is the following IPP response when the device is finally up:
> 12-02-2024 08:20:16: ! IPP: IPP: client-error-bad-request

Maybe the device isnt already initialised in this stage but dont know either.

I posted it as a ticket because it looks like a more generic problem with the HP OfficeJet series and maybe it could be solved better than the qurik I've used.

alexpevzner, 2024-02-12:
Hi @Matze1224,

thank you for reporting this issue.

I see, even on succes/fixed cases, device goes through the number of libusb_bulk_transfer timeout/device reset attempts.

Have you tried to play with the `init-delay` quirk?

Matze1224, 2024-02-13:
Thanks for the reply!

As the quirks arent well documented, I accidentially set `36ms` for the 8730 printers but it somehow worked in the end:

```ini
[HP OfficeJet Pro 8730]
  init-delay = 36
```
[03f0-7a12-HP-OfficeJet-Pro-8730_init-delay-36.log](https://github.com/OpenPrinting/ipp-usb/files/14257592/03f0-7a12-HP-OfficeJet-Pro-8730_init-delay-36.log)

`init-delay` doesnt look to work on the 8710 printers, as it repeated itself always - even if the printer was fully booted:

```ini
[HP OfficeJet Pro 8710]
  init-delay = 24000
```
[03f0-7c12-HP-OfficeJet-Pro-8710_init-delay-24000.log](https://github.com/OpenPrinting/ipp-usb/files/14257629/03f0-7c12-HP-OfficeJet-Pro-8710_init-delay-24000.log)

From the previous log files, I've calculated `36s` for the 8730 printers and `24s` for the 8710 printers, but maybe I should leave a bit more buffer there.

Using the `0.9.23-1ubuntu0.23.10.1` version backported from Ubuntu mantic to Ubuntu jammy on my test machine.

alexpevzner, 2024-02-22:
> As the quirks arent well documented

What's wrong with quirks documentation? I'd be glad to improve it with a help of your feedback

> I've calculated 36s for the 8730 printers and 24s for the 8710 printers, but maybe I should leave a bit more buffer there.

May I ask you to try to experiment with values smaller that these? Your calculations are probably assumes the fact, that while ipp-usb performs a series of retries, printer initialization goes its way. Actually, ipp-usb's retry may restart the printer initialization from some point.

I feel half-minute delay is a way too much and may be uncomfortable for users.

Matze1224, 2024-02-28:
For the quirks documention; I've got the impression there are more options than documented in the README. Maybe its because the options quickly evolve in the later versions and already handed with 3 versions. Currently, I see the `ignore-ipp-status` option which is undocumented, but looks good elsewhere - now found out there also documented in the manpage which is more complete than the README; good to know.

The 8710s work fine with the `init-reset = soft`. Using a `init-delay` keeps looping, but need to try lower values. Thought about the same you said with the 24 secs at testing. But I'm curious if the init-delay will help, as its negate the effect of the soft-reset I think. The "Device initialization timed out" needs around 5 secs for itself, so there is already some delay. Still testing if a delay helps, but it needs time (cause only happens if its off for multiple hours).

8730 printers make more trouble, as `init-reset` or `init-delay` wont help. No soft- or hardreset. Always anwsers IPP with `client-error-bad-request` except after a printer reboot. Maybe there is some IPP retrying or to wait for a DNSSD `_ipp._tcp` record appear? Can also test the brand new `ignore-ipp-status` quirk with the newest release.

alexpevzner, 2024-03-03:
So, finally:

* 8710 requires `init-reset = soft`
* 8730 requires more investigations

Correct?

alexpevzner, 2024-03-03:
I've added HP 8710 quirk, that you've discovered, to the base of known quirks. After the next update (which is planned soon), it will come to all users.

I don't close this issue, because workaround for 8730 is not found yet.