OpenPrinting/ipp-usb #84: "HP LaserJet Pro P1100 plus" BasicCaps Err

NullYing, 2024-10-18:
The HP LaserJet Pro P1100 plus is a printer but does not have scanning functionality.

BasicCaps bytes: [8 33 1 1 0 4 17 0], it's length less then 10
https://github.com/OpenPrinting/ipp-usb/blob/05cf67a81f1cf8d43372acd546cf1bc30e028b26/usbio_libusb.go#L554C14-L555C1

[image]

NullYing, 2024-10-18:
hp support: https://support.hp.com/cn-zh/product/details/hp-laserjet-pro-p1100-plus-series/2101440622

alexpevzner, 2024-10-18:
Hi @NullYing,

are you able to rebuild `ipp-usb` by yourself? If so, I'd prefer to experiment a little bit at your side (and with your help)...