OpenPrinting/ipp-usb #91: Canon Selphy CP1500 recognized but not printing.

7b7bruno, 2024-11-25:
I am having issues setting up a Canon Selphy CP1500 with ipp-usb and CUPS on a Raspberry PI 5(Raspberry PI OS Bookworm). 
When I first set up ipp-usb from the Apt package, ipp-usb threw an error. I then compiled the newest version from GitHub and It seemed to work. I successfully printed a CUPS test page, but after that I tried printing my own picture and the printer showed an error on the screen "Cannot print incompatible images or memory card not readable". Since then, It doesn't even respond at all, when I try printing a picture. I have tried rebooting and recompiling several times, but the result has not changed.
Here is the output from ipp-usb in debug mode:
```shell
 ===============================
  ipp-usb started in "debug" mode, pid=991
  ctrlsock: listening at "/var/ipp-usb/ctrl"
+ PNP Bus 003 Device 002: added

  ===============================
+ Bus 003 Device 002: added SELPHY CP1500
  Device info:
    Ident:         04a9-3302-C224011020331979-Canon-SELPHY-CP1500
    Manufacturer:  Canon
    Product:       SELPHY CP1500
    SerialNumber:  C224011020331979
    MfgAndProduct: Canon SELPHY CP1500
    BasicCaps:     print,scan,fax,http

  Device quirks:
    from [Canon SELPHY CP1500] (/usr/share/ipp-usb/quirks/Canon.conf:7)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      init-delay = 0s
      request-delay = 0s
      ignore-ipp-status= true
    from [Canon SELPHY CP1500] (/tmp/ipp-usb/ipp-usb-quirks/Canon.conf:7)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      init-delay = 0s
      request-delay = 0s
      ignore-ipp-status= true
    from [*] (/usr/share/ipp-usb/quirks/default.conf:3)
      blacklist = false
      usb-max-interfaces = 0
      disable-fax = false
      init-delay = 0s
      request-delay = 0s
      ignore-ipp-status= false
      http-connection = ""

  USB stack parameters
    authorized_default:          -1
    autosuspend:                 2
    blinkenlights:               N
    initial_descriptor_timeout:  5000
    nousb:                       N
    old_scheme_first:            N
    quirks:
    usbfs_memory_mb:             16
    usbfs_snoop:                 N
    usbfs_snoop_max:             65536
    use_both_schemes:            Y

  USB interfaces:
    Config Interface Alt Class SubClass Proto
       1       0      0   7      1       2
*      1       0      1   7      1       4
       1       1      0   255    0       0
*      1       1      1   7      1       4

  USB[0]: open: Bus 003 Device 002 Interface 0 Alt 1
  USB[1]: open: Bus 003 Device 002 Interface 1 Alt 1
> HTTP[000]: POST http://localhost:60000/ipp/print
> HTTP[000]: request body: got 462 bytes; closed
> HTTP[000]: body is small (462 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[000]: connection 0 allocated
< HTTP[000]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[000]: response body: got 1129 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
> HTTP[001]: POST http://localhost:60000/ipp/faxout
> HTTP[001]: request body: got 463 bytes; closed
> HTTP[001]: body is small (463 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[001]: connection 1 allocated
< HTTP[001]: POST http://localhost:60000/ipp/faxout - 404 Not Found
< HTTP[001]: client has gone; draining response from USB
! IPP FaxOut probe failed: HTTP: 404 Not Found
> HTTP[002]: GET http://localhost:60000/eSCL/ScannerCapabilities
> HTTP[002]: body is chunked, sending as is
  USB[0]: connection allocated, 2 in use: a-- a--
  HTTP[002]: connection 0 allocated
< HTTP[002]: GET http://localhost:60000/eSCL/ScannerCapabilities - 404 Not Found
< HTTP[002]: client has gone; draining response from USB
! ESCL: eSCL: HTTP status: 404 Not Found
> Canon SELPHY CP1500: _printer._tcp TXT record:
> Canon SELPHY CP1500: _ipp._tcp TXT record:
    air=none
    mopria-certified=2.1
    rp=ipp/print
    priority=50
    kind=photo
    PaperMax=<legal-A4
    URF=W8,SRGB24,V1.5,RS300,IS7,MT11,PQ4,OB9,IFU0,OFU0,CP99
    UUID=b649f780-754a-4bd3-809e-40f8dfeb9758
    Color=T
    Duplex=F
    note=
    qtotal=1
    usb_MDL=SELPHY CP1500
    usb_MFG=Canon
    usb_CMD=URF
    ty=Canon SELPHY CP1500 HTTP
    product=(Canon SELPHY CP1500 HTTP)
    pdl=image/urf,application/octet-stream,image/jpeg,image/pwg-raster
    txtvers=1
    Fax=F
    Scan=F
> Canon SELPHY CP1500: _http._tcp TXT record:
> Canon SELPHY CP1500: _ipp-usb._tcp TXT record:
  DNS-SD: Canon SELPHY CP1500 (USB): trying
  USB[1]: connection released, 1 in use: ar- ---
  DNS-SD: FQDN: "raspberrypi.local"
  DNS-SD: FQDN: "raspberrypi.local"->"localhost"
  DNS-SD: +subtype: "_universal._sub._ipp._tcp"
+ DNS-SD: Canon SELPHY CP1500 (USB): publishing requested
  DNS-SD: Canon SELPHY CP1500 (USB): AVAHI_ENTRY_GROUP_REGISTERING
  USB[0]: connection released, 0 in use: --- ---
  DNS-SD: Canon SELPHY CP1500 (USB): AVAHI_ENTRY_GROUP_ESTABLISHED
  DNS-SD: Canon SELPHY CP1500 (USB): published
> HTTP[003]: POST http://localhost:60000/ipp/print
> HTTP[003]: request body: got 207 bytes; EOF
> HTTP[003]: body is small (207 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[003]: connection 1 allocated
< HTTP[003]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[003]: response body: got 165 bytes; EOF
  USB[1]: connection released, 0 in use: --- ---
> HTTP[004]: POST http://localhost:60000/ipp/print
> HTTP[004]: request body: got 714 bytes; EOF
> HTTP[004]: body is small (714 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[004]: connection 0 allocated
< HTTP[004]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[004]: response body: got 1130 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
> HTTP[005]: POST http://localhost:60000/ipp/print
> HTTP[005]: request body: got 297 bytes; EOF
> HTTP[005]: body is small (297 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[005]: connection 1 allocated
> HTTP[006]: POST http://localhost:60000/ipp/print
> HTTP[006]: request body: got 744 bytes; EOF
> HTTP[006]: body is small (744 bytes), prefetched before sending
  USB[0]: connection allocated, 2 in use: a-- ar-
  HTTP[006]: connection 0 allocated
< HTTP[006]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[005]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[006]: response body: got 1130 bytes; EOF
  USB[0]: connection released, 1 in use: --- ar-
< HTTP[005]: response body: got 94 bytes; EOF
  USB[1]: connection released, 0 in use: --- ---
> HTTP[007]: POST http://localhost:60000/ipp/print
> HTTP[007]: request body: got 744 bytes; EOF
> HTTP[007]: body is small (744 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[007]: connection 0 allocated
< HTTP[007]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[007]: response body: got 1130 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
> HTTP[008]: POST http://localhost:60000/ipp/print
> HTTP[008]: request body: got 744 bytes; EOF
> HTTP[008]: body is small (744 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[008]: connection 1 allocated
< HTTP[008]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[008]: response body: got 1130 bytes; EOF
  USB[1]: connection released, 0 in use: --- ---
> HTTP[009]: POST http://localhost:60000/ipp/print
> HTTP[009]: request body: got 744 bytes; EOF
> HTTP[009]: body is small (744 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[009]: connection 0 allocated
< HTTP[009]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[009]: response body: got 1130 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
> HTTP[010]: POST http://localhost:60000/ipp/print
> HTTP[010]: request body: got 744 bytes; EOF
> HTTP[010]: body is small (744 bytes), prefetched before sending
  USB[1]: connection allocated, 1 in use: --- a--
  HTTP[010]: connection 1 allocated
< HTTP[010]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[010]: response body: got 1130 bytes; EOF
  USB[1]: connection released, 0 in use: --- ---
> HTTP[011]: POST http://localhost:60000/ipp/print
> HTTP[011]: request body: got 297 bytes; EOF
> HTTP[011]: body is small (297 bytes), prefetched before sending
  USB[0]: connection allocated, 1 in use: a-- ---
  HTTP[011]: connection 0 allocated
< HTTP[011]: POST http://localhost:60000/ipp/print - 200 OK
< HTTP[011]: response body: got 94 bytes; EOF
  USB[0]: connection released, 0 in use: --- ---
```

alexpevzner, 2024-11-26:
Hi @7b7bruno,

I don't thing this is the `ipp-usb` issue. Log looks pretty well. But `ipp-usb` is only the transport. If transport works well, but printer doesn't like the image it has received, printing will not work.

How do you use your printer? Have you explicitly added it to CUPS, or you print driverlessly, without explicit configuration? Did you specify any PPD when configuring your printer?

7b7bruno, 2024-11-29:
I reflashed the newest firmware a few times, while I was experimenting with It. Before I was printing without explicitly adding the printer to CUPS, but now I added It and It seems to work pretty reliably, but only through the Raspberry PI OS image viewer. When I try printing with the `lp` command, the printer throws the same `Cannot print incompatible images or memory card not readable`. Do you know what the Image viewer could be doing differently compared to the `lp` command?

alexpevzner, 2024-11-30:
Hi @7b7bruno,

how did you added the printer explicitly? Have you explicitly set a PPD file?

Hi @michaelrsweet,

it seems to be a cups-filters issue. The printer behaves differently when using temporary vs permanent queue.

Could you please have a look?

7b7bruno, 2024-11-30:
@alexpevzner , I set the driver to `Canon SELPHY CP1500 HTTP, driverless, cups-filters 1.28.17 (color)`. It was already in the list of drivers when I went to add the printer to CUPS.

NullYing, 2025-03-02:
I found that the Canon CP1500 can access the web via USB protocol on macOS, but it doesn't work when using ipp-usb.