OpenPrinting/ipp-usb #92: 7/1/2 printer

rwbot, 2024-11-29:
Hi, I'm trying to configure a Jiose J500 thermal printer. It's a USB only printer without AirPrint or network features, but I'm trying to convert it into such. 

I currently have it set up on cups via the web interface using the usb connection and with a modified PPD driver but am running into recurring issues depending on which device i'm printing from. 

Going down the rabbit hole led me to the ipp-usb section of the cups wiki, and upon trying to set it up I discovered it doesn't meet the 7/1/4 criteria for ipp-usb via  running `lsusb -v | grep -A 3 bInterfaceClass.*7` which gives
```
bInterfaceClass         7 Printer
bInterfaceSubClass      1 Printer
bInterfaceProtocol      2 Bidirectional
iInterface              0
```

But the fact i'm able to currently print *sometimes* via iPhone and iPad gives me some hope that there's some workaround. So...despite the requirement for a 7/1/4 interface, Is there a way to configure ipp-usb for a 7/1/2 type printer? Thanks in advance

alexpevzner, 2024-11-29:
Hi @rwbot,

I'm sorry, but no. ipp-usb is only for 7/1/4 printers...

closed 2024-11-29 by alexpevzner

zdohnal, 2024-12-02:
@rwbot I guess the reason why you can print via other devices is because cupsd is sharing the print queue associated with USB printer and the devices pick that up.