OpenPrinting/ipp-usb #93: Create an OCI container image of ipp-usb

tillkamppeter, 2024-12-17:
As we have OCI container images for CUPS and for the 4 retro-fitting Printer Applications at OpenPrinting, we also want to have an OCI container image of the IPP-over-USB daemon ipp-usb.

This way we can provide the full printing stack as OCI container images, ideally on DockerHub. This allows to add printing functionality to most immutable distributions, and also to distributions in general, using official packages of OpenPrinting.

alexpevzner, 2024-12-18:
Hi @tillkamppeter,

if you plan to put a new release tag on this project, please coordinate it with me, because I also plan to do in in a near feature.

Basically, I'm ready to the next release and waiting for feedback from @fabled under the #64 issue (which is expected to close 3 related issues).

tillkamppeter, 2024-12-18:
I am not intending to do a release for now. This feature request is not urgent, I posted it as a project idea for the upcoming [Winter of Code](https://winterofcode.tech/) which has its coding period on January 20, 2025 - February 20, 2025. If somebody creates the OCI container image then (or later), we will most probably do a release including this feature.

rudra-iitm, 2024-12-18:
To maintain consistency with other OpenPrinting (Rocks) OCI images, it would be ideal to use [Rockcraft](https://documentation.ubuntu.com/rockcraft/en/stable/) for creating the OCI images.