OpenPrinting/ipp-usb #95: Support logging to syslog

fabled, 2025-02-28:
It would be beneficial if logging to syslog instead of (or in addition to) files would be supported.

alexpevzner, 2025-02-28:
OK, accepted. Thanks a lot, @fabled, for your patience and collaboration!

fabled, 2025-03-07:
> OK, accepted. Thanks a lot, [@fabled](https://github.com/fabled), for your patience and collaboration!

Thanks!

I'm actually trying to debug an issue where a HP printer randomly drops the last page of a job. Usually in a duplex printout the last half of the page is empty. The printer errors the job, but it seems that on IPP level (based on Cups backend debug logs) everything seems OK. So I'm trying to figure out if something weird happens on lower USB levels. Or if its just the printer firmware broken.

fabled, 2025-03-07:
> The printer errors the job, but it seems that on IPP level (based on Cups backend debug logs) everything seems OK. So I'm trying to figure out if something weird happens on lower USB levels.

To clarify. Even if we send the same exact print data, it usually works ok, but randomly this issue will happen.

The exact data sent to printer is not identical because Go http will rework the chunked transfer-encoding chunks. Any idea if it would be easily possibly to change so that the HTTP body chunks sent are bit identical?

alexpevzner, 2025-03-07:
Changing this is not easy, and I actually depend on a fact that chunks sent to the printer are independent from the chunks received from the client, because it allows me to always finish the HTTP transaction correctly, even if client drops the connection in the middle and without need to buffer the entire client's data before forwarding to the printer.

However, data **actually** sent to the printer (at the real USB level) can be logged by enabling the **trace-usb** log level.

alexpevzner, 2025-03-07:
You may play a little bit with the **ipp-usb** predecessor, the **ippusbxd**  program (https://github.com/OpenPrinting/ippusbxd/commits/master/).

Unlike ipp-usb, this program just relays incoming TCP connection to the USB, without any modifications or interpretation.

fabled, 2025-03-13:
> You may play a little bit with the **ipp-usb** predecessor, the **ippusbxd** program (https://github.com/OpenPrinting/ippusbxd/).
> 
> Unlike ipp-usb, this program just relays incoming TCP connection to the USB, without any modifications or interpretation.

Integrating to something else is not feasible at this time.

However, I managed to finally figure out the issue in `wireshark`. So I'm now able to dump the usbmon interface, have wireshark stitch together the `Send-Document` data payload, verify http chunking, extract the file payload, and view the raster data in external viewer. And all this seems to be bit identically correct even in my error corner cases.

Turns out, that adding a 1 millisecond sleep in `func (iface *UsbInterface) Send` after a successful transmission of large URB. So smells definitely a hardware/firmware issue.  `request-delay` or any other tweak is not enough. It really seems to require throttling of the data transmitted. Though, I will need to verify that this indeed reliably fixes the issue. But the testing so far indicates that the issue is now gone.

The actual issue is that printing largeish raster jobs will abort randomly (usually about 10-20% of time). Typically the last page is left blank, but occasionally the print just aborts midway and arbitrary amount of pages can be missing from the print job.

fabled, 2025-03-14:
> Well. We can easily add an inter-URB delay quirk.

That might be an interesting / appreciated option.

> URB size can also be limited. To try, you can just add a couple of lines into the `(*UsbInterface)Send`:
> The only limitation, the resulting size must be a multiple of 1024, so we will not split USB wire packets even on the USB 3.x.
> 
> It will not break things, because HTTP stack doesn't assume that all the requested data will go away in a single request, it will not fail and will just come back with the remaining data bytes.

I did not realize this. But I actually modified the `go` `io.Copy` buffering to limit the http chunk and as a result the URB size to 2k. That did not fix the issue.

At this point it feels like an obscure buffering issue. If the USB buffers at the device end fill too much at wrong point in time, then the job would fail. Though, the other side effect of the delay is that `cups` can render more raster data in the streaming pipeline. So the delay possibly also alters the http chunks to be more likely large or maximum sized. So another wild guess for issues is if the raster page header gets split over multiple http chunks. But I have not verified this yet.

> BTW, does it happen only on a large jobs? What matters, the size of a single page or amount of pages?

This happens when sending raster jobs. It seems to not be tied to amount of pages. However, there seems to be some correlation with the payload size - e.g. printing data that was originally scanned seem to fail more likely than data that was generated directly from an editor. Possibly due to the scanned data containing more "noise" in the raster data.

The issue also does not happen always. Sending the same exact document payload succeeds about 50-95% of time, and fail the rest. Depending on luck, timing and the phase of moon ;)

> Some users around me complain that some printers (much more low-level that you use) fail to print a large, multipage jobs.
>
> Looks that the printer attempts to process a whole job at once, and if it has not enough resources, it just fails the entire job.

This depends. Generally the only document formats that can be "streamed" and printed on a page by page basis are the `urf` and `pwg-raster` raster formats. E.g. PDF cannot be rendered until the complete file is received by the printer - so printing will not begin until all data is sent.

Typically the printers advertise in `pdf-k-octets-supported` attribute with the maximum PDF size it accepts. The low end printers typically limit at some tens or a hundred of megabytes. The HP X677 advertises 250MB limit.

> May be you know, can CUPS split a large job into the sequence of smaller ones? It could help in both situations, like yours and like mine.

Not automatically to my understanding. If the PDF fails, the job fails. There is support in the ipp backend to retry as raster if the printer fails with format unsupported error.

IPP allows supporting multi-document jobs by send multiple `Send-Document` commands, but this is optional. The support for this is advertised by `multiple-document-jobs-supported` and even many enterprise things do not support it (including X677).

The automatically generated temporary cups queue or printers added with `lpadmin -p <print_queue_name> -v <URI> -E -m everywhere` will have the `pdf` as preferred format. And thus will have all the limitations.

Because of this, the only way to print large jobs (say 400+ pages of scanned / graphics containing stuff) is the raster format. For this reason, I print in this mode. The raster mode is enabled by configuring the printer with `lpadmin -p <print_queue_name> -v <URI> -E -m driverless:<URI>`.

It probably would be a good improvement to cups to honor `pdf-k-octets-supported` and to disable pdf as an option if the file size exceeds that. That would allow falling back to raster format automatically.

alexpevzner, 2025-03-14:
> That might be an interesting / appreciated option.

Done (at the GitHub master).
```
usb-send-delay = 1ms
```
I will add documentation when you confirm that it works. Unfortunately, I can't test by myself as currently I don't have a compatible printer near me.

fabled, 2025-03-14:
> Done (at the GitHub master).
> 
> ```
> usb-send-delay = 1ms
> ```

Thanks. I'll try this. My exact code adjustment was to sleep 1ms if the sent URB size is larger than 2048 bytes. The rationale was to not sleep after the small `\r\n` packets the go sends separately due to "fluash after chunk" behavior (https://github.com/golang/go/blob/master/src/net/http/transfer.go#L357-L359 and https://github.com/golang/go/blob/master/src/net/http/internal/chunked.go#L258-L260).

The `Send` pattern to USB seems to be as follows (assuming huge reads from local socket):
 1. The request writer gets wrapped as `bufio` at https://github.com/golang/go/blob/master/src/net/http/request.go#L663-L671
 2. `io.Copy` is called for IN:chunked and OUT:(chunked to bufio to usb) stack. This results in the default 32 kB copy buffer being used at https://github.com/golang/go/blob/master/src/io/io.go#L418.
 3. The reads happen pretty much according to what can be read from the input, this assumes large reads being done
 4. The read fills the `io.Copy` 32kB buffer.
 5. First the chunk size is written, which is buffered to bufio.
 6. The 32kB data is written. First the rest of bufio buffer is filled, and flushed. Resulting in 4 kB usb `Send` call.
 7. Rest of the io copy buffer gets directly sent to usb `Send`, typically 4-28kB.
 8. The end-of-chunk linefeed pair is sent and flushed, triggering 2 byte usb `Send` call.
 9. repeat from step 4 until end-of-chunked-http

So my logic was to just throttle the transfer speed.

I also noticed that different HP products have different USB receive speeds. Also, the timing between multiple steps 6 above vary depending on how fast the to-be-printed data is generated.

I'm wondering if the quirk should be similar to `request-delay` that the delay is guaranteed minimum between URBs sent (counting also the time the reset of the code is doing between the calls) instead of unconditional delay. Thoughts?

alexpevzner, 2025-03-14:
> I'm wondering if the quirk should be similar to request-delay that the delay is guaranteed minimum between URBs sent (counting also the time the reset of the code is doing between the calls) instead of unconditional delay. Thoughts?

Well, the good question indeed :-)

Honestly? Intuitively.

But after some reflection on it I think the following.

HTTP requests are rare, paced by the client and delays between them will be, most likely, quite long. So `request-delay` uses this logic to prevent this unneeded long delay if actually interface was not used for some time.

By contrast, USB send requests are frequent, a single HTTP request may contain hundreds of USB sends. Delays there are 1 ms or so. Smart logic, like I've used with `reqest-delay` will save only one small pause per a whole HTTP request.

So for simplicity I didn't bother here to optimize.

alexpevzner, 2025-03-14:
OK. I've also added the `usb-send-delay-threshold` quirk:
```
usb-send-delay-threshold = 2048
```
Please notice it takes into the account the actual transfer size (as returned by libusb after the transfer completion), not the requested size.

fabled, 2025-03-14:
> OK. I've also added the `usb-send-delay-threshold` quirk:

Thanks, this is enough to replicate the behavior of my local patch.

Though, I'm wondering why this works and things fails miserably if a delay is added after every URB. I'm wondering if the issue is caused that golang's http library sends the `\r\n` boundary in separate `Send`. And that if the condition is triggered by the printer's receive buffer being empty after this stand alone line break URB.

Do you have any idea what would be easiest way to alter the http library to not call flush after boundary?  I suppose I could just modify the go library code and test if my hypothesis is right.

I'll experiment with this a bit more next week to see if I get things working by adjusting the buffering.

alexpevzner, 2025-03-14:
> Do you have any idea what would be easiest way to alter the http library to not call flush after boundary?

You probably can, in the `(*UsbTransport)RoundTripWithSession` function call `outreq.Write()` (at the **usbtransport.go:480**) into the intermediate buffer, then copy it into the connection.

> I'll experiment with this a bit more next week to see if I get things working by adjusting the buffering.

A want to create a new release quickly, because latest (0.9.29) panics.

May be the better way would be if your publish your quirks file that works for now, as a temporary solution, and I'll make a new release in the next 1-2 days, but later, if you will be able to find the better approach, we will include it into the subsequent release.

alexpevzner, 2025-03-15:
> Again, sending huge raster data, cannot cache in memory.

I have a suspicion, which is hard to prove analytically (Go HTTP stack is complex and hard to read), that your HP printer fails when receives the chunk's trailing `\r\n` in the separate USB packet.

The delay that we've have introduced may have an effect of actually merging this data into the single USB packet. Who knows, how all this data transferring pipeline spread around involved goroutines?...

Probably the easiest way to try is to make the following modifications to the usbConn code:
1. add `bytes.Buffer` into the usbConn structure  
2. rename `(*usbConn)Write` into something else (say, `(*usbConn)doWrite`)
3. add `(*usbConn)flush` method, that actually sends data, collected in the buffer, added at the step (1) by calling `(*usbConn)doWrite` in loop (remember, `Send` may write out only part of data). `flush` must break the look and return an error, if `Send` returns an error.
4. add new implementation of the `(*usbConn)Write`, which writes data to the buffer and calls `flush` only if buffer exceeds, say, 16K
5. add call to the `flush` method into very beginning of the `(*usbConn)Read`. If `flush` returns an error, `Read` should terminate here with this error.

If it helps, it worse addition of the next quirk :)

> Also thinking if the UsbInterface should just call the quirk getters at init time to avoid parsing the value on every bulk transfer?

It actually doesn't parse at every GetXXX call. The cost of this call is only the map lookup, where parsed values are cached. I think, it is a reasonable tradeoff between simplicity and performance.

fabled, 2025-03-15:
> I have a suspicion, which is hard to prove analytically (Go HTTP stack is complex and hard to read), that your HP printer fails when receives the chunk's trailing `\r\n` in the separate USB packet. [...]

So I've been spending looking at the `usbmon` captures that show the kernel seen URBs. The delays have no affect on the packetization on kernel / sending side. When using `libusb` to send URBs the sent transfers are sent exactly as is and result in direct mapping to USB packets.

It is possible that the peripheral side internally converts the URBs to some pipe/fifo/uart and results in this.

So far I've seen three issues:
 1. Printing last page fails. (The behaviour without sending quirks, happens 5-40% of time.)
 2. Printer goes to error state. (If 1ms delay after every URB, high likeliness.)
 3. Printing fails earlier than last page. (Happens very rarely.)

I've seen none of the above happen with the delay after large packets only.

Since the other changes that made the `\r\n` get merged to the preceding URB (usbipp `Send` limits transmit to 4k) or following URB (golang library modified to not flush after chunk boundary) both had no effect the primary issue. The latter change also modified the final chunk terminator to get merged so it was not standalone. And the fact that even without delays we get successful prints, would indicate that this is not packetization related.

I'm probably going to stick with the current workaround and try to escalate at HP side since its clearly a device issue.

Perhaps the two final workaround tests might be:
 - add the small delay before sending the final end-of-chunking packet
 - try something like you suggest that to make sure only full 4k usb transfers are sent out until the very last end

I'll likely try this sometime next week.

alexpevzner, 2025-03-15:
I have added your quirks to the common collection, and probably tomorrow or at Monday will create a release.