OpenPrinting/ipp-usb #108: Containerization-friendly file/directory placement

tillkamppeter, 2025-04-09:
The `ipp-usb` daemon uses some files, for configuration, USB quirk rules, remembereing the use of the ports, and logging. For this, paths are had-coded in the `paths.go` file (comments removed):
```
package main
const (
        PathConfDir = "/etc/ipp-usb"
        PathConfQuirksDir = "/etc/ipp-usb/quirks"
        PathQuirksDir = "/usr/share/ipp-usb/quirks"
        PathProgState = "/var/ipp-usb"
        PathLockDir = PathProgState + "/lock"
        PathLockFile = PathLockDir + "/ipp-usb.lock"
        PathControlSocket = PathProgState + "/ctrl"
        PathProgStateDev = PathProgState + "/dev"
        PathLogDir = "/var/log/ipp-usb"
        PathLogFile = PathLogDir + "/main.log"
)
```
In a container, like a Snap or an OCI container there is often not the same file system available as in a classic operating system environment without containerization/sandboxing. Also applications can be shipped in immutable file systems, meaning that one cannot write into the directories in which the application itself is delivered.

Therefore there must be an easy way to redefine the places of the files which ipp-usb is using, so that one can easily package it in a containerized, snadboxed, and/or immutable format.

The Snap for example has the immutable file system with which ipp-usb is shipped in `/snap/ipp-usb/current/` and it can write files only into subdirectories of `/var/snap/ipp-usb/current/` and `/var/snap/ipp-usb/common/`, meaning that the file paths need to be appropriately redefined for the Snap build. Currently, we go the awkward way of modifying `paths.go` with a command line editor (in our case `perl`) before building ipp-usb in `snapcraft.yaml`.

To make it easier I suggest to add one or more (ideally all) of the following methods:
1. Build command line options
2. Command line options when starting the daemon `ipp-usb`
3. Environment variables to be read by `ipp-usb`
4. Entries in the config file `ipp-usb.conf` (requires (2) or (3) for `ipp-usb` to find the config file at least)

to change the paths/file names.

rudra-iitm, 2025-04-11:
LGTM

im0vishal, 2025-04-11:
> [@im0vishal](https://github.com/im0vishal) [@rudra-iitm](https://github.com/rudra-iitm) WDYT? Is this all what is needed or did I forget anything?

Seems all okay to me also.

alexpevzner, 2025-04-22:
@tillkamppeter,

I've added the following options:

```
   -path-conf-files-srch dir1[:dir2...]
        List of directories where configuration files (ipp-usb.conf)
	are searched (/etc/ipp-usb)

    -path-log-dir dir
        Path to the directory where log files (main.log and per-device
	<DEVICE>.log) are written (/var/log/ipp-usb)

    -path-lock-file file
        Path to the program's lock file (/var/ipp-usb/lock/ipp-usb.lock)

    -path-dev-state-dir dir
        Path to the directory where per-device state files are written
	(/var/ipp-usb/dev)

    -path-ctrl-sock file
        Path to the program's control socket
	(/var/ipp-usb/ctrl)

    -path-quirks-files-srch dir1[:dir2...]
        List of directories where quirks files (*.conf) is searched
	(/etc/ipp-usb/quirks:/usr/share/ipp-usb/quirks)
```

Please confirm that this is OK for you. And if so, I will proceed with the next release.

tillkamppeter, 2025-04-22:
Looks OK for me. But please edit the `snapcraft.yaml` first, removing the `perl`-based editing of `paths.go` and instead, let `ipp-usb` called with appropriate paths on the command line, by editing the scripts in `snap/local/`. Test-build and run the Snap.

Similar should be done for the rockcraft build of ipp-usb. @im0vishal @rudra-iitm perhaps you could do this part.

alexpevzner, 2025-04-22:
@tillkamppeter,

may I ask you to do this update to `snapcraft.yaml` by yourself? I'm absolutely not familiar with Snap...

alexpevzner, 2025-04-29:
Dear gentlemen, @tillkamppeter, @im0vishal, @rudra-iitm,

I want to release the `ipp-usb` as early as possible, as it contains important improvements for the Xerox B205/B210/B215 devices.

May I ask you to give some feedback soon? Or if in doubt, I can release it as is and fix possible issues in the next release.

closed 2026-10-26 by alexpevzner