OpenPrinting/ipp-usb #109: Add make/model to TXT record of DNS-SD broadcast

tillkamppeter, 2025-04-16:
The TXT record of my printer's DNS-SD broadcast (printer connected by network) is:
```
Service Type: _ipps._tcp
Service Name: Brother MFC-L8390CDW series
Domain Name: local
Interface: wlp0s20f3 IPv4
Address: BRW58CDC9DF8E51.local/192.168.68.116:443
TXT mopria-certified = 2.1
TXT print_wfds = T
TXT UUID = e3248000-80ce-11db-8000-94ddf83a5d08
TXT TLS = 1.3
TXT PaperMax = legal-A4
TXT kind = document,envelope,label,postcard
TXT URF = SRGB24,W8,CP1,IS19-20,MT1-3-4-5-8-11,OB10,PQ4,RS600,V1.5,DM1
TXT rfo = ipp/faxout
TXT TBCP = F
TXT Transparent = T
TXT Binary = T
TXT PaperCustom = T
TXT Scan = T
TXT Fax = T
TXT Duplex = T
TXT Copies = T
TXT Color = T
TXT usb_CMD = PJL,PCL,PCLXL,URF
TXT usb_MDL = MFC-L8390CDW series
TXT usb_MFG = Brother
TXT priority = 25
TXT adminurl = https://BRW58CDC9DF8E51.local./net/net/airprint.html
TXT product = (Brother MFC-L8390CDW series)
TXT ty = Brother MFC-L8390CDW series
TXT note = 
TXT rp = ipp/print
TXT pdl = application/octet-stream,image/urf,image/jpeg,image/pwg-raster,application/pdf
TXT qtotal = 1
TXT txtvers = 1
```

and if I connect the same printer via USB, the TXT record provided by `ipp-usb` is:
```
Service Type: _ipp._tcp
Service Name: Brother MFC-L8390CDW series (USB)
Domain Name: local
Interface: lo IPv4
Address: till-t14g5.local/127.0.0.1:60001
TXT air = none
TXT mopria-certified = 2.1
TXT rp = ipp/print
TXT priority = 50
TXT kind = document,envelope,label,postcard
TXT PaperMax = legal-A4
TXT URF = SRGB24,W8,CP1,IS19-20,MT1-3-4-5-8-11,OB10,PQ4,RS600,V1.5,DM1
TXT UUID = e3248000-80ce-11db-8000-94ddf83a5d08
TXT Color = T
TXT Duplex = T
TXT note = 
TXT qtotal = 1
TXT pdl = application/octet-stream,image/urf,image/jpeg,image/pwg-raster,application/pdf
TXT txtvers = 1
TXT adminurl = http://localhost:60001/net/net/airprint.html
TXT Fax = T
TXT rfo = ipp/faxout
TXT Scan = T
```

There are much less entries. Most importantly `ty`, `product`, `usb_MFG`, `usb_MDL`, and `usb_CMD` are missing. These allow a client (like a print dialog, and not only those of Linux, but of any other OS, too) to list available printers with make and model (and already discover many capabilities, including PDLs) without needing to do a `get-printer-attributes` IPP request.

Therefore it would be great if you could add these items.

To get a code example (unfortunately only in C), especially for `usb_CMD`, check the `ippusbxd` repo here on OpenPrinting.

alexpevzner, 2025-04-17:
Hi Till,

this is **very** strange. `ty`, `product`, `usb_MFG`, `usb_MDL` and `usb_CMD` could be missed, if `printer-make-and-model` or `printer-device-id` IPP printer attributes are missed. But there are also `usb_SER` and `usb_HWID` which are added inconditionally started from the `0.9.28`, and they are also missed...

This is, fop comparison, the TXT record from my Kyocera:
```
"air=none"
"mopria-certified=1.3"
"rp=ipp/print"
"priority=50"
"kind=document,envelope"
"PaperMax=legal-A4"
"URF=CP255,DM4,IFU0,IS19-20,OB1-10,PQ4,RS600,V1.4,W8"
"UUID=4509a320-00a0-008f-00b6-002507510eca"
"Color=F"
"Duplex=T"
"note=Living Room"
"qtotal=1"
"usb_MDL=ECOSYS M2040dn"
"usb_MFG=Kyocera"
"usb_CMD=PCLXL,PostScript Emulation,PCL5E,PJL"
"ty=ECOSYS M2040dn"
"product=(ECOSYS M2040dn)"
"pdl=application/octet-stream,application/pdf,image/tiff,image/jpeg,image/urf,ap
"txtvers=1" 
"adminurl=http://localhost:60000/airprint" 
"Fax=F" 
"Scan=T" 
"usb_SER=VCF9192281" 
"usb_HWID=0482&069d"
```

So I would appreciate if you could publish here the `ipp-usb` logs for this device (see `/var/log/ipp-usb/`).

And what version of the `ipp-usb` are you using?

alexpevzner, 2025-04-17:
> What I did is stop ipp-usb, unplug printer from USB, remove all old logs...

Thanks! This is exactly what I wanted!

The problem was is that Brother returns textWithLanguage for `printer-make-and-model` etc while `ipp-usb` (onerously) expected only the bare text.

> ipp-usb is 0.9.28

I wonder, why is so old? The current version is 0.9.30...

> release of the distro today.

My congratulations to you and Canonical folks!

alexpevzner, 2025-04-17:
FIXED. Please, confirm.

tillkamppeter, 2025-04-17:
> > ipp-usb is 0.9.28
> 
> I wonder, why is so old? The current version is 0.9.30...
>

Feature Freeze for Ubuntu 25.04 was loooong 2 months ago ...
 
> > release of the distro today.
> 
> My congratulations to you and Canonical folks!

Thank you very much.

tillkamppeter, 2025-04-17:
FF for Ubuntu 25.04 was Feb 20 or so, missing 0.9.29 by a week ...

tillkamppeter, 2025-04-17:
The current snapshot does not build for me. Does it require a newer version of goipp?

alexpevzner, 2025-04-17:
It comes with the proper version of goipp (it is called *vendoring* in go terms), so in theory it should not depend on the external version of goipp.

How can I reproduce this problem?

tillkamppeter, 2025-04-17:
With just `make` I get
```
gotags -R . > tags
go build -ldflags "-s -w" -tags nethttpomithttp2 -mod=vendor
# github.com/OpenPrinting/goipp
vendor/github.com/OpenPrinting/goipp/message.go:267:7: undefined: Formatter
make: *** [Makefile:13: all] Error 1
```
and `make clean` does not work.

alexpevzner, 2025-04-17:
Missed commit :(

FIXED

tillkamppeter, 2025-04-17:
Great, it seems to work now! Thank you very much!

Here is what I get with my printer:
```
Service Type: _ipp._tcp
Service Name: Brother MFC-L8390CDW series (USB)
Domain Name: local
Interface: lo IPv4
Address: till-t14g5.local/127.0.0.1:60001
TXT air = none
TXT mopria-certified = 2.1
TXT rp = ipp/print
TXT priority = 50
TXT kind = document,envelope,label,postcard
TXT PaperMax = legal-A4
TXT URF = SRGB24,W8,CP1,IS19-20,MT1-3-4-5-8-11,OB10,PQ4,RS600,V1.5,DM1
TXT UUID = e3248000-80ce-11db-8000-94ddf83a5d08
TXT Color = T
TXT Duplex = T
TXT note = 
TXT qtotal = 1
TXT usb_MDL = MFC-L8390CDW series
TXT usb_MFG = Brother
TXT usb_CMD = PJL,PCL,PCLXL,URF
TXT ty = Brother MFC-L8390CDW series
TXT product = (Brother MFC-L8390CDW series)
TXT pdl = application/octet-stream,image/urf,image/jpeg,image/pwg-raster,application/pdf
TXT txtvers = 1
TXT adminurl = http://localhost:60001/net/net/airprint.html
TXT Fax = T
TXT rfo = ipp/faxout
TXT Scan = T
TXT usb_SER = E82414G4N583607
TXT usb_HWID = 04f9&0501
```
It is not exactly the same as when network-connected but it is much more usefull now and assures a much higher compatibility with clients and a better representation in user interfaces.

alexpevzner, 2025-04-17:
Perfect!

So looks like #108 is the only issue to be resolved before the new release.

closed 2025-04-17 by alexpevzner