OpenPrinting/ipp-usb #110: OCI/ROCK Container: Avahi sees printer as remote service not as local service

tillkamppeter, 2025-04-17:
When running the (immutable) OCI container of ipp-usb, from PR #106, from https://github.com/im0vishal/ipp-usb/tree/immutable-ipp-usb, commit 44b2745f ("Updated ReadMe to include runtime flags for data persistence"), started with the command lines
```
sudo docker volume create ipp-usb-data
sudo docker run -d --network host -v /dev/bus/usb:/dev/bus/usb:ro -v ipp-usb-data:/var/lib/ipp-usb --device-cgroup-rule='c 189:* rmw' --name ipp-usb ipp-usb:latest
```
the avahi-daemon on the host system sees ipp-usb as a remote service and not as a local service. So running `ippfind --local` does not list the printer but `ippfind --remote` lists it.

Especially client tools then do not replace the network host name (`XXX.local`) by `localhost`, and if ipp-usb is providing the printer only on the local machine (`lo` network interface) the printer cannot be accessed as the URI `ipp://XXX.local:60000/ipp/print` is advertised and used, while `ipp://localhost:60000/ipp/print` works.

This can be caused by the container running its own avahi-daemon and so the host's avahidaemon receiving the service from the container's avahi-daemon and not directly.

In avahi (I am using 0.8.0), the function `avahi_server_is_service_local()` in the file `avahi-core/server.c` of the Avahi source code is actually determining whether a service is local or not. 

The function is left by the `return 0;` in these lines:
```
    e = find_entry(s, interface, protocol, key);
    avahi_key_unref(key);
    if (!e)
        return 0;
```
and therefore the service is see as not local. This I found out by applying the attached patch to the Avahi source code and looking for `XXX` in the `/var/log/syslog` file and I saw `XXX6` but not `XXX10` right afterwards.

[check-local.patch.txt](https://github.com/user-attachments/files/19798404/check-local.patch.txt)

tillkamppeter, 2025-04-17:
@im0vishal @rudra-iitm Could you also have a look here?

alexpevzner, 2025-04-17:
> the avahi-daemon on the host system sees ipp-usb as a remote service and not as a local service. So running ippfind --local does not list the printer but ippfind --remote lists it.

It seems to be Avahi bug. In the loopback-only mode, `ipp-usb` uses the "localhost` name and the loopback interface index for announcing itself. Regarding of that, if system has at least one real network interface, Avahi silently replaces "loopback" with the "XXX.local" hostname.

tillkamppeter, 2025-04-17:
The fact that Avahi considers a service local or remote is independent of the host name it reports.

If I run ipp-usb as classic installation (DEB package) or as the Snap, the DNS-SD record as I see with `avahi-browse` or `avahi-discover` also reports the `XXX.local` host name but `ippfind --local` lists the service anyway. With this, Avahi reporting the service as local, CUPS/libcups correct the `XXX.local` host name back to `localhost` when accessing.

Only if I use the OCI container the service gets listed by `ippfind --remote` and not by `ippfind --local`.

alexpevzner, 2025-04-17:
What exactly it means, does Avahi considers service local or not?

alexpevzner, 2025-04-17:
> My feeling is that the OCI container is introducing its own avahi-daemon and this makes the service looking remote.

Me too. After quick looking to the Avahi source, I think Avahi sets this flag only for the entries it manages by itself (i.e., for the services, registered in this instance of the Avahi daemon).

tillkamppeter, 2025-04-17:
So @im0vishal @rudra-iitm could you modify **all** the OpenPrinting OCI container images (CUPS, ipp-usb, Printer Applications) so that they do not run their own avahi-daemon but register their services (printers) on the host's avahi-daemon, like the Snaps do.

alexpevzner, 2025-04-17:
What if we tweak `ippfind` to consider services that use one of the `127.0.0.1` or `::1` addresses as local, regardless of the `AVAHI_LOOKUP_RESULT_LOCAL` flag?

tillkamppeter, 2025-04-17:
@alexpevzner Now I realize what "local" actually is for Avahi, it is just a service registered on the avahi-daemon with which we are actually talking ...

tillkamppeter, 2025-04-17:
> What if we tweak `ippfind` to consider services that use one of the `127.0.0.1` or `::1` addresses as local, regardless of the `AVAHI_LOOKUP_RESULT_LOCAL` flag?

I see now that Avahi also reports the network interface to the browse callback. One could check whether it is `lo` and consider this local, too. But this would need to be done for every client, not only for `ippfind`. So we need to fix at several places.

Ideal would be that Avahi does this extra check (and also finds out about the hosts own name to be able to consider services from non-"lo" interfaces as local).

Could you report a bug on Avahi?